nkolban/esp32-snippets #490: Any direction to read from a BLE 4.0 keyboard into ESP32

volome, 2018-04-21:
Hi thanks for all this contribution, its wonderful and I am very thankful for it.

I need a direction to be able to read characters on a BLE 4.0 keyboard into ESP32.
Yes I have played with the samples files, still I need some more help.

Thanks

chegewara, 2018-04-21:
If i understand you want to use standard bluetooth keyboard to connect with esp32 and on esp32 read it.

To simplify code we can just assume that keyboard has all settings valid and dont have to check it or parse report map. This is how i would do it:
- when you scan find device that have appearance as keyboard 
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEHIDDevice.h#L21

- find all characteristics with UUID 0x2a4d,
- now from here find characteristic that value of descriptor 0x2908 is: (value && 0xff) == 0x01; its input report characteristic, register for notify on it
- or maybe even easier: find all characteristics 0x2a4d and there most likely is only one with descriptor 0x2902, register for notify on it
- parse every value received as notification as follow (there should be always 8 bytes):
- byte 0: flags, says if shift, ctrl, alt etc are pressed or not
- byte 1, reserved, should be always 0x00,
- bytes 2-7 are keys that are actually pressed (up to 6 keys at the same time), values are NOT ascii code
- decode bytes 2-7, those are arrays to decode key values:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/SampleKeyboardTypes.h#L82-L399

volome, 2018-04-22:
Thank you very much chegewara

Will try it out

Thanks

kellertobias, 2020-06-14:
Has anyone completed something here in the meantime? I am trying to connect a Bluetooth Keyboard with a Trackpoint to an ESP32 and forward it to my computer (Building a Bluetooth to USB HID Gateway in order to use the same bluetooth peripherals with multiple computers on a USB switcher).

chegewara, 2020-06-14:
> Has anyone completed something here in the meantime?

Yes

> Building a Bluetooth to USB HID Gateway in order to use the same bluetooth peripherals with multiple computers on a USB switcher

Not possible, because esp32 does not have native usb. You will need external chip that will works as usb hid (esp32 s2?).

kellertobias, 2020-06-14:
My plan was to use an esp32 for the bluetooth part and the heavy lifting (e.g. keymapping or maybe even some shortcuts for e.g. some mqtt smarthome stuff and keyboard macros) and then connect one arduino leonardo per computer.

however, I've tried your HID Keyboard Client (https://github.com/chegewara/esp32-hid-keyboard-client) and it doesn't find my Thinkpad Bluetooth Keyboard at all. Just a list of "BLE Advertised Device found", but none has the Address of my Keyboard. Any suggestions?

boernsen-development, 2022-08-31:
@kellertobias Did you solve your issue?

I am planning exactly the same project, i.e. using an ESP32 + USB-HID-capable device (maybe Arduino Micro) as an interface to connect an Apple Magic Trackpad (Bluetooth only) via USB so that I can use it without installing any driver, add some custom gestures, and easily switch computers with a single USB hub.

The Apple Magic Trackpad only supports Bluetooth Classic 2.0 (no BLE!), so I guess I will be facing the same issue than you pairing the Trackpad with the ESP32. Did it work for you in the end?