nkolban/esp32-snippets #491: BLEAdvertisementData.getPayload() compile error

rweller59, 2018-04-22:
Having some issues transmitting expected advertising data so tried grabbing the payload. I get the following error compiling:
  'components/cpp_utils/./BLEAdvertising.h:38:14: error: 'std::__cxx11::string 
   BLEAdvertisementData::getPayload()' is private
   std::string getPayload();               // Retrieve the current advert payload.'

Code looks like this:
	pService->start();
	//BLEAdvertising* pAdvertising = pServer->getAdvertising();
	pAdvertising = pServer->getAdvertising();
	BLEAdvertisementData adData;
	adData.setName(deviceManager::getMacID()); // sets the 'local name'
	adData.setShortName("shortname");
	std::string data = "id data";
	adData.setServiceData((BLEUUID)(uint16_t)0x20, data.c_str());
	adData.setManufacturerData("mfg_001");
	ESP_LOGI(LOG, "BLE advertising payload: %s", adData.getPayload().c_str());
	pAdvertising->setAdvertisementData(adData);
	pAdvertising->start();

chegewara, 2018-04-22:
Probably you have issue here:
`adData.setServiceData((BLEUUID)(uint16_t)0x20`
0x20 is uint8_t but you need uint16_t

rweller59, 2018-04-22:
doesn't the (uint16_t)0x20 cast it properly?

chegewara, 2018-04-22:
Im not sure, maybe youre right. Just its something i would check first. You can try to create `BLEUUID uuid = (uint16_t)0x20;` and you will see if it will throws error or not.