nkolban/esp32-snippets #494: Please don't include internal header files of bluedroid of esp-idf

TianHao-Espressif, 2018-04-23:
Hi, @nkolban ,
There's a minor thing should be corrected.

`find . -name "*.cpp" |xargs grep -nr "\<gatt_api.h\>" *`
`/cpp_utils/BLEServer.cpp:21:#include <gatt_api.h>`
It should include `esp_gatts_api.h` or `esp_gattc_api.h`

Due to I cleanup the `component.mk` of `esp-idf/component/bt/`(the MR is ready in our internal GIT server, still not merged), the internal header files of bluedroid will be invisible to application. Application should only include the header files which prefix is `esp_`. 
Could you modify this to use `esp_xxx` APIs?

nkolban, 2018-04-23:
No problem my friend.  We'll change it ASAP.

kosso, 2018-12-08:
I just tried installing and testing the ESP32 BLE Arduino library and am getting error: `fatal error: gatt_api.h: No such file or directory`

I assume this is related to this issue? 
Any ideas?

chegewara, 2018-12-08:
Yes, update library. `gatt_api.h` header name has been changed a long time ago to `esp_gatt_api.h`.

closed 2018-12-08 by chegewara

kosso, 2018-12-08:
OK. The problem that just lost me some hours today is that the Arduino IDE Library Manager is only showing and installing up to version `0.4.7`.   Also the Git Repo 'Releases' section is only up to `0.4.7`. (https://github.com/nkolban/ESP32_BLE_Arduino/releases) 

Whereas the current master of the library is at version `1.0.1`.

https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/library.properties#L2

I will get the Zip from Github.