nkolban/esp32-snippets #496: Enhancement: Ability to perform a BLEScan without blocking

nkolban, 2018-04-24:
The current implementation of BLEScan has a `start()` method that takes the duration for which scanning is to be performed.  The `start()` method blocks until that duration has completed.  While on the surface this sounds just fine, there is a request for alternate semantics.  When we perform a BLEScan() we can register a callback that will be invoked for each device found.  This is great ... and the callback can request that the scan stop.  But what if we want to scan indefinitely?

Obviously we can perform a BLEScan::start() for a huge duration (or 0?) and the scan will simply proceed forever ... however, that then ties up the task in which the start() was called.  The request was to provide an option which would make a call to `start()` asynchronous.  That means that when `start()` is called, it would return immediately and just keep on calling the callback when new devices were detected. To accommodate this, we have added a new optional parameter to start.  The signature of start is now:

```
BLEScanResults BLEScan::start(uint32_t duration, void (*scanCompleteCB)(BLEScanResults))
```

The `scanCompleteCB` is optional.  If **not** specified or supplies as `nullptr`, the call to `start()` will block for the duration and return the set of results.   If `scanCompleteCB` is supplied as a function reference, then the call to `start()` will return immediately and after scanning has been completed, `scanCompleteCB()` will be invoked to indicate that scanning has completed.

A sample called `SampleAsyncScan.cpp` has been supplied to illustrate.

bot1131357, 2019-08-15:
Hi @nkolban 

I tried running SampleAsyncScan.cpp and I get the following output on the console:

`Async Scanning sample starting
About to start scanning for 10 seconds
Now scanning in the background ... scanCompleteCB() will be called when done.
Tick! - still alive
Advertised Device: Name: AnotherESP32, Address: 30:ae:a4:25:02:b2, txPower: 3
Advertised Device: Name: , Address: 01:72:43:a4:54:06, manufacturer data: 0600010920024fab7c3260b8639546b5de3048644f26c7d4bf7c3b58d3
Advertised Device: Name: , Address: 77:37:d3:97:2f:27, manufacturer data: 4c001005131c1958eb
Advertised Device: Name: , Address: 42:6b:19:66:8c:8e, manufacturer data: 4c0007190102200ba90f010000025a73fbf64d07113e126194eba09557
Advertised Device: Name: , Address: 6d:15:ae:be:0c:e5, manufacturer data: 4c000c0e003632377b93a73926864a83fe741005011cde0316
Advertised Device: Name: , Address: 49:fc:8d:65:1b:dd, manufacturer data: 4c001005031c0ae27d
Advertised Device: Name: , Address: 3c:43:c9:09:a5:ec, manufacturer data: 06000109200237b36d9fcbc707a1bb2f4421608b95258cd45081d2dcf0
Advertised Device: Name: fenix 5 APAC, Address: f7:73:bb:1c:e3:c5, manufacturer data: 87000aec, serviceUUID: 6a4e3e10-667b-11e3-949a-0800200c9a66
Advertised Device: Name: , Address: 7b:65:32:b8:1b:4f, manufacturer data: 4c0010050118395d87
Advertised Device: Name: , Address: 5e:d1:43:1d:74:04, manufacturer data: 4c00100503184a5cf0
Tick! - still alive
Tick! - still alive
Tick! - still alive
Tick! - still alive
Tick! - still alive
Tick! - still alive
Tick! - still alive
Tick! - still alive
Tick! - still alive
Tick! - still alive
Scan complete!
We found 10 devices
Tick! - still alive
Tick! - still alive
Tick! - still alive
Tick! - still alive
Tick! - still alive
Tick! - still alive
Tick! - still alive
Tick! - still alive
Tick! - still alive`

The dump does work for the first time, but not for subsequent scans. 

(Thanks for your contribution on esp32. You are an inspiration!)

bot1131357, 2019-08-15:
In the meantime, I am using freeRTOS to create another task, as suggested [here](https://esp32.com/viewtopic.php?t=4980#p23416) by chegewara.

chegewara, 2019-08-15:
@bot1131357 Check option in menuconfig tofor scan duplicates.

bot1131357, 2019-08-19:
@chegewara : is that a problem caused by scan duplicates? Also, do you know how I can access menuconfig from esp32-arduino? 

In any case, I'm happy to report that moving the blocking code into a separate thread works. I just need to include `FreeRTOS::sleep(1);` in my loop() function.

chegewara, 2019-08-21:
@bot1131357 Here you can change if you want or not to see duplicates:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEScan.h#L52

abianchargy, 2021-02-09:
Hi!
I've working with this great function but I've had some issues. When It is working for a while without while loop, an error appears. 

[...]
Async Scanning sample starting
abort() was called at PC 0x401695d3 on core 0

Backtrace: 0x40091448:0x3ffcf580 0x40091679:0x3ffcf5a0 0x401695d3:0x3ffcf5c0 0x4016961a:0x3ffcf5e0 0x40168b0d:0x3ffcf600 0x40168890:0x3ffcf620 0x40168f41:0x3ffcf640 0x401693eb:0x3ffcf660 0x40169435:0x3ffcf690 0x400d26ef:0x3ffcf6b0 0x400d2783:0x3ffcf6d0 0x400d2805:0x3ffcf700 0x400d2867:0x3ffcf730 0x400d1cf6:0x3ffcf820 0x40100bf5:0x3ffcf870 0x400fbd66:0x3ffcf890 0x4008e089:0x3ffcf8c0

Rebooting...
[...]

If I change the Start function by BLEScan::start(uint32_t duration, bool is_continue), It works correctly (with blocking)
So, I don't understan why is this happening. 
Thank you!

chegewara, 2021-02-09:
https://github.com/me-no-dev/EspExceptionDecoder