nkolban/esp32-snippets #498: Question: How to tear down BLE Client connection correctly?

Tuet, 2018-04-24:
Hi, within an own class for a BLE device I connect to the device and in the end I build up the following hierarchy:
BLEClient* pBLEClient = BLEDevice::createClient()
  BLEREmoteServer* pBLEService = pBLEClient->getService("...")
    BLERemoteCharacteristic* pBLECharaA = pBLEService->getCharacteristic("...")
      pBLECharaA->registerForNotify(pCallbackA)

Then, I receive a onDisconnect event via my class that is public of BLEClientCallbacks. The question is now how to handle this and which and how should I destroy/delete the objects that I can reconnect later? I tried using delete pBLECharaA, then delete pBLEService. Do I have to delete all characteristic pointers myself? Do I have to unregister the notifications first? Or is there some kind of auto-delete that I just have to delete pBLEClient? 
Currently, it either crashes the CPU with some sort of heap-Errors or I am not able to connect again if I just don't delete anything. Please give me an advise how to do it correctly. Thank you very much for the very great BLE library so far!

chegewara, 2018-04-26:
Im not good at C++ and garbage collecting, but i think you can setup BLERemoteServis* and BLERemoteCharacteristic* as global variables and reuse on every connection, unless you want to have multiple connections at the same time.

Tuet, 2018-04-26:
Hi, of course I can reuse the pointers to class instances, but the problem is that the getService and getCharacteristic functions seems to allocate some resources on the heap as well as in the lower BLE stack areas. When I just try to reconnect to the same device then I get messages like "BT: application already registered.". So what I am looking for is a function that de-register the stuff I have got via getService and getCharacteristics to release the resources properly, so that I can re-register them later correctly. I tried to do so by invoking a delete on the pointers I have got in the hope that the destructor is called and does the clean up. Unfortunately I am not a C++ expert and maybe I am just doing something simple quite wrong...

chegewara, 2018-04-26:
This may be another bad idea, but you can try to create task. In this task create BLEClient, service, characteristics etc, all you need during connectection. Then on disconnect you can try to `vTaskDestroy()` that task. I have no idea how this will affect heap and stack, but worth to try.

Tuet, 2018-04-26:
I solved it for me! Short answer first:

1. When a onDisconnect() callback comes in, you have to manually invoke the disconnect() function of BLEClient.
2. After that, the instance of the BLEClient can be deleted. The destructor within this function seems to clean up all the hierarchy of many BLERemoteService and BLERemoteCharacteristics by itself. Don't try to delete these instances manually, otherwise a heap corruption will appear. It seems that most of the resources get released this way.
3. You can now connect and recreate the hierarchy again - without issues like "..already registered".

Long answer:
I am doing all this with Arduino, out of an Arduino library I am writing. I don't know much about creating tasks and due to the nature of the BLE implementation I try to call all BLE functions out of the main loop. As soon as I call any BLE function from within a BLE callback, everything crashes or does not work. 

However, it seems I understood the nkolban BLE stack a little bit more: When I tried to delete any BLERemoteService manually the system crashed with heap corruption. In the exception decoder I realized that the destructor of BLEREmoteService still tried to destruct the BLERemoteCharacteristics, that I already deleted myself. It seems that there is already a mechanism in nkolban's stack that tears down all the BLERemoteService and BLERemoteCharacteristics instances. That should be triggered on BLEClient-Level, but a call to disconnect() seems to be usefull as well.

chegewara, 2018-04-26:
You just saved me a lot of time and tests.

biccius, 2021-01-17:
Hi,

i'm running a BLE client on a ESP32-PICO-D4 (revision 1) with these settings:

[image]

The snippet code i'm testing is the following and the purpose is to init ble device, scan for a specific ble advertising name, connect to the service, read the answer of a specific characteristic and de-init ble device.

`#include "Arduino.h"
#include "BLEDevice.h"

#define CONNECT_TIMEOUT_MS 5000
#define DISCONNECT_TIMEOUT_MS 5000

static BLEUUID 	  				serviceUUID	  ("49535343-fe7d-4ae5-8fa9-9fafd205e455");
static BLEUUID    				charUUID_read ("49535343-1e4d-4bd9-ba61-23c647249616");
static BLEUUID    				charUUID_write("49535343-8841-43f4-a8d4-ecbe34729bb3");
static const char 				UUID_NAME[] = "XXXXXXXXXXXXXXXXXXXXXX";
static boolean 					connected = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;
static BLERemoteService* 		pRemoteService;
static BLEAdvertisedDevice* 	myDevice;
static boolean doConnect = false;
static boolean fScanInProgress = false;

bool ble_initialized = false;
volatile bool commandSequenceComplete = false;
volatile bool messageReceived = false;
volatile bool fScanFinished = false;
volatile uint8_t foundBLEDevices = 0;

char TX_BUFFER[50];
char RX_BUFFER[256];
uint8_t RX_INDEX = 0;
uint8_t TEMP_INDEX = 0;
uint8_t RX_LENGTH = 0;

static BLEScan*    pBLEScan;
static BLEClient*  pClient;

void printHeapStatus()
{
	Serial.print("RAM: "); Serial.print( ESP.getFreeHeap() ); Serial.println();
}

class MyClientCallback : public BLEClientCallbacks
{

  void onConnect(BLEClient* pclient)
  {
	  Serial.println("onConnect: OK");
	  connected = true;
  }

  void onDisconnect(BLEClient* pclient)
  {

	  if (fScanInProgress) BLEDevice::getScan()->erase(pclient->getPeerAddress());

	    printHeapStatus();
	    Serial.println("onDisconnect: OK");
	    Serial.println("pClient->disconnect();");

	  	pClient->disconnect();

	    printHeapStatus();

	  	connected = false;

  }
[... 553 lines cut ...]
*** Added device:  XXXXXXXXXX
*** Added device: 
*** Added device: 
*** Added device: 
scan complete!
Device(s) found: 5
Create client...
RAM: 172664
Set callback...
RAM: 172664
Connecting...
onConnect: OK
.....

it seems that I have a loss of heap of approximately 3k
I think the problem is that I don't call the destructors of the following objects

```
pClient->~BLEClient();	//CRASH
pRemoteService->~BLERemoteService();	//CRASH
```

but if I do I get a crash

where am i going wrong?

chegewara, 2021-01-17:
> 		  	myDevice = new BLEAdvertisedDevice(advertisedDevice);
> MyClientCallback* my_callback = new MyClientCallback();
> MyAdvertisedDeviceCallbacks* my_advertisedcallback = new MyAdvertisedDeviceCallbacks();

For sure you are having memory leak here. I did not read full code, but i am guessing you never delete any of these.

biccius, 2021-01-17:
Hi chegewara,

wow, thank you for your super fast answer!

myDevice is already deleted in the function closeConnections();

added my_advertisedcallback->~BLEAdvertisedDeviceCallbacks();   and    my_callback->~BLEClientCallbacks();  inside closeConection that is executed in the deinit phase but i'm still having heap leakage

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
----------------- BOOT ---------------
RAM: 220808
Started @240Mhz
change_CpuFrequencyMhz: 160
change_CpuFrequencyMhz: 80
init_BLE: OFF
init_WiFi: OFF
init_pins()
init_vref
init_i2c()
init_RTC
init_accelerometer()
init_display
RAM: 220292
--------------------------------------
RAM: 220292
BLEDevice::init()
RAM: 129776
pBLEScan: ok
RAM: 129356
Scan in progress...
FOUND -->  XXXXXXXXXX
*** Added device: XXXXXXXXXX
*** Added device: 
*** Added device: 
*** Added device: 
*** Added device: 
*** Added device: 
scan complete!
Device(s) found: 6
Create client...
RAM: 126556
Set callback...
RAM: 126556 
Connecting...
onConnect: OK
RAM: 122480
Waiting connection...
Write command :0116D
INCOMING: :0110020000000000000     length:20
INCOMING: 00000000000000000000     length:20
INCOMING: 00000060021000000002     length:20
INCOMING: 30001000000000004001     length:20
INCOMING: 06518183012550000005     length:20
INCOMING: A     length:1
PACKETS DECODED
Write command :01403A
INCOMING: :0140000000002600000     length:20
INCOMING: 0000720201218111216     length:19
PACKETS DECODED
Write command :01D0101540001539
INCOMING: :01D0101540001500015     length:20
INCOMING: 00007020200001100037     length:20
INCOMING: 00011000150000702020     length:20
INCOMING: 00011000370001365535     length:20
INCOMING: 6553500005eeeeeBB     length:17
PACKETS DECODED
Write command :0116D
RAM: 118868
pClient->disconnect();
RAM: 119460
wait...
RAM: 122160
onDisconnect: OK
pClient->disconnect();
RAM: 122116
closeConnections();
RAM: 123404
pBLEScan->clearResults();
RAM: 125312
delete myDevice;
RAM: 125520
BLEDevice::deinit(false);
RAM: 211468
pRemoteCharacteristic->~BLERemoteCharacteristic(); 
RAM: 211748
free(pRemoteCharacteristic);
RAM: 216528
free(pRemoteService);
RAM: 216708
my_callback->~BLEClientCallbacks();
RAM: 216708
my_advertisedcallback->~BLEAdvertisedDeviceCallbacks();
RAM: 216708
UNIT_TEST: BLE scan terminated
--------------------------------------
RAM: 216708
BLEDevice::init()
RAM: 126424
pBLEScan: ok
RAM: 126180
Scan in progress...
*** Added device: 
*** Added device: 
*** Added device: 
FOUND -->  XXXXXXXXXX
*** Added device: XXXXXXXXXX
*** Added device: 
*** Added device: 
scan complete!
Device(s) found: 6
Create client...
RAM: 123436
Set callback...
RAM: 123436
Connecting...
onConnect: OK
RAM: 119360
Waiting connection...
Write command :0116D
INCOMING: :0110020000000000000     length:20
INCOMING: 00000000000000000000     length:20
INCOMING: 00000060021000000002     length:20
INCOMING: 30001000000000004001     length:20
INCOMING: 06518183012550000005     length:20
INCOMING: A     length:1
STATUS PACKETS DECODED
Write command :01403A
INCOMING: :0140000000002600000     length:20
INCOMING: 0000720201218111216     length:19
CTP PACKETS DECODED
Write command :01D0101540001539
INCOMING: :01D0101540001500015     length:20
INCOMING: 00007020200001100037     length:20
INCOMING: 00011000150000702020     length:20
INCOMING: 00011000370001365535     length:20
INCOMING: 6553500005eeeeeBB     length:17
CALIB PACKETS DECODED
Write command :0116D
RAM: 115740
pClient->disconnect();
RAM: 116468
wait... 
RAM: 118972
onDisconnect: OK
pClient->disconnect();
RAM: 118928
closeConnections();
RAM: 120284
pBLEScan->clearResults();
RAM: 122192
delete myDevice;
RAM: 122400
BLEDevice::deinit(false);
RAM: 208332
pRemoteCharacteristic->~BLERemoteCharacteristic(); 
RAM: 208616
free(pRemoteCharacteristic);
RAM: 208908
free(pRemoteService);
RAM: 213576
my_callback->~BLEClientCallbacks();
RAM: 213576
my_advertisedcallback->~BLEAdvertisedDeviceCallbacks();
RAM: 213576
UNIT_TEST: BLE scan terminated
--------------------------------------
RAM: 213576
BLEDevice::init()
RAM: 123288
pBLEScan: ok
RAM: 123044
Scan in progress...
*** Added device: 
*** Added device: 
FOUND -->  XXXXXXXXXX
*** Added device: XXXXXXXXXX
*** Added device: 
*** Added device: 
*** Added device:

chegewara, 2021-01-17:
Scan object is a single instance, so you dont have to delete it. 
Scanned devices you can clear anytime you want.
pClient should be deleted by library, but you can try to delete it (most likely it will crash app).
My suggestion is to create variable/object for any callback you are passing. That way you dont have to worry about memory leak from that, because you will have only one instance which you can pass instead of creating `new Callback()`.

biccius, 2021-01-20:
The only instances i cannot delete are pRemoteService and pClient
I call 'delete pRemoteService' and 'delete pClient' but what i have is an error like this:

`delete(pRemoteService);
assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free

Backtrace: 0x4009456c:0x3ffcf020 0x4009479d:0x3ffcf040 0x4010e0db:0x3ffcf060 0x400941e1:0x3ffcf090 0x40085c72:0x3ffcf0b0 0x40087731:0x3ffcf0d0 0x4000bec7:0x3ffcf0f0 0x40090b9f:0x3ffcf110 0x400ddb5b:0x3ffcf130 0x400db857:0x3ffcf150 0x400dbf47:0x3ffcf170 0x400dc322:0x3ffcf190 0x400dc341:0x3ffcf1b0 0x400d1bf1:0x3ffcf1d0 0x400d235c:0x3ffcf1f0 0x400d3bb2:0x3ffcf230 0x400d3bcd:0x3ffcf250 0x400d3ec1:0x3ffcf270 0x400e0ecd:0x3ffcf2c0 0x40090c81:0x3ffcf2e0

Rebooting...
ets Jun  8 2016 00:22:57`

This step of destroying used instances is done after OnDisconnect has been executed.
I'm not really familiar with C++ but I'd say that 'delete' call the destructor and free lists of elements previously allocated
What's wrong with that?

chegewara, 2021-01-25:
pClient is deleted by library after disconnection, dont remember about remote services, probably too.