nkolban/esp32-snippets #499: BLE Bonding and Whitelist

medamarko, 2018-04-24:
Hi all ive seen that security for BLE has been introduced, but i could not find a way to bond device and keep it in whitelist. Is this available in library?
Thanks!

chegewara, 2018-04-24:
After bonding device is stored in NVS by bluedroid stack. Recently has been added some API in esp-idf to retrieve info about whitelisted devices but i did not study it yet and isnt added to this library. In this library bonding is working and secure connection with PIN should work, and here is function to allow to connect only whitelisted devices:
https://github.com/nkolban/esp32-snippets/blob/17d31202823dfd23aa147cf86f6a76530c80d489/cpp_utils/BLEAdvertising.cpp#L110

medamarko, 2018-04-24:
Thanks on fast answer, so connetion seems to work i can debug server as well. But unfortunatly write to char is not working until i get this message:
E (125993) BT: l2cble_process_conn_update_evt: Error status: 59 only after this message i get successuful write to char. To get this message it may take few minutes.

medamarko, 2018-04-24:
Changing log didnt helped, for some reason it start to work after one of timeouts in l2cap of bluedroid. Im trying to get more info.