nkolban/esp32-snippets #500: Raw advertising data

HanvanHulst, 2018-04-25:
Hi,

Is it possible to use an ESP32 to scan and gather raw advertising data?
At this moment i use a raspberry pi with HCIDUMP -Rx to get the following data;

> 0000: 04 3e 2b 02 01 00 00 84  34 49 0e f0 ec 1f 02 01  .>+.....4I......
  0010: 06 0f 09 41 69 72 20 4d  65 6e 74 6f 72 20 50 72  ...Air Mentor Pr
  0020: 6f 0b ff 12 21 00 8a 1a  df 39 20 00 16 d8        o...!....9 ...

> 0000: 04 3e 0f 02 01 04 00 84  34 49 0e f0 ec 03 02 0a  .>......4I......
  0010: 00 c9                                             ..

> 0000: 04 3e 2b 02 01 00 00 84  34 49 0e f0 ec 1f 02 01  .>+.....4I......
  0010: 06 0f 09 41 69 72 20 4d  65 6e 74 6f 72 20 50 72  ...Air Mentor Pr
  0020: 6f 0b ff 12 21 00 8a 1a  df 39 20 00 16 c4        o...!....9 ...

> 0000: 04 3e 0f 02 01 04 00 84  34 49 0e f0 ec 03 02 0a  .>......4I......
  0010: 00 c4                                             ..

> 0000: 04 3e 2b 02 01 00 00 84  34 49 0e f0 ec 1f 02 01  .>+.....4I......
  0010: 06 0f 09 41 69 72 20 4d  65 6e 74 6f 72 20 50 72  ...Air Mentor Pr
  0020: 6f 0b ff 12 21 00 80 1a  fa 39 22 00 1d d8        o...!....9"...

I would like to store this raw BLE advertising data in a database.
The data is from a Air Mentor Pro and contains PM2.5,PM10,CO2,TVOC, TEMP and humidity data.

Is there an example program to scan for raw data? or should i switch over to esp idf?

Kindly regards Han

chegewara, 2018-04-25:
Lets start from a joke:
https://github.com/nkolban/esp32-snippets/tree/master/BLE/scanner
I think its oldest ble code that mr. Neil posted.

Now more serious. With this example and few more lines of code you have all you need:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/SampleScan.cpp

I thought its done, but it requires to add few more lines here:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEAdvertisedDevice.cpp#L237

HanvanHulst, 2018-04-25:
Sorry to ask i am quite new to the ESP board, but it seams that you use the ESP IDF environment to write code for the board and net like me the Arduino IDE.

Wat is by the way the best environment to write code for the ESP8266 and the ESP32 including all kind of sensors?

regards Han

chegewara, 2018-04-25:
From my point of view both are good. And arduino can library for less common sensors. I will make required changes so you can get and use raw advertising data, i already have it was required for my other (abandoned) project, i just need to prepare PR.

HanvanHulst, 2018-04-26:
Great thanks for your help. I am looking forward to it

chegewara, 2018-04-27:
Added in PR #507

HanvanHulst, 2018-04-30:
I found a solution and it seems to work fine; i still have to cleanup the code

 
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      CapAddress = advertisedDevice.getAddress().toString().c_str();
      CapData = String(advertisedDevice.toString().c_str()).substring(String(advertisedDevice.toString().c_str()).indexOf("data:")+6);
      CapName = advertisedDevice.getName().c_str();
Serial.println(" CO2 : " + String(strtoul(CapData.substring(4,8).c_str(),0,16)));
          Serial.println(" PM2.5 : " + String(strtoul(CapData.substring(8,12).c_str(),0,16)));
          Serial.println(" PM10 : " + String(strtoul(CapData.substring(12,16).c_str(),0,16)));
          Serial.println(" CO : " + String(strtoul(CapData.substring(16,18).c_str(),0,16)));
          Serial.println(" O3 : " + String(strtoul(CapData.substring(18,20).c_str(),0,16)));

chegewara, 2018-04-30:
You choice, code is ready anyway.

hansmbakker, 2018-12-04:
@HanvanHulst this issue seems solved, could you please close it to keep the issue list clean?

@chegewara