nkolban/esp32-snippets #502: BLE Gateway - Impossible to use WiFi and BLE at the same time

Brunez3BD, 2018-04-25:
Hi, as in the title I'm trying to create a BLE gateway. The problem is that if I am already connected to the WiFi the BLE part doesn't work.

Here is the [main code](https://pastebin.com/B1gk90bZ), see also my [FlowerCare library](https://github.com/Brunez3BD/ESP32_FlowerCare) if necessary.

The following is the terminal output I'm able to obtain:
```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11404
entry 0x40078a28
Create flora 1
Create flora 2
Create flora 3
Connecting to FASTWEB-Brunelli
..Connected!
Setup done!
Get data flora 1
E (35770) BT: btc_gattc_call_handler()
```

Brunez3BD, 2018-04-27:
After some testing I located the "error". The program blocks inside `FC_1->getData()`, at the call to [`_BLEClient->connect(*_addr)`](https://github.com/Brunez3BD/ESP32_FlowerCare/blob/4344fa324403ddfa0ca7f11ed24936828e804011/src/FlowerCare_BLE.cpp#L68) inside `FlowerCare_BLE.cpp`. 
Going deeper I found that it blocks inside `BLEClient.cpp` when calling [`m_semaphoreOpenEvt.wait("connect")`](https://github.com/nkolban/esp32-snippets/blob/756e8d80b1b553e7f2ba6c27d2ceecec7cf023b3/cpp_utils/BLEClient.cpp#L121). The problem is that despite I located the error I don't know what to do, can you help me?

chegewara, 2018-04-27:
I would start with commenting out this line and see what would happend:
`m_semaphoreOpenEvt.wait("connect")`

This wont fix the issue, maybe even cause some other issues but it can give you more info.

Brunez3BD, 2018-05-16:
I don't know if it is a good idea, but I can try it. I was hoping that someone else encountered this problem before and resolved it. I know that there is some applications that use BLE and Wifi at the same time (like [this](https://github.com/pcbreflux/espressif/tree/master/esp32/app/ESP32_gatts_mqtt_gateway)) but with arduino core I'm unable to get it working without modify the "standard" BLE library for arduino core

chegewara, 2018-05-17:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/SampleClientWithWiFi.cpp

johnhmacleod, 2018-06-01:
I did run into an issue with BLE+WiFi and found that ensuring my BLE connection was complete before connecting to WiFi avoided the problem.  That was a while ago (6 months perhaps) and I haven't retested since.  Doing that, my BLE->WiFi gateway now runs happily for hours on end!