nkolban/esp32-snippets #503: libbt.a issue and new 3.0 version

PhilColbert, 2018-04-25:
updated the idf to the latest 3.0 version and compiling gets this error 

CC build/xtensa-debug-module/trax.o
AR build/xtensa-debug-module/libxtensa-debug-module.a
LD build/AirWhere_v4.2.elf
C:/msys32/esp/esp-idf/AirWhere_v4.2/build/arduino\libarduino.a(BluetoothSerial.o):(.literal._ZL10esp_spp_cb18esp_spp_cb_event_tP18esp_spp_cb_param_t+0x34): undefined reference to `esp_bt_gap_set_scan_mode'
C:/msys32/esp/esp-idf/AirWhere_v4.2/build/arduino\libarduino.a(BluetoothSerial.o):(.literal._ZL10esp_spp_cb18esp_spp_cb_event_tP18esp_spp_cb_param_t+0x38): undefined reference to `esp_spp_start_srv'
C:/msys32/esp/esp-idf/AirWhere_v4.2/build/arduino\libarduino.a(BluetoothSerial.o):(.literal._ZN15BluetoothSerial5writeEh+0x0): undefined reference to `esp_spp_write'
C:/msys32/esp/esp-idf/AirWhere_v4.2/build/arduino\libarduino.a(BluetoothSerial.o):(.literal._ZN15BluetoothSerial5beginE6String+0x20): undefined reference to `esp_spp_register_callback'
C:/msys32/esp/esp-idf/AirWhere_v4.2/build/arduino\libarduino.a(BluetoothSerial.o):(.literal._ZN15BluetoothSerial5beginE6String+0x24): undefined reference to `esp_spp_init'
C:/msys32/esp/esp-idf/AirWhere_v4.2/build/arduino\libarduino.a(BluetoothSerial.o): In function `esp_spp_cb(esp_spp_cb_event_t, esp_spp_cb_param_t*)':
C:/msys32/esp/esp-idf/AirWhere_v4.2/components/arduino/libraries/BluetoothSerial/src/BluetoothSerial.cpp:153: undefined reference to `esp_bt_gap_set_scan_mode'
C:/msys32/esp/esp-idf/AirWhere_v4.2/components/arduino/libraries/BluetoothSerial/src/BluetoothSerial.cpp:153: undefined reference to `esp_spp_start_srv'
C:/msys32/esp/esp-idf/AirWhere_v4.2/build/arduino\libarduino.a(BluetoothSerial.o): In function `BluetoothSerial::write(unsigned char)':
C:/msys32/esp/esp-idf/AirWhere_v4.2/components/arduino/libraries/BluetoothSerial/src/BluetoothSerial.cpp:153: undefined reference to `esp_spp_write'
C:/msys32/esp/esp-idf/AirWhere_v4.2/build/arduino\libarduino.a(BluetoothSerial.o): In function `BluetoothSerial::write(unsigned char const*, unsigned int)':
C:/msys32/esp/esp-idf/AirWhere_v4.2/components/arduino/libraries/BluetoothSerial/src/BluetoothSerial.cpp:153: undefined reference to `esp_spp_write'
C:/msys32/esp/esp-idf/AirWhere_v4.2/build/arduino\libarduino.a(BluetoothSerial.o): In function `BluetoothSerial::flush()':
C:/msys32/esp/esp-idf/AirWhere_v4.2/components/arduino/libraries/BluetoothSerial/src/BluetoothSerial.cpp:153: undefined reference to `esp_spp_write'
C:/msys32/esp/esp-idf/AirWhere_v4.2/build/arduino\libarduino.a(BluetoothSerial.o): In function `BluetoothSerial::begin(String)':
C:/msys32/esp/esp-idf/AirWhere_v4.2/components/arduino/libraries/BluetoothSerial/src/BluetoothSerial.cpp:153: undefined reference to `esp_spp_register_callback'
C:/msys32/esp/esp-idf/AirWhere_v4.2/components/arduino/libraries/BluetoothSerial/src/BluetoothSerial.cpp:153: undefined reference to `esp_spp_init'

the functions are defined in 

components\arduino\tools\sdk\include\bluedroid\esp_spp_api.c
components\arduino\tools\sdk\include\bluedroid\api\esp_spp_api.h

but for some unknown reason not getting compiled into the new libbt.a?

Any ideas ?

thanks

chegewara, 2018-04-25:
This is better place to ask this question/open this issue:
https://github.com/espressif/arduino-esp32/issues

Now my question: when did you update arduino-esp32 stack?

PhilColbert, 2018-04-25:
Well it worked before I updated the esp idf.

I had been copying in a btlib version, arduino esp32 stack was done couple weeks ago.

Using as a component, just seems to be like the required , I dont think its arduino issue, more of the fact the new versions of the esp_spp_api isnt being compiled into btlib.a for some weird reason ?

chegewara, 2018-04-25:
Some time ago in esp-idf stack bt.h has been renamed to esp_bt.h. It was not merged into arduino for some time. Its small chance it is merged now, because 2 days ago esp-idf get status release v3.0-rc1. I see that arduino-esp32 last time has been updated 9 days ago, so its long shot.

PhilColbert, 2018-04-25:
Thanks,

just updated the arduino core and its done the same thing.

So its not compiling the functions from the esp_spp_api  into the btlib.a function :(

I dont know why not, confused!

Thanks for the reply tho

chegewara, 2018-04-25:
mr @me-no-dev will be more helpful in this matter.

PhilColbert, 2018-04-25:
Hi, thanks for the reply.
 
I downloaded the entire thing again and set it up, and it worked this time.... bizarre, sorry for bothering :)

Thanks

PhilColbert, 2018-04-25:
Getting further, its not crashing now like it did before v3.0 was added.... which is a step in the right direction, also more heap is good.

Unfortunately theres some memory corruption going on now....

The error message "E (4050) BT: btc_gatts_arg_deep_copy 9 no mem" comes out as its trying to assign a variable for the length of the characteristic and this length variable is in the corrupted memory location...... 

E (3634) BLE-: current free heap: 82408, minimum ever free heap: 82408
*********************1
I (3637) BTDM_INIT: BT controller compile version [48877dc]

I (3643) system_api: Base MAC address is not set, read default base MAC address                             from BLK0 of EFUSE

E (4050) BT: btc_gatts_arg_deep_copy 9 no mem

Waiting a client connection to notify...
Heap after start BLE
E (4059) BLE-: current free heap: 11088, minimum ever free heap: 11056
AirWhr[e 0;04302m00I0  (c4o06nf1)i gBurluatetiooon thcSoemprliaetle: d,ES lP_eSt                            sP P_goI NI!T
_EVT

Any ideas ?

Thanks

closed 2018-05-26 by PhilColbert

Rajkumar-007, 2020-10-13:
> hmm... I pulled the latest IDF and have no issues with compilation. Are you use you have BT and SPP enabled in menuconfig?

Thank you so much, its solved esp32 bluetooth undefined errors and perfectly working!!! Path is (menuconfig->component config-> Bluetooth->bluedroid enble->SPP  and BT).