nkolban/esp32-snippets #504: read characteristics from multiple services?

programmer131, 2018-04-26:
in connectToServer() function inside ble_client example, is it possible to subscribe to multiple notifications from different services?
BLERemoteService* pRemoteService1 = pClient->getService(serviceUUID_1);
BLERemoteService* pRemoteService2 = pClient->getService(serviceUUID_2);
when I load this code, esp32 finds first service/characteristic and then reset always.

chegewara, 2018-04-26:
Yes, it is possible. There is one thing, you register for notify in BLERemoteCharacteristic not in BLERemoteService. When you have 2 services then your code is good, then for each service you need to find characteristic, check if that characteristic canNotify and register for notify.

programmer131, 2018-05-02:
it works. thanks.

closed 2018-05-02 by programmer131