nkolban/esp32-snippets #505: Client loses connection and stop (/hang) Need advise

JeroenBeemster, 2018-04-26:
Hi There

I am using ESP32_BLE_Arduino library on a esp-idf environment trying to read multiple (10) BLE-servers
I am able to read a BLE server using the client. 
There are a lot of Error passing through and everything is still working 
I do not care about these error but on different moments the connection got lost and everything stops.
And that is what I do not like (/cannot have)

The places with odd errors and still working fine are: 
----
Code:	
pRemoteService->getCharacteristics();  
or  
Result:
E (3678) BLERemoteCharacteristic: esp_ble_gattc_get_all_descr: ESP_GATT_NOT_FOUND
E (3678) BLERemoteCharacteristic:
E (3678) BLERemoteCharacteristic:

Code: 
pClient->disconnect();
Result:
E (19898) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0016

The places wich stop ESP32 to work are (not always, but certainly within 20 times):
----
Code: 
std::string value = pRemoteCharacteristic->readValue();   	
Result:
E (26488) BT: GATTC_Read GATT_BUSY conn_id = 3
E (26488) BT: No pending command
E (26498) BT: No pending command
E (35048) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0013

Code: 
BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
Result:
E (103488) BT: btc_gattc_cback transfer failed
E (103578) BT: No pending command
E (112008) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0013

Things already tried:
-Latest version from esp-idf, 
-Latest version from ESP32_BLE_Arduino
-Using isConnected() before executing code.

I expect that it is due to losing the connection during execution 

question
----
So my major question is. How to avoid stopping the client executing. (catch all the issues and just continue)
Minor question. How to reduce errors that might cause the problem

Additional question: How to show the messages that are in the library: ESP_LOGD(LOG_TAG,"someting"); 
  I tried already using this command  esp_log_level_set("*", ESP_LOG_ERROR); 
  including the settings "make menuconfig"

chegewara, 2018-04-26:
`E (3678) BLERemoteCharacteristic: esp_ble_gattc_get_all_descr: ESP_GATT_NOT_FOUND`
means that characteristic does not have any descriptors, no need to worry, its not error

`E (19898) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0016`
this probably also its not an error, sometimes espressif devs use LOGE for logging, but in your case it can just be info that disconnection reason is 0x0016, especialy if its after pClient->disconnect(); (im not sure, though)

Here 
- std::string value = pRemoteCharacteristic->readValue();
- BLERemoteService* pRemoteService = pClient->getService(serviceUUID);

try to add check:
```
if(pClient->isConnected()){
BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
}
```

JeroenBeemster, 2018-04-26:
Thanks. I tried the isConnected() already. Actually and does not help

chegewara, 2018-04-26:
Maybe a little delay before read/write operation that causes crash?

chegewara, 2018-04-26:
> Additional question: How to show the messages that are in the library: ESP_LOGD(LOG_TAG,"someting");
> I tried already using this command esp_log_level_set("*", ESP_LOG_ERROR);
> including the settings "make menuconfig"

In arduino you have Debug core level to set, in esp-idf its log level in menuconfig that you need to setup first. I have set it up to verbose (helps in debug).

JeroenBeemster, 2018-04-26:
> Maybe a little delay before read/write operation that causes crash?

I  tried a delay of 100ms, I thought that should be enough. That did not help  :(

chegewara, 2018-04-26:
Sorry, but i cant perform real life simulation with so many devices so i cant any solution or advice. I wish i can be more helpful.

JeroenBeemster, 2018-04-26:
> I Sorry, but i cant perform real life simulation with so many devices so i cant any solution or advice. I wish i can be more helpful.
No worries I will come back with more details. 
I am almost sure that this is just due to the BLE-server is sending data and during transmit the connection is lost. And there is no fallback scenario. I think I found the "Verbose" setting. So I will know more soon

closed 2018-07-12 by JeroenBeemster