nkolban/esp32-snippets #514: REST-Error: Not enough space

Donderda, 2018-05-08:
Hi there!
First of all: This is a great repo! It saves me a lot of time =).

I've got a small problem trying to send some data to a REST-endpoint:

I get the message `RESTClient: post(): Send failure: Not enough space`.
this is the message I want to send: `{"type":"iBeacon","RSSI:"-45,"powerTX":-59,"identifier":"e2c56db5dffb48d2b060d0f5a71096e0-12-34","MAC":"6f:49:a2:9b:65:1c","major": 12,"minor":34}`

This is my sourcecode where I establish the WiFi-Connection, getting the string and trying to post the string:
```
void WiFiClient::run(void *data) {
	QueueHandle_t xQueue = (QueueHandle_t) data;
	char buffer[constants::MAX_ITEM_SIZE];
	std::stringstream ss;
	ss << "http://" << constants::server_ip << ":8081/beacon";
	std::string url = ss.str();
	ESP_LOGI("WiFiClient", "Endpoint: %s", url.data());

	RESTClient client;
	client.setURL(url);
	client.addHeader("Content-Type", "application/json");
	client.addHeader("device", constants::name);
	while (true) {
		if (uxQueueMessagesWaiting(xQueue)) {
			if (xQueueReceive(xQueue, &buffer, 10) == pdPASS) {
				if(wifi.isConnectedToAP()){
					RESTTimings *timings = client.getTimings();

					ESP_LOGI("WiFiClient", "ResponseCode: %li", client.post(buffer));
					ESP_LOGI("WiFiClient", "Result: %s", client.getResponse().c_str());
					timings->refresh();
					ESP_LOGI("WiFiClient", "timings: %s", timings->toString().c_str());

				} else {
					ESP_LOGE("WiFiClient", "Not connected...");
					this->stop();
				}
				ESP_LOGI("WiFiClient", "incoming %s", buffer);
			}
		}
		this->delay(10);
	}
}
```
Do you have an idea how this might be fixed?

chegewara, 2018-05-08:
Hi, this error message is not clear to me, but you can try to do some tweaks in menuconfig wifi options tx buffers.

Donderda, 2018-05-08:
Hi and thank you for your reply. I will have a look at this settings and reply as soon as possible :)

Donderda, 2018-05-08:
Thank you! This really helped me :)

closed 2018-05-08 by Donderda