nkolban/esp32-snippets #516: HttpServer with SSL enable

snahmad, 2018-05-17:
Hi,

Does any one used HttpServer class with enable SSL.  see below code. also check debug logs

```
class HttpTask: public Task {
	void run(void *data) {
		ESP_LOGD("http", "Testing http ...");
		
		SSLUtils::setKey(key);
		SSLUtils::setCertificate(cert);
		
		HttpServer* pHttpServer = new HttpServer();
		pHttpServer->addPathHandler(
		HttpRequest::HTTP_METHOD_POST,
		"/",
		httpPostHandler);
		
		pHttpServer->start(80, true);
		
		return;
	}
};

```

C:/Work/LibDev/esp32/esp-idf/components/mbedtls/library/ssl_tls.c:6788: mbedtl
s_ssl_handshake() returned -30976 (-0x7900)
D (49899) Socket: rc=-30976, MBEDTLS_ERR_SSL_WANT_READ=-26880
D (49905) Socket: << accept: sockFd: 4100
D (49911) HttpServerTask: HttpServer that was listening on port 80 has receive
d a new client connection; sockFd=4100
D (49919) HttpParser: >> parse: socket: fd: 4100
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited)
. Exception was unhandled.

snahmad, 2018-05-17:
I was using port=80 instead 443 for https.

Now handshake successfully. 

size_t Socket::receive(uint8_t* data, size_t length, bool exact)  cause crash on post data receive.

s_ssl_read_record_layer() returned -29312 (-0x7280)
D (29204) Socket: rc=0, MBEDTLS_ERR_SSL_WANT_READ=-26880
D (29210) Socket: << accept: sockFd: 4100
D (29216) HttpServerTask: HttpServer that was listening on port 443 has receiv
ed a new client connection; sockFd=4100
D (29224) HttpParser: >> parse: socket: fd: 4100
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited)

chegewara, 2018-05-17:
Try to play with this values:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/HttpServer.cpp#L55

snahmad, 2018-05-17:
ok will try tomorrow.

snahmad, 2018-05-17:
To let you know this example works with my openssl client and web browser.
https://github.com/espressif/esp-idf/tree/master/examples/protocols/openssl_server

snahmad, 2018-05-18:
You meant change 
From 
HttpServerTask(std::string name): Task(name, 16*1024) {
to sometime higher. 

HttpServerTask(std::string name): Task(name, 20*1024) {

stack overflow is not issue.

snahmad, 2018-05-18:
I tried out changed to 
HttpServerTask(std::string name): Task(name, 20*1024) 

Now SSL shake is successfully. but later on receive Http Parsing failed.
See this thread:
https://www.esp32.com/viewtopic.php?p=24985#p24985

chegewara, 2018-05-18:
Its hard to say what can be the reason, espesialy that i didnt try SSL myself. But if i had to guessing it looks like free heap or stack issue. I would play with menuconfig options and try to free as much as its possible.

snahmad, 2018-05-18:
how can you tell heap or stack issue.
when plenty free
Stack free: 10368
Internal Heap free: 74012
Heap free: 8420:74012

I will try to tweak other settings.

snahmad, 2018-05-19:
I have tried all options. no luck. kindly let me know if any have used HttpServer class with SSL successfully with any self signed SSL certificate.

snahmad, 2018-05-22:
D (72689) Socket: << receive: rc: 1
D (72693) Socket: << accept: sockFd: 8196
D (72699) HttpServerTask: HttpServer that was listening on port 443 has receiv
ed a new client connection; sockFd=8196
D (72707) HttpParser: >> parse: socket: fd: 8196
D (72721) Socket: >> receive: sockFd: 8196, length: 1, exact: 0
D (72727) Socket: before mbedtls_ssl_read
Guru Meditation Error: Core 0 panic'ed (InstrFetchProhibited)
. Exception was unhandled.

mbedtls_ssl_read cause crash.

CODE: SELECT ALL

```
size_t Socket::receive(uint8_t* data, size_t length, bool exact) {
   ESP_LOGD(LOG_TAG, ">> receive: sockFd: %d, length: %d, exact: %d", m_sock, length, exact);
   if (exact == false) {
      int rc;
      if (getSSL()) {
         do {
            ESP_LOGD(LOG_TAG, "before mbedtls_ssl_read");
            rc = mbedtls_ssl_read(&m_sslContext, data, length);
            ESP_LOGD(LOG_TAG, "rc=%d, MBEDTLS_ERR_SSL_WANT_READ=%d", rc, MBEDTLS_ERR_SSL_WANT_READ);
         } while(rc == MBEDTLS_ERR_SSL_WANT_WRITE || rc == MBEDTLS_ERR_SSL_WANT_READ);
      } else {
         rc = ::lwip_recv_r(m_sock, data, length, 0);
         if (rc == -1) {
            ESP_LOGE(LOG_TAG, "receive: %s", strerror(errno));
```

chegewara, 2018-05-22:
Maybe its the issue with m_sslContext? Or length:
https://github.com/espressif/esp-idf/blob/bae9709a7950e2ee08e14c65be27831bcb547105/examples/protocols/https_mbedtls/main/https_mbedtls_example_main.c#L279-L281

snahmad, 2018-05-24:
Fixed some issues. Still some issues accessing from web browser using Http get method call for files.

https://github.com/nkolban/esp32-snippets/issues/527