nkolban/esp32-snippets #518: An error occurred when the notification was sent in the touch interrupt function.

dtboy1995, 2018-05-18:
- code fragment
```c
void gotTouch(){
  if (deviceConnected) {
       pCharacteristic->setValue(&value, 1);
       pCharacteristic->notify();
       value++;
  }
}
void setup(){
  Serial.begin(115200);

  // Create the BLE Device
  BLEDevice::init("MyESP32");
  ...
  ...
  touchAttachInterrupt(T0, gotTouch, 80);
}
```

- error log
```log
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x4008ca3a  PS      : 0x00060e34  A0      : 0x8008bbfe  A1      : 0x3ffc0aa0  
A2      : 0x3ffe4920  A3      : 0x3ffcfda8  A4      : 0x00060e21  A5      : 0x00000001  
A6      : 0x00060e21  A7      : 0x00000000  A8      : 0x3ffcfda8  A9      : 0x3ffcfda8  
A10     : 0x00000019  A11     : 0x00000019  A12     : 0x00000006  A13     : 0x00000001  
A14     : 0x00060e23  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
Core 1 was running in ISR context:
EPC1    : 0x400daa1e  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008ca3a

Backtrace: 0x4008ca3a:0x3ffc0aa0 0x4008bbfb:0x3ffc0ac0 0x4008aa17:0x3ffc0ae0 0x400d6e38:0x3ffc0b20 0x400d376e:0x3ffc0b60 0x400d29f1:0x3ffc0bd0 0x40080fb3:0x3ffc0bf0 0x40081725:0x3ffc0c10 0x40154f73:0x00000000

Core 0 register dump:
PC      : 0x4008b2dd  PS      : 0x00060434  A0      : 0x8008bcad  A1      : 0x3ffd50b0  
A2      : 0x3ffc111c  A3      : 0x3ffd5140  A4      : 0x00000010  A5      : 0x3ffd8140  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x0000cdcd  A9      : 0x0000cdcd  
A10     : 0xb33fffff  A11     : 0x0000abab  A12     : 0x3ffe5b94  A13     : 0x00000001  
A14     : 0x00060423  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x4008b2dd:0x3ffd50b0 0x4008bcaa:0x3ffd50d0 0x4008a651:0x3ffd50f0 0x400fa41d:0x3ffd5130 0x4012a145:0x3ffd5160 0x40127e2b:0x3ffd5180 0x400e2c2e:0x3ffd51a0 0x400de605:0x3ffd5450

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13256
entry 0x40078a90
```

chegewara, 2018-05-18:
Could you use this tool to debug stacktrace:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md#decoding-an-exception-stack-trace

chegewara, 2018-05-18:
This looks strange. Line 410 is debug line:
https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/src/BLEAdvertisedDevice.cpp#L410

Could you try to change log level in arduino to WARN or ERROR? Also test your code how it work without touch procedure, sending some notifications and with touch but without sending nootifications.

dtboy1995, 2018-05-18:
in `loop` function, it works well

i change log level 
```log
[D][BLECharacteristic.cpp:664] setValue(): >> setValue: length=1, data=00, characteristic UUID=beb5483e-36e1-4688-b7f5-ea07361b26a8
[D][BLECharacteristic.cpp:671] setValue(): << setValue
[D][BLECharacteristic.cpp:524] notify(): >> notify: length: 1
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x4008ca3a  PS      : 0x00060e34  A0      : 0x8008bbfe  A1      : 0x3ffc0aa0  
A2      : 0x3ffe496c  A3      : 0x3ffcfde8  A4      : 0x00060e21  A5      : 0x00000001  
A6      : 0x00060e21  A7      : 0x00000000  A8      : 0x3ffcfde8  A9      : 0x3ffcfde8  
A10     : 0x00000019  A11     : 0x00000019  A12     : 0x00000006  A13     : 0x00000001  
A14     : 0x00060e23  A15     : 0x00000079  SAR     : 0x00000008  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  
Core 1 was running in ISR context:
EPC1    : 0x400de4c6  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008ca3a

Backtrace: 0x4008ca3a:0x3ffc0aa0 0x4008bbfb:0x3ffc0ac0 0x4008aa17:0x3ffc0ae0 0x400da650:0x3ffc0b20 0x400d42f9:0x3ffc0b60 0x400d3091:0x3ffc0bd0 0x40080fb3:0x3ffc0bf0 0x40081725:0x3ffc0c10 0x40158daf:0x00000000

Core 0 register dump:
PC      : 0x4008b2da  PS      : 0x00060a34  A0      : 0x8008bcad  A1      : 0x3ffd50f0  
A2      : 0x3ffc111c  A3      : 0x3ffd5180  A4      : 0x00000010  A5      : 0x3ffd8240  
A6      : 0x00000015  A7      : 0x82b87d47  A8      : 0x0000cdcd  A9      : 0x0000cdcd  
A10     : 0xb33fffff  A11     : 0x0000abab  A12     : 0x3ffe5bd4  A13     : 0x00000001  
A14     : 0x00060a23  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x4008b2da:0x3ffd50f0 0x4008bcaa:0x3ffd5110 0x4008a651:0x3ffd5130 0x400fe049:0x3ffd5170 0x4012dd25:0x3ffd51a0 0x4012ba0b:0x3ffd51c0 0x400e66d6:0x3ffd51e0 0x400e20ad:0x3ffd5490

Rebooting...

```
- decocde 
```log
PC: 0x4008ca3a: vListInsert at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./list.c line 188
EXCVADDR: 0x00000000

Decoding stack results
0x4008ca3a: vListInsert at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./list.c line 188
0x4008bbfb: vTaskPlaceOnEventList at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 2922
0x4008aa17: xQueueGenericReceive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1589
0x400da650: FreeRTOS::Semaphore::wait(std::__cxx11::basic_string  , std::allocator  >) at C:\Users\dtboy\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src\FreeRTOS.cpp line 85
0x400d42f9: BLECharacteristic::notify() at C:\Users\dtboy\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src\BLECharacteristic.cpp line 564
0x400d3091: gotTouch() at C:\Users\dtboy\Documents\Arduino\generated_examples\BLE_notify/BLE_notify.ino line 104
0x40080fb3: __touchISR at C:\Users\dtboy\Documents\Arduino\hardware\espressif\esp32\cores\esp32\esp32-hal-touch.c line 45
0x40158daf: esp_vApplicationIdleHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./freertos_hooks.c line 62
```
please forgive me for not understanding.

Thank you very much.

chegewara, 2018-05-18:
No problem, all is good.

I think i know where is the problem, but you will have to make some workaround to make it working. Probably the issue is with `pCharacteristic->notify();`, this is blocking function (with semaphores) and probably `touchAttachInterrupt` has problem with it. Try something like that:

```
void gotTouch(){
   // if touched ...
   if (deviceConnected) {
       pCharacteristic->setValue(&value, 1);
       pCharacteristic->notify();
       value++;
  }
  isTouch = false;
}

bool isTouch = false;
void gotTouchSet(){
  isTouch = true;
}
touchAttachInterrupt(T0, gotTouchSet, 40);

void loop(){
if(isTouch){
  gotTouch();
}
...
}
```

dtboy1995, 2018-05-18:
I think you're right. It's a very good idea.  :+1:
Thank you very much.