nkolban/esp32-snippets #520: Http/SSL get file

snahmad, 2018-05-18:
Hi,

Have someone written C++ class to  do https get file from secured Http server piece by piece.
either using libCurl or socket. 
Current RESTClient does not support get file.

Thanks,
Naeem

HanVertegaal, 2018-05-18:
Yes, see attached files

[ota.zip](https://github.com/nkolban/esp32-snippets/files/2016888/ota.zip)

"It-works-on-my-system" guarantee.

snahmad, 2018-05-18:
ok, Thanks. I will try it later.

snahmad, 2018-05-21:
I am getting this error. any idea why?
D (14416) Download: >> Get
C:/Work/LibDev/esp32/esp-idf/components/freertos/tasks.c:1072 (prvAddNewTaskTo
ReadyList)- assert failed!
abort() was called at PC 0x4008bf0e on core 0

HanVertegaal, 2018-05-21:
Apparently you cannot create a task pinned to core 1. Perhaps you only have a single core?
Try to replace the final parameter in [xCreateTaskPinnedToCore](url) to `tskNO_AFFINITY`.

snahmad, 2018-05-21:
This my logs.  Content-Length is missing. This is encrypted traffic. for some reason, I cannot see any ssl traffic in wireshark. Can you guess form logs what is going wrong?

I (19726) openssl: ssl_tls.c:3754 <= read record

I (19730) openssl: ssl_tls.c:7042 <= read

D (19736) Download: http_response length(13)
D (19740) HttpParser: >> ParseStatusLine: "HTTP/1.1 200 " [13]
D (19746) HttpParser: << ParseStatusLine: method: , version: HTTP/1.1, status:
 200
E (19753) Download: Missing 'Content-Length'
V (19759) sdmmc_cmd: sending cmd slot=1 op=24 arg=12000 flags=1c10 data=0x3ffb
ed94 blklen=512 datalen=512 timeout=5000
V (19769) sdspi_host: sdspi_host_start_command: slot=1, CMD24, arg=0x00012000
flags=0x7, data=0x3ffbed94, data_size=512 crc=0x6a
V (19785) sdspi_transaction: r1 = 0x00 hw_cmd.r[0]=0xffffffff
V (19787) sdmmc_cmd: cmd response 00000000 00000000 00000000 00000000 err=0x0
state=0
I (19795) openssl: ssl_tls.c:7186 => write close notify

I (19801) openssl: ssl_tls.c:4124 => send alert message

I (19807) openssl: ssl_tls.c:4125 send alert level=1 message=0

I (19813) openssl: ssl_tls.c:2701 => write record

I (19819) openssl: ssl_tls.c:1258 => encrypt buf

I (19825) openssl: ssl_tls.c:1269 dumping 'before encrypt: output payload' (2
bytes)

I (19833) openssl: ssl_tls.c:1269 0000:  01 00
            ..

I (19843) openssl: ssl_tls.c:1314 dumping 'computed mac' (20 bytes)

I (19849) openssl: ssl_tls.c:1314 0000:  29 14 1c 2f 52 00 a4 d8 37 a3 68 a7 e
1 1d 0d 16  )../R...7.h.....

I (19859) openssl: ssl_tls.c:1314 0010:  5c 8c f9 12
            \...

I (19871) openssl: ssl_tls.c:1481 before encrypt: msglen = 48, including 16 by
tes of IV and 10 bytes of padding

I (19881) openssl: ssl_tls.c:1560 <= encrypt buf

I (19885) openssl: ssl_tls.c:2838 output record: msgtype = 21, version = [3:2]
, msglen = 48

I (19895) openssl: ssl_tls.c:2841 dumping 'output record sent to network' (53
bytes)

I (19903) openssl: ssl_tls.c:2841 0000:  15 03 02 00 30 b8 a6 c8 d8 85 d1 ce 6
7 4e d7 a2  ....0.......gN..

I (19915) openssl: ssl_tls.c:2841 0010:  ac 2f 5e 96 68 90 35 7a f2 1b 04 90 9
7 60 fd a9  ./^.h.5z.....`..

I (19925) openssl: ssl_tls.c:2841 0020:  bb 55 92 a3 ad ef 16 8e a6 9a 87 65 6
d 2b 01 82  .U.........em+..

I (19935) openssl: ssl_tls.c:2841 0030:  e7 80 bd 61 40
            ...a@

I (19944) openssl: ssl_tls.c:2416 => flush output

I (19950) openssl: ssl_tls.c:2435 message length: 53, out_left: 53

I (19961) openssl: ssl_tls.c:2441 ssl->f_send() returned 53 (-0xffffffcb)

I (19966) openssl: ssl_tls.c:2460 <= flush output

I (19970) openssl: ssl_tls.c:2850 <= write record

I (19976) openssl: ssl_tls.c:4137 <= send alert message

I (19982) openssl: ssl_tls.c:7202 <= write close notify

snahmad, 2018-05-21:
It could be Content-Length is case sensitive matching. Let me check

snahmad, 2018-05-21:
Content-Length us not issue. 
D (19728) Download: http_response length(13), however I am sending 112 bytes file. Something going wrong. I am using  the attached certificate and private key.
Kindly check it.

[server_certificate.zip](https://github.com/nkolban/esp32-snippets/files/2023181/server_certificate.zip)

HanVertegaal, 2018-05-21:
If you got to the message 'Missing Content-Length', then you're almost there. At least the SSL handshake is working.
Have you tried downloading the file using a browser? Do you then see the Content-Length using the browser's Developer Tools?

snahmad, 2018-05-21:
I have added these additional logs. Download using browser works and using postman by disabling SSL self signed verification. I am getting only http first header line  in my recv_buf below. any idea?
// parse message
std::string response = std::string(recv_buf, ret);
ESP_LOGD(TAG, "http_response length(%u)", response.size());
ESP_LOGD(TAG, "http_response str(%s)", response.c_str());
		

D (19523) Download: http_response length(13)
D (19529) Download: http_response str(HTTP/1.1 200 )
D (19533) HttpParser: >> ParseStatusLine: "HTTP/1.1 200 " [13]
D (19539) HttpParser: << ParseStatusLine: method: , version: HTTP/1.1, status:
 200
E (19547) Download: Missing 'Content-Length'
V (19553) sdmmc_cmd: sending cmd slot=1 op=24 arg=12000 flags=1c10 data=0x3ffb
ed94 blklen=512 datalen=512 timeout=5000
V (19563) sdspi_host: sdspi_host_start_command: slot=1, CMD24, arg=0x00012000
flags=0x7, data=0x3ffbed94, data_size=512 crc=0x6a
V (19577) sdspi_transaction: r1 = 0x00 hw_cmd.r[0]=0xffffffff
V (19579) sdmmc_cmd: cmd response 00000000 00000000 00000000 00000000 err=0x0

snahmad, 2018-05-21:
Is my parameters are correct?

	download_params.sourceURL = "/test.html";
	download_params.sourceHost = "192.168.2.100";
	download_params.targetPath = "/sdcard/test.html";
	download_params.sema = xSemaphoreCreateMutex();
	download_params.progress = 0.0;
	download_params.success = false;

chegewara, 2018-05-21:
For some reason this part of code is not working:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/HttpParser.cpp#L287-L292

Maybe its problem with lineTerminator? Try to print log message after this line:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/HttpParser.cpp#L287

HanVertegaal, 2018-05-22:
@snahmad: You may need to use a more fully specified sourceURL such as 'https://server.com/folder/file.ext'

snahmad, 2018-05-22:
I tried out fully specified sourceURL . no use.
This example works
https://github.com/espressif/esp-idf/tree/master/examples/protocols/openssl_client

I think some different between my menuconfig and this example config.
I have attached zip contains for both project config files.
[sdkconfig.zip](https://github.com/nkolban/esp32-snippets/files/2026225/sdkconfig.zip)

My project config is new_sdkconfig

snahmad, 2018-05-22:
Fix the issue. Update Download::WriteFile.  We need to SSL_read in a loop till we read all contents including http header and body. All working now.

https://github.com/espressif/esp-idf/tree/master/examples/protocols/openssl_client example
also use loop to read all http response.

```
bool Download::WriteFile(SSL* ssl, std::ofstream& ofs, float* progress)
{
    ESP_LOGD(TAG, ">> WriteFile");

    // read first packet containing status line and headers
    char recv_buf[RECV_BUF_LEN];
	int recv_bytes = 0;
	int ret = 0;
    std::string response;
	do {
        ret = SSL_read(ssl, recv_buf, RECV_BUF_LEN - 1);
        if (ret <= 0) {
			break;
        }
        recv_buf[ret] = '\0';
        recv_bytes += ret;
        //ESP_LOGI(TAG, "%s", recv_buf);
		response += recv_buf;
    } while (1);
    
	ESP_LOGI(TAG, "totaly read %d bytes data", recv_bytes);
	
    
	ESP_LOGD(TAG, "http_response length(%u)", response.size());
	ESP_LOGD(TAG, "http_response str(%s)", response.c_str());
	
	
    HttpParser* parser = new HttpParser();
    parser->parseResponse(response);

    int httpVersionMajor = 0;
    int httpVersionMinor = 0;
    if (2 != sscanf(parser->getVersion().c_str(), "HTTP/%d.%d", &httpVersionMajor, &httpVersionMinor))
    {
        recv_buf[ret] = 0;
        ESP_LOGE(TAG, "Invalid HTTP status line '%s'", recv_buf);
        delete parser;
        return false;
    }
    if (httpVersionMajor != 1) {
        recv_buf[ret] = 0;
        ESP_LOGE(TAG, "HTTP version not supported '%s'", recv_buf);
        delete parser;
        return false;
    }

    if (atoi(parser->getStatus().c_str()) != HttpResponse::HTTP_STATUS_OK)
    {
        recv_buf[ret] = 0;
        ESP_LOGE(TAG, "Non-200 HTTP status code received '%s'", recv_buf);
        delete parser;
        return false;
    }

    if (!parser->hasHeader(HttpRequest::HTTP_HEADER_CONTENT_LENGTH))
    {
        ESP_LOGE(TAG, "Missing 'Content-Length'");
        delete parser;
        return false;
    }
    int contentLength = atoi(parser->getHeader(HttpRequest::HTTP_HEADER_CONTENT_LENGTH).c_str());
    ESP_LOGD(TAG, "Content-Length = %d", contentLength);

    std::string body = parser->getBody();
	
	ESP_LOGD(TAG, "Body length(%u)", body.size());
	ESP_LOGD(TAG, "Body='%s'", body.c_str());
	
    ofs.write(body.data(), body.length());
    int bytesLeft = contentLength - body.length();
    delete parser;

    while ((bytesLeft > 0) && (ret > 0))
    {
        ret = SSL_read(ssl, recv_buf, std::min(bytesLeft, RECV_BUF_LEN - 1));
        if (ret <= 0)
        {
            ESP_LOGE(TAG, "Read error");
            return false;
        }
        else
        {
            ofs.write(recv_buf, ret);
            bytesLeft -= ret;
            *progress = 1.0 - (float)bytesLeft / contentLength;
        }
    }

    ESP_LOGD(TAG, "<< WriteFile");
    return true;
}
```

snahmad, 2018-05-22:
I did not see last loop. My above will not work for large file. It can cause out of heap memory while accumulating big file. Need to come with better solution.

HanVertegaal, 2018-05-22:
The first 1KB read should load all the headers, and perhaps a couple of bytes from the body. The file write loop copies the first part of the body to the file and then reads the remainder of the body.
Apparently 1 KB is not large enough to contain the headers in your case. Perhaps you can increase the size of the first read, or read until you see the header terminator `"\r\n\r\n"`.

snahmad, 2018-05-22:
It is reading line by line http response header.

I (24321) Download: SSL_read:'HTTP/1.0 200 '
I (24440) Download: SSL_read:'OK'
I (24444) Download: SSL_read:'
'
I (24452) Download: SSL_read:'Content-Type: text/html
'
I (24456) Download: SSL_read:'Content-Length: 143
'
I (24460) Download: SSL_read:'
'
I (24464) Download: SSL_read:'<!DOCTYPE html>
<html>
<body>

<h1>My First Heading</h1>

<p>My first paragraph.</p>

<p>My second paragraph.</p>

</body>
</html>
'

```
    char recv_buf[RECV_BUF_LEN];
	int recv_bytes = 0;
	int ret = 0;
    std::string response;
	do {
        ret = SSL_read(ssl, recv_buf, RECV_BUF_LEN - 1);
        if (ret <= 0) {
			break;
        }
        recv_buf[ret] = '\0';
        recv_bytes += ret;
        ESP_LOGI(TAG, "SSL_read:'%s'", recv_buf);
		response += recv_buf;
    } while (1);

```

chegewara, 2018-05-22:
Most likely its not the case, but most your logs has:
`ParseStatusLine: "HTTP/1.1 200`
except last one, which is:
`Download: SSL_read:'HTTP/1.0 200 '`
Two different http versions or last one is from different server?

HanVertegaal, 2018-05-22:
The request is sent as HTTP version 1.1, but the server may well respond with version 1.0. It shouldn't really matter.

snahmad, 2018-05-22:
I have changed version from 1.1 to 1.0 in Http get request. I do not have any other server to test against. 
I need to handle these responses and first get HTTP reader and check http reader is ok. then read body chunk by chunk of 1k and write to file.

snahmad, 2018-05-22:
My solution. read few times first to get all http reader info.

```
bool Download::WriteFile(SSL* ssl, std::ofstream& ofs, float* progress)
{
    ESP_LOGD(TAG, ">> WriteFile");

    // read first packet containing status line and headers
    char recv_buf[RECV_BUF_LEN];
	int recv_bytes = 0;
	int ret = 0;
    std::string response;
	for(int i=1 ; i<=5 ; i++)
	{
        ret = SSL_read(ssl, recv_buf, RECV_BUF_LEN - 1);
        if (ret <= 0) {
			ESP_LOGE(TAG, "No data");
			return false;
        }
        recv_buf[ret] = '\0';
        recv_bytes += ret;
        ESP_LOGI(TAG, "SSL_read:'%s'", recv_buf);
		response += recv_buf;
    }
    
	ESP_LOGI(TAG, "totaly read %d bytes data", recv_bytes);
	
    
	ESP_LOGD(TAG, "http_response length(%u)", response.size());
	ESP_LOGD(TAG, "http_response str(%s)", response.c_str());
	
	
    HttpParser parser;
    parser.parseResponse(response);

    int httpVersionMajor = 0;
    int httpVersionMinor = 0;
    if (2 != sscanf(parser.getVersion().c_str(), "HTTP/%d.%d", &httpVersionMajor, &httpVersionMinor))
    {
        recv_buf[ret] = 0;
        ESP_LOGE(TAG, "Invalid HTTP status line '%s'", recv_buf);
        return false;
    }
    if (httpVersionMajor != 1) {
        recv_buf[ret] = 0;
        ESP_LOGE(TAG, "HTTP version not supported '%s'", recv_buf);
        return false;
    }

    if (atoi(parser.getStatus().c_str()) != HttpResponse::HTTP_STATUS_OK)
    {
        recv_buf[ret] = 0;
        ESP_LOGE(TAG, "Non-200 HTTP status code received '%s'", recv_buf);
        return false;
    }

    if (!parser.hasHeader(HttpRequest::HTTP_HEADER_CONTENT_LENGTH))
    {
        ESP_LOGE(TAG, "Missing 'Content-Length'");
        return false;
    }
    int contentLength = atoi(parser.getHeader(HttpRequest::HTTP_HEADER_CONTENT_LENGTH).c_str());
    ESP_LOGD(TAG, "Content-Length = %d", contentLength);

    std::string body = parser.getBody();
	
	ESP_LOGD(TAG, "Body length(%u)", body.size());
	ESP_LOGD(TAG, "Body='%s'", body.c_str());
	
    ofs.write(body.data(), body.length());
    int bytesLeft = contentLength - body.length();

    while ((bytesLeft > 0) && (ret > 0))
    {
        ret = SSL_read(ssl, recv_buf, std::min(bytesLeft, RECV_BUF_LEN - 1));
        if (ret <= 0)
        {
            ESP_LOGE(TAG, "Read error");
            return false;
        }
        else
        {
            ESP_LOGI(TAG, "SSL_read:'%d'", ret);
			ofs.write(recv_buf, ret);
            bytesLeft -= ret;
            *progress = 1.0 - (float)bytesLeft / contentLength;
        }
    }

    ESP_LOGD(TAG, "<< WriteFile");
    return true;
}

```

snahmad, 2018-05-22:
Hi HanVertegaal ,
Are you downloading only file? Is your download C++ class is part of project? where you download given list of files and write to file system.

Is this part of OTA which include firmware update as well? Are you doing OTA firmware update using SSL/HTTPS as well?

Would like to share your experience. 

I need to implement OTA firmware than all my web files and some image files as well download from https server and write disk.

As separate task.
I still need to check  C++HttpServer  which was crashing when used on port 43 with SSL is enabled for any Http post request. HttpParser was failing. It could be the same issue above not passing all Http headers info.

Thanks,
Naeem

HanVertegaal, 2018-05-22:
Hello Naeem,

The Download module is indeed part of an OTA solution for my project. My ESP has 2 OTA partitions, and a single SPIFFS partition containing a webapp. Both firmware and webapp are stored on a website together with a versioninfo file. This file lists the current version numbers, the location of the firmware and webapp images and a link to the release notes release notes. The Ota module can check whether an update is available and if true, download the new images and write them to flash.

Note that the Ota module uses other modules in my project, but I think you can remove those without too much trouble.

[ota.zip](https://github.com/nkolban/esp32-snippets/files/2027759/ota.zip)

Han

snahmad, 2018-05-23:
Hi Han,

Thanks for sharing. 
Do you have single web file or many web files such as .html. js, .css etc.

Are you using two OTA partitions?

HanVertegaal, 2018-05-23:
My 1MB SPIFFS partition contains a full Angular 6 webapp.
There are two 1.5MB OTA partitions.

snahmad, 2018-05-23:
ok, I am also going to use Angular latest version. Does his webapp contains bootsrap css as well.
Is this zip file contains all web files?

I was thinking of using sd card for large web files contains some image files  as well.
How you can fit into 1Mb?

HanVertegaal, 2018-05-23:
I use [spectre CSS](https://picturepan2.github.io/spectre/index.html), but bootstrap will fit too. Currently the webapp uses some 300KB when I build it using ng-cli (ng build --prod). The bundles created by ng-cli are zipped and the .gz extension is added to them. They can then be sent to the client using the header 'Content-Encoding: gzip'. However, I do not have any large images, and those can quickly add to the size of the webapp.

HanVertegaal, 2018-05-23:
I do not unzip. The Ota images and the webapp are retrieved uncompressed.

snahmad, 2018-05-24:
ok. angular cli generate distribution folder which contains not one file but many files. How your one files contains many files?

HanVertegaal, 2018-05-24:
I use mkspiffs to create the spiffs filesystem which can then be flashed to the spiffs partition.
My the spiffs target in my makefile looks somewhat like this:

`.PHONY: spiffs
spiffs:
ifneq ($(wildcard $(WEB_PATH)/3rdpartylicenses.txt),)
	# make version file
	$(eval VERSION=$(shell date +%y%m%d-%H%M%S))
	echo $(VERSION) > $(VERSION_PATH)/webversion.txt
	cp $(VERSION_PATH)/webversion.txt $(WEB_PATH)/webversion.txt
	# get rid of pesky txt file
	rm -f $(WEB_PATH)/3rdpartylicenses.txt
	# shorten the filenames
	mv $(WEB_PATH)/main* $(WEB_PATH)/main.$(VERSION).js; fi
	mv $(WEB_PATH)/polyfills* $(WEB_PATH)/polyfills.$(VERSION).js; fi
	mv $(WEB_PATH)/runtime* $(WEB_PATH)/runtime.$(VERSION).js; fi
	mv $(WEB_PATH)/styles* $(WEB_PATH)/styles.$(VERSION).css; fi
	# compress files
	gzip -f -9 $(WEB_PATH)/*.js
	gzip -f -9 $(WEB_PATH)/*.css
	# fix filenames in index.html
	sed 's/main\.[0-9a-zA-Z]*\.js/main\.$(VERSION)\.js/g' < $(WEB_PATH)/index.html > $(WEB_PATH)/index2.html
	sed 's/polyfills\.[0-9a-zA-Z]*\.js/polyfills\.$(VERSION)\.js/g' < $(WEB_PATH)/index2.html > $(WEB_PATH)/index3.html
	sed 's/runtime\.[0-9a-zA-Z]*\.js/runtime\.$(VERSION)\.js/g' < $(WEB_PATH)/index3.html > $(WEB_PATH)/index4.html
	sed 's/styles\.[0-9a-zA-Z]*\.css/styles\.$(VERSION)\.css/g' < $(WEB_PATH)/index4.html > $(WEB_PATH)/index5.html
	rm $(WEB_PATH)/index.html $(WEB_PATH)/index2.html $(WEB_PATH)/index3.html $(WEB_PATH)/index4.html
	mv $(WEB_PATH)/index5.html $(WEB_PATH)/index.html
	# build and flash spiffs file
	~/github/mkspiffs/mkspiffs -c $(WEB_PATH) -b 4096 -p 256 -s 0xF0000 $(BUILD_PATH)/spiffs.bin
endif`

snahmad, 2018-05-24:
This is good for first time make flash via serial link.

How do you update Over the air web files.

snahmad, 2018-05-24:
Can you share your contact details? may need more discussion with you offline.

snahmad, 2018-06-05:
Hi,

Just to clarify.
spiffs.bin contains your firmware binary image as well as all web files (css. html).
How do you extract while running within firmware using c/c++. I thought you download each file one by one using HTTP get. I don't need to use SSL for now.

Thanks,
Naeem

snahmad, 2018-06-05:
Is your web applications contains WIFI config page. scan all wifi etc? or any other page similar to ESP32 Explorer.

HanVertegaal, 2018-06-05:
Hi,

My spiffs.bin just contains the web app (.html, .css and .js files - or webpack bundles actually). The firmware is in another .bin file. The spiffs.bin is actually a complete SPIFFS file system made using 'makefs', and it is written to the spiffs partition in flash. From there it is accessed as any spiffs file system using the normal file I/O functions.

My partitions are as follows:
# Name,Type,SubType,Offset,Size,Flags
nvs,data,nvs,0x9000,0x5000,
otadata,data,ota,0xE000,0x2000
ota_0,app,ota_0,0x10000,1536K,
ota_1,app,ota_1,,1536K,
storage,data,spiffs,,0xF0000,

Reading from the file system (flash or SD card) in answer to GET requests from clients can be rather slow. I use a separate thread to read the requested file contents into a buffer while the HttpServer thread sends the data out to the client. This helps a little, especially with the larger files.

My web app does contain a page to scan WiFi access points, and to set the login credentials. Also, to switch between access point and station modes. It is all rather rudimentary at this point and needs some 'beautifying' for actual users. 

Han

HanVertegaal, 2018-06-06:
I cannot share all my sources. But scanning is fairly straightforward. The included code will give you an impression of handling the scan in C++, writing the results in JSON and displaying them in an Angular component.
[scan.zip](https://github.com/nkolban/esp32-snippets/files/2075653/scan.zip)

snahmad, 2018-08-30:
Hello Hen,

Does your OTA works with encrypted bin files?

Naeem

HanVertegaal, 2018-08-30:
There is no provision for encrypted files. It just uses an encrypted connection to the server. However, I see no reason why you couldn't add an option to use encrypted files.

snahmad, 2019-01-04:
Hi,

I need to download file via JavaScript/angular cli http get file from esp32 device into selected folder via web browser. any idea how? example code.

zheyuejiang, 2021-04-24:
where is the "download.h"?