nkolban/esp32-snippets #527: HttpServer with SSL is not handling error gracefully.

snahmad, 2018-05-23:


snahmad, 2018-05-23:
Hi Kolban,
When you think you can look at this issue of using HttpServer with SSL enabled for handling http get or post request from application or browser. 

Can you test at your end with simple application. is website works with web browser? does it handle wrong SSL handshake and connection close.

We like to use this class for production in 1-2 months time period.

Thanks,
Naeem

snahmad, 2018-05-24:
I have updated Http classes to use std::shared_ptr<Socket>. It is working for Https(SSL) post method call.
Each post method need to be  closed before next Http post request. It is better. After few calls. we run out of ram. I think close socket does not de-allocated SSL session ram allocation.

[Http.zip](https://github.com/nkolban/esp32-snippets/files/2035625/Http.zip)

chegewara, 2018-05-25:
https://github.com/nkolban/esp32-snippets/issues/286#issuecomment-392150678
I am moving here, because SSL issue is here.
Now it is important how it is used in user code, and see if it is handled in proper way. Without code and/or full logs its hard to say.

snahmad, 2018-05-25:
ok, i will put usage code soon.

squonk11, 2018-07-06:
Now I have taken a look into the sources of HttpServer.cpp and Socket.cpp. My impression is that SSL code is not really finished in these sources; it looks as if someone stopped implementing it here: [https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/Socket.cpp#L80-L83](https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/Socket.cpp#L80-L83). If I take a look into the exaple provided by Expressif here: [https://github.com/espressif/esp-idf/blob/master/examples/protocols/https_mbedtls/main/https_mbedtls_example_main.c](https://github.com/espressif/esp-idf/blob/master/examples/protocols/https_mbedtls/main/https_mbedtls_example_main.c) I see that they use mbedtls_net_connect() for the client and probably mbedtls_net_accept() (example not provided) for the server. These calls I do not find in the Socket.cpp sourcecode. 
So, I think there is some more work to be done...

snahmad, 2018-07-11:
ok, I am now using mongoose now.

chegewara, 2018-07-11:
To be honest i am using esp_http_client now. Its very easy to use and works great if you dont need websockets. Maybe there is websocket implemented but i dont need it as of yet.

squonk11, 2018-07-11:
But esp_http_client is not a webserver...

chegewara, 2018-07-11:
My bad, sorry.

squonk11, 2018-07-11:
I am currently trying to finalize the SSL integration into Socket.cpp. But it needs several modifications. I am asking myself if it could make more sense to implement a SSLSocket class instead of integrating it into the Socket class. What do you think?

chegewara, 2018-07-11:
At this point you can fork this repo and create branch for SSLSocket and Socket with SSL, then you can play with it and see what will suits you.

chegewara, 2018-07-12:
https://github.com/chmorgan/esphttpd-freertos/blob/master/main/user_main.c

squonk11, 2018-07-12:
Thank you for this  link - it is also an interesting project. But I think it does not use SSL. 
The HttpServer class in this repository basically works for me but I am currently missing the SSL connection. I think/hope that I will be able to make the Socket class work with SSL quite soon. Then I can provide this as pull request. But since there are many modifications to the existing code it would be helpful if someone else but me could check what I have done.

squonk11, 2018-07-16:
For whatever reason I was not able to make the SSL work using mbedtls. But then I tried OpenSSL. With this I got it finally working. The API of OpenSSL is much easier to handle than mbedtls. 
Now the question is if and how I should provide a pull request?

nkolban, 2018-07-16:
Howdy @squonk11 ... life has been super hectic and this repo hasn't been getting anywhere near the love I would like to give it.  However that doesn't mean its anywhere near dead ... one need only call my name 3 times in the mirror and I'll be there.

I like to think I'm very available.  Email at kolban1@kolban.com or a direct message on IRC in channel ESP32 to user kolban.

As for the SSL changes ... my personal preference would be to leverage the mbedTLS library directly without going through the OpenSSL API wrappers.  To the best of my understanding, OpenSSL is as much an API spec as it is an implementation and on ESP-IDF, this maps to the mbedTLS which (again my understanding) is the stack chosen in ESP-IDF for TLS/SSL.

Perhaps we could collaborate together to achieve the goals we are looking for using the mbedTLS APIs?  When something doesn't work the way it should, then we can put our minds together and see what the puzzle might be?

That all said, if the community of users of this repo would prefer to see the OpenSSL APIs leveraged internally then the voices will be heard and that's where we will take it.  If there is no prevailing and valid argument on why one API is "better" than another ... then I'd say stick with mbedTLS.  If there is an argument for why OpenSSL should be used in preference to mbedTLS API  **or** there is a community preference to move to OpenSSL ... then OpenSSL here we go!!!

squonk11, 2018-07-16:
Hi Neil, thanks for your reply. 
I have the code changes for OpenSSL almost ready and I can send it to you today for having a check. The changes are only to Socket.cpp and SSLUtils.cpp. The OpenSSL was much easier to handle and seems to fit better to lwip. 
I have also a version with mbedtls but this does not work: the ssl handshake works, the browser sends a request which is correctly read and interpreted by the HttpServer class but the response seems to get lost somewhere in the mbedtls stack. In wireshark I see no response and the mbedtls_ssl_write reports error -78. No idea how to solve that.

chegewara, 2018-07-16:
I dont know what are OpenSSL requirements but mbedTLS is memory hungry. mbedTLS requires 75kB or more (dont know exact number) for handshake, about half is for library and 30kB its for rx/tx buffers (15kB each).

squonk11, 2018-07-16:
How can I check the memory footprint of OpenSSL? Then we could compare.

chegewara, 2018-07-16:
Use this line for debugging (before, inside and after OpenSSL):
`ESP_LOGI(TAG, "%d: free heap %d", __LINE__, esp_get_free_heap_size());`

squonk11, 2018-07-16:
what do you mean by "before", "inside" and "after" OpenSSL? 
"before" integrating the OpenSSL sources or before invoking the initialization?

chegewara, 2018-07-16:
There can be many "before". In your example code, before you first time invoke/initialize variable with OpenSSL or Socket class. For sure after integrating your code.

squonk11, 2018-07-16:
o.k. I will check

squonk11, 2018-07-16:
Free heap before OpenSSL init: 142284
Free heap after OpenSSL init: 138368
That makes a difference of 3916 bytes.
During runtime I have only 96316 byte free. That is a difference to the beginning of 45968 bytes
After the page has been loaded the free heap is: 129008 bytes. That is a difference to the beginning of 13276 bytes.

squonk11, 2018-07-16:
I was trying to upload my sourcecode using the attach funtion below this input window. But I always receive an error message. Is there a spechial trick to be used?

chegewara, 2018-07-16:
Some extensions only are allowed, if the one you are trying to attack is not allowed add txt extension or upload code to gist.

squonk11, 2018-07-16:
<script src="https://gist.github.com/squonk11/ac65740dadb6f5fc03a9a2a17e522103.js"></script>

squonk11, 2018-07-16:
is it o.k. to upload to gist in this way?

squonk11, 2018-07-17:
now I spent again some more hours on getting mbedtls to work; but: no success!
For the time being I think I will stick to OpenSSL - this seems to works.

squonk11, 2018-07-17:
[https://cpp.libhunt.com/compare-mbedtls-vs-openssl](https://cpp.libhunt.com/compare-mbedtls-vs-openssl)