nkolban/esp32-snippets #530: Problem to add several caracterictics to the same service

dsolheid, 2018-05-24:
Hi all,

I have tested different samples with my ESP-WROOM-32 Lolin ..
But 
1) I always see the same UUID for the service / caracteristic, like : 
SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

Even if I perform a new UPLOAD of another sample app ... (yes, I know, it's seems crazy ..)
I'mean, even if i upload the UART Service app, the server name change to "UART Service", but I still see the UUIDs named above.

2) When I try to adapte sample with more than one caracteristic .. I still see only the first caracteristic above ..

Completly crazy ... the hardware/memory  is corrupted ?? I dont know ..

Did you experience something similar ?

Best Regards 

Didier

ps: I tested with a Android Studio App, a Cypress BLE app, another cutom app

chegewara, 2018-05-24:
Hi,
for power save android is caching bluetooth device, so if you connect to device with the same MAC old data are displayed.
Try to run android app twice or use nRF connect to test your esp32 application and connect, disconnect an connect again.

vuanh-minh, 2018-05-28:
Hi,

I had exactly the same problem with my ESP32 DevKitC, when i changed my characteristics, and upload it to the ESP, only the name of the device changed but the characteristic were still the same. So i tried to upload it to another ESP32, and there was no problems, and the characteristics changed. So i think the problem is from the first ESP, i don't know if the hardware/memory is corrupted or something... If someone have an answer.