nkolban/esp32-snippets #533: BLE connection between ESPs32

luizahiginoss, 2018-05-24:
I'm trying to connect one ESP as a client and another as a server, to do that I made the service UUID and the characteristic UUID the same. But using your example codes (latest version), client and server do not connect and the client code crashes after line 47 

Serial.println(" - Found our service");

How can I make this connection happens and read the characteristic value? I'm using arduino IDE plataform

chegewara, 2018-05-24:
When you last time updated arduino-esp32 library and ESP32BLE_library?

luizahiginoss, 2018-05-24:
Today! I'm using the latest version. Now the connection not even starts. The only alteration that I've made in the code were the UUIDs and the name of the server for "ESP32" 
This is what the serial monitor shows me 

[image]

chegewara, 2018-05-24:
I just tested it and all works with changes in BLE_server:
```
void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

  BLEDevice::init("ESP32");
  BLEServer *pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);
  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );

  pCharacteristic->setValue("Hello World says Neil");
  pService->start();
  BLEAdvertising *pAdvertising = pServer->getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);    //<--- ADD THIS LINE
  pAdvertising->start();
  Serial.println("Characteristic defined! Now you can read it in your phone!");
}
```

Assuming that client and server UUID's match.

luizahiginoss, 2018-05-24:
Thanks, it works now!

closed 2018-05-24 by luizahiginoss