nkolban/esp32-snippets #534: Support removing services

jim-ber, 2018-05-24:
Currently, there is no way to remove a service that we have added to the server.

That would be useful for applications that change the GATT Server services during runtime.

chegewara, 2018-05-25:
I will try to add this when i end with 
https://github.com/nkolban/esp32-snippets/issues/532

chegewara, 2018-06-19:
Sorry it takes so long but i am busy with paid projects. I will try to add this this week, but no promise.

chegewara, 2018-06-19:
@jim-ber Could you try this before i make PR? 
https://github.com/chegewara/esp32-snippets

jim-ber, 2018-06-19:
@chegewara I'll do that today. Thanks for all your work!

Let you know what I find.

jim-ber, 2018-06-19:
I'm making a few changes tonight and i'll let you know.

chegewara, 2018-06-20:
@jim-ber It looks pretty solid, did you test it with some example code?
I think you can make PR

hansmbakker, 2018-12-15:
@jim-ber I see you made a PR to fix this, thank you! Could you please close this issue to keep the issue list clean?

closed 2018-12-30 by chegewara