nkolban/esp32-snippets #537: How I can realize PROPERTY_READ?

einiemand, 2018-05-28:
Hey guys.

I've been studying BLE and trying to read data sent from PC. After modifying the demo BLE_uart, I can receive messages sent from iPhone and view them on my PC. However, reading data from my PC via ESP32 and displaying them on my iPhone never succeeded. I modified my code mutiple times but still couldn't figure out how to write the callback function onRead() correctly. You may view my code in the attachment.
[rw.zip](https://github.com/nkolban/esp32-snippets/files/2044282/rw.zip)

chegewara, 2018-05-28:
onRead method is called when central device is reading from characteristic. This way for example, when central reads from characteristic you can write new value, or track how many times characteristic has been read or something like that.

`pReadCharacteristic->addDescriptor(new BLE2902());`
You can have this line in your code but this will do nothing, because client app will never register for notify or indications because characteristic does not have `PROPERTY_NOTIFY`.

One question, how do you plan to read from PC to send data to iPhone?

einiemand, 2018-05-28:
Thanks for answering! What I want to do is to read the message, which is sent from PC, on a bluetooth app on my cellphone. I tried to combine notify with read by adding 
`pReadCharacteristic = pService->createCharacteristic(`
    `CHARACTERISTIC_UUID_READ,`
    `BLECharacteristic::PROPERTY_READ | `
    `BLECharacteristic::PROPERTY_NOTIFY`
  `);`

`class ReadCallbacks : public BLECharacteristicCallbacks {`
 ` void onRead(BLECharacteristic *pCharacteristic) {`
   ` pCharacteristic->notify();`
  `}`
`};`
Although notify works well independently as I can see data displayed on my cellphone, the combination doesn't work inside the onRead() function. One problem about notify is that you have to call `pTxCharacteristic->setValue()` before sending data, which means only limited kinds of messages can be sent. But I want to receive what I type on PC. Could you please tell me how to do it? I believe many new learners have the same question as I do.

chegewara, 2018-05-28:
I can try to help you, but still, i dont know (i dont understand) how you communicate PC with esp32. Is it over ble or over UART?

einiemand, 2018-05-28:
it’s uart. The cellphone communicates with esp32 over ble.

chegewara, 2018-05-28:
Ok, i can help you with ble part. About how to deal with UART you have to read on forum or here:
https://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/uart.html
Or study some examples.

Basically, when you will read byte or bytes from UART then you will update txCharacteristic value and send notify.

einiemand, 2018-05-28:
Thanks for your information! Will
try it