nkolban/esp32-snippets #538: Multiple Servers

luizahiginoss, 2018-05-28:
I'm trying to develop a bluetooth network with one client and several servers, but they communicate with the client based on "notify". If they have the same service UUID and characteristic UUID i won't be able to do that? Probably i'm just getting confused with GATT and GAP aplications for BLE... Can u help me? 

I haven't started the tests yet, but I'm using the BLE examples...

luizahiginoss, 2018-06-13:
I tried to use one ESP32 as the client and other 3 ESPs as servers. The client only connect with one server and receive the value from its characteristic. The other 2 continue to advertise, but their characteristics values are never showed and the connection never completed.

luizahiginoss, 2018-06-13:
The client code I'm using is the one below
````
/**
 * A BLE client example that is rich in capabilities.
 */

#include "BLEDevice.h"
//#include "BLEScan.h"

// The remote service we wish to connect to.
static BLEUUID serviceUUID("91bad492-b950-4226-aa2b-4ede9fa42f59");
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID("0d563a58-196a-48ce-ace2-dfec78acc814");

static BLEAddress *pServerAddress;
static boolean doConnect = false;
static boolean connected = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;
const uint8_t v[]={0x1,0x0};

static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
    Serial.print("Notify callback for characteristic ");
    Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
    Serial.print(" of data length ");
    Serial.println(length);
}

bool connectToServer(BLEAddress pAddress) {
    Serial.print("Forming a connection to ");
    Serial.println(pAddress.toString().c_str());
    
    BLEClient*  pClient  = BLEDevice::createClient();
    Serial.println(" - Created client");

    // Connect to the remove BLE Server.
    pClient->connect(pAddress);
    Serial.println(" - Connected to server");

    // Obtain a reference to the service we are after in the remote BLE server.
    BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
    if (pRemoteService == nullptr) {
      Serial.print("Failed to find our service UUID: ");
      Serial.println(serviceUUID.toString().c_str());
      return false;
    }
    Serial.println(" - Found our service");

    // Obtain a reference to the characteristic in the service of the remote BLE server.
    pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
    if (pRemoteCharacteristic == nullptr) {
      Serial.print("Failed to find our characteristic UUID: ");
      Serial.println(charUUID.toString().c_str());
      return false;
    }
    Serial.println(" - Found our characteristic");

    // Read the value of the characteristic.
    std::string value = pRemoteCharacteristic->readValue();
    Serial.print("The characteristic value was: ");
    Serial.println(value.c_str());

    pRemoteCharacteristic->registerForNotify(notifyCallback);
    pRemoteCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)v,2,true);
}	
/**
 * Scan for BLE servers and find the first one that advertises the service we are looking for.
[... 152 lines cut ...]
  Serial.println("Waiting a client connection to notify...");
}

void loop() {
    // notify changed value
    if (deviceConnected) {
        pCharacteristic->setValue("Hello world says Luli");
        pCharacteristic->notify();
        value++;
        delay(20000); // bluetooth stack will go into congestion, if too many packets are sent
    }
    // disconnecting
    if (!deviceConnected && oldDeviceConnected) {
        delay(500); // give the bluetooth stack the chance to get things ready
        pServer->startAdvertising(); // restart advertising
        Serial.println("start advertising");
        oldDeviceConnected = deviceConnected;
    }
    // connecting
    if (deviceConnected && !oldDeviceConnected) {
        // do stuff here on connecting
        oldDeviceConnected = deviceConnected;
    }
}
````

nsamala, 2018-06-13:
It looks like the code above doesn't try connecting because `doConnect` gets set to `false` after the first connection. I can post an example of it connecting to multiple servers later today.

luizahiginoss, 2018-06-14:
Ok! Ty

luizahiginoss, 2018-06-15:
I was able to connect two devices (now using server code) making a few changes in the Client code. It works for at least 1 ou 2 minutes. 

````
Forming a connection to 30:ae:a4:92:0b:be
 - Created client
 - Connected to server
 - Found our service
 - Found our characteristic
The characteristic value was: Hello World says Xastre
We are now connected to the BLE Server. 1
E (167657) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0016
E (167659) BT: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x0016
Forming a connection to 30:ae:a4:92:6b:72
 - Created client
 - Connected to server
 - Found our service
 - Found our characteristic
The characteristic value was: Hello World says Luli
We are now connected to the BLE Server 2.
E (171227) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0016
E (171230) BT: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x0016
Forming a connection to 30:ae:a4:92:0b:be
 - Created client
 - Connected to server
 - Found our service
 - Found our characteristic
The characteristic value was: Hello World says Xastre
We are now connected to the BLE Server. 1
E (174902) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0016
E (174905) BT: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x0016
````
 --> So far, the messages are completely normal because I am using pClient->disconnect();, and then this happens 

````
Forming a connection to 30:ae:a4:92:6b:72
 - Created client
E (178064) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x003e
E (178065) BT: p_conn is NULL in bta_gattc_conn_cback

E (178066) BT: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x003e
E (178073) BT: p_conn is NULL in bta_gattc_conn_cback

We are now connected to the BLE Server 2.

We are now connected to the BLE Server 2.
Forming a connection to 30:ae:a4:92:0b:be
 - Created client
 - Connected to server
 - Found our service
 - Found our characteristic
The characteristic value was: Hello World says Xastre
We are now connected to the BLE Server. 1
E (181757) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0016
E (181760) BT: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x0016
E (181767) BT: bta_gattc_conn_cback() - cif=5 connected=0 conn_id=5 reason=0x0016
Forming a connection to 30:ae:a4:92:6b:72
 - Created client
E (184756) BT: application already registered.
E (184756) BT: Register with GATT stack failed.

E (184758) BT: application already registered.
E (184762) BT: Register with GATT stack failed.

E (184766) BT: bta_gattc_process_api_open Failed, unknown client_if: 0
````

After this, everything stops. 

The code I'm using for client is in this txt file 

[BLE_client.txt](https://github.com/nkolban/esp32-snippets/files/2107312/BLE_client.txt)

Donderda, 2018-06-27:
Were you able to fix this problem?

luizahiginoss, 2018-06-28:
No, I wasn't...

mangooose, 2018-07-25:
Can you please provide multiple servers connection example? 
I was able to connect to 2 servers and get the characteristics but somehow writeValue works only for the first server and hangs there.

chegewara, 2018-10-10:
@luizahiginoss @mangooose @Donderda @nsamala 
Ok, i have good news. I have working code for one esp32 client connecting to multiple peripherals. As for now works connecting, disconnecting and register for notifications with receiving notifications of course.

Only thing that does not work for me, but im pretty sure its because of my simplified main_app code, is write to remote characteristic. In theory it suppose to work too.

All i have to do now is to cleanup code and make some improvements, because i have to marge multiple clients with multiple servers.

mangooose, 2018-10-10:
@chegewara Great :D thanks a lot. Which ESP32 module are you using?

chegewara, 2018-10-10:
I am using devKitC and analog lamb esp32 wrover mini.

chegewara, 2018-10-10:
I would like to apology for mess in code, because i am working on many additions, changes and bugfixes now. This will be code for testing purposes only. I will create new branch for it and all issues or questions please post on my repository:

https://github.com/chegewara/esp32-snippets-enchancements-test/tree/multi_connect_test

Starting point to write own client and server are SampleServer and SampleClient from main subfolder (its just suggestion).

chegewara, 2018-10-11:
Added arduino compatibility, you just need to export library to arduino on your own.
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md#replacing-the-version-that-comes-with-arduino-esp32

chegewara, 2018-10-12:
Hi,
i have to solve one or two issues and i can post nicely working version that can be used both ways (one server to multi client and one client to multi peripherals). Biggest issue now is how esp-idf stack works. When esp32 is running in central/master role (BLEClient) and there is few peripherals connected, when one peripheral got disconnected bt stack is disconnecting all peripheral devices. This and few more small issues drives me crazy, but i am working on it. If there is request to publish current version library to my repository just tell me.

mangooose, 2018-10-12:
@chegewara currently I use the latest code in master and i couldn't connect to multiple servers should i use [this](https://github.com/chegewara/esp32-snippets-enchancements-test/tree/multi_connect_test) repo ?

chegewara, 2018-10-12:
Yes, but its not published yet. Give me few minutes.

I have nice rig setup now:
- 1xesp32 working as central client,
- 4xesp32 working as peripheral server,
- 1xsmartphone (or more) with nRF connect connected to all peripherals at the same time for monitoring.

Now, esp32 client is performing infinite scanning for peripherals, when peripheral is found client is connecting to it and starts sending write message to remote characteristic on that peripheral. This is happening every 1000 ms with `vTaskDelayUntil`. When peripheral receive write request then its sending echo with notification to central. As for now all works fine, from time to time i have central crash, but it seems to be esp-idf bt stack related.
With smartphone i can monitor peripherals.
So i can have one central esp32 connected to many peripherals that also can be connected to many centrals.

chegewara, 2018-10-12:
@mangooose Most API has not been changed, but in BLEScan is few changes. Those are not breaking older code changes but are useful during multi connecting.

mangooose, 2018-10-12:
@chegewara which repo should I use?

chegewara, 2018-10-12:
https://github.com/chegewara/esp32-snippets-enchancements-test/tree/multi_connect_test

chegewara, 2018-11-29:
Implemented in latest version.

victorChB, 2019-01-07:
> https://github.com/chegewara/esp32-snippets-enchancements-test/tree/multi_connect_test

@chegewara I can't acces to the link, is it merged somewhere?

chegewara, 2019-01-07:
Its merged already in master.

Bryanbaring, 2019-02-08:
"404 didnt find anyrhing here" . i cant open the link.

chegewara, 2019-02-08:
Its because repository is no longer public since code is merged here.

aaronchan1992, 2019-04-26:
> Yes, but its not published yet. Give me few minutes.
> 
> I have nice rig setup now:
> 
> * 1xesp32 working as central client,
> * 4xesp32 working as peripheral server,
> * 1xsmartphone (or more) with nRF connect connected to all peripherals at the same time for monitoring.
> 
> Now, esp32 client is performing infinite scanning for peripherals, when peripheral is found client is connecting to it and starts sending write message to remote characteristic on that peripheral. This is happening every 1000 ms with `vTaskDelayUntil`. When peripheral receive write request then its sending echo with notification to every connected client device. As for now all works fine, from time to time i have central crash, but it seems to be esp-idf bt stack related.
> With smartphone i can monitor peripherals.
> So i can have one central esp32 connected to many peripherals that also can be connected to many centrals.

Hi Sir, I am currently working on a parking project which is similar to your implementation above. A ble central client for receiving data from multiple peripheral server (parking sensor). Once the BLE server devices are connected, the data can be display onto the web server.  

May I know how can I go about doing it.

Thank you.

helmym, 2019-06-02:
@chegeware 
I can’t access to the link 
And I need esp32 ble code to connect BLE server to 4 Clint have the same uuid

Donderda, 2019-06-03:
Its already in the latest master 

Written with two thumbs on my mobile device...

Jerry0045, 2019-07-01:
@chegewara 
Hello,
I´m trying to connect one ESP32 as the client and other 6 ESPs as servers but i'm able to connect client to only one server. You said that you are able to connect multiple servers and its in the master but i'm not able to find it. Can you please help me where i can find it or is there any example?
Thanks for response.

chegewara, 2019-07-01:
No, there is no example, at least i dont know any.

You have to remember that with arduino its limited to 3 simultaneous connection and with esp-idf you need to chanage setting  with make menuconfig.

There is nothing special you have to do, just create BLEClient for every server you want to connect to.

luizahiginoss, 2019-08-09:
@chegewara I tried to find your multiple client code, but i couldn't. Can you show me the link?

chegewara, 2019-08-09:
This is very old code and its for sure not production ready, but it should works:
```
/**
 * Create a sample BLE client that connects to a BLE server and then retrieves the current
 * characteristic value.  It will then periodically update the value of the characteristic on the
 * remote server with the current time since boot.
 */
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <esp_log.h>
#include <string>
#include <sstream>
#include <sys/time.h>
#include "BLEDevice.h"
#include "FreeRTOS.h"
#include <nvs_flash.h>

#include "BLEAdvertisedDevice.h"
#include "BLEClient.h"
#include "BLEScan.h"
#include "BLEUtils.h"
#include "Task.h"

#include "sdkconfig.h"
#define TICKS_TO_DELAY 1000
BLEScan *pBLEScan;
static const char* LOG_TAG = "TEST";
// esp_ble_addr_type_t addressType;
TaskHandle_t handle;
TaskHandle_t xTaskGetHandle( const char *pcWriteBuffer );

static BLEUUID serviceUUID("4fafc201-1fb5-459e-8fcc-c5c9c331914b");
static BLEUUID    charUUID1("beb5483e-36e1-4688-b7f5-ea07361b26a8");
static BLEUUID    charUUID2("beb5483e-36e2-4688-b7f5-ea07361b26a8");
static BLEUUID    charUUID3("beb5483e-36e3-4688-b7f5-ea07361b26a8");

void scan1(void*);

static void notifyCallback(
	BLERemoteCharacteristic* pBLERemoteCharacteristic,
	uint8_t* pData,
	size_t length,
	bool isNotify) {
		ESP_LOGE(LOG_TAG, "Notify callback for characteristic %s of data %s length %d",
				pBLERemoteCharacteristic->getUUID().toString().c_str(), ((char*) pData), length);
}

class MyCallbacks : public BLEClientCallbacks {
	void onConnect(BLEClient* pC){
        	// xTaskCreate(scan1, "scan", 4048, NULL, 5, NULL);
	}
	void onDisconnect(BLEClient* pC) {
		// pBLEScan->erase(pC->getPeerAddress());
	}
};
esp_ble_addr_type_t type;
class MyClient: public Task {
	void run(void* data) {
 		ESP_LOGI(LOG_TAG, "Advertised Device: %s", ((BLEAddress*)data)->toString().c_str());
		 BLEAddress addr = *(BLEAddress*)data;
       BLEClient*  pClient  = BLEDevice::createClient();
        pClient->setClientCallbacks(new MyCallbacks());
		// esp_ble_gap_set_prefer_conn_params(*addr.getNative(), 100,125, 2, 0x03e8);
        pClient->connect(addr, type);

        // xTaskCreate(scan1, "scan", 4048, NULL, 5, NULL);

		BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
		if (pRemoteService == nullptr) {
			ESP_LOGD(LOG_TAG, "Failed to find our service UUID: %s", serviceUUID.toString().c_str());
[... 107 lines cut ...]
}
// this code is only for debug purpose, to see if we have problem wit esp-idf btuT task stack
TaskHandle_t xTaskGetHandle( const char *pcWriteBuffer )
{
TaskStatus_t *pxTaskStatusArray;
volatile UBaseType_t uxArraySize, x;
uint32_t ulTotalRunTime;
	uxArraySize = uxTaskGetNumberOfTasks();
	pxTaskStatusArray = (TaskStatus_t *)pvPortMalloc( uxArraySize * sizeof( TaskStatus_t ) );
	if( pxTaskStatusArray != NULL )
	{
    	uxArraySize = uxTaskGetSystemState( pxTaskStatusArray, uxArraySize, &ulTotalRunTime );
		for( x = 0; x < uxArraySize; x++ )
		{
			ESP_LOGW(LOG_TAG, "%s highwater--> %d, state: %d ", pxTaskStatusArray[x].pcTaskName, uxTaskGetStackHighWaterMark(pxTaskStatusArray[x].xHandle), (uint8_t)pxTaskStatusArray[x].eCurrentState);
			if(strncmp(pxTaskStatusArray[x].pcTaskName, pcWriteBuffer, strlen(pcWriteBuffer)) == 0){
				vPortFree( pxTaskStatusArray );
				return  pxTaskStatusArray[x].xHandle;
			}
		}
    	vPortFree( pxTaskStatusArray );
	}
	return nullptr;
}
```

Chedeshri, 2019-08-16:
> This is very old code and its for sure not production ready, but it should works:
>
> ```
> /**
>  * Create a sample BLE client that connects to a BLE server and then retrieves the current
>  * [...]

Hello chegewara, I am trying the same example and i am getting guru meditation error which i opened as other issue , my application is same as mentioend in this example, i am working on that error since a month, could you give me step by step guidance on this one, i think i am missing somehting, i coulnt thank you more ! 

Regards,

chegewara, 2019-08-17:
I can try to help with errors, but you need to provide some more info and logs.

AS123-y, 2021-04-15:
hi @luizahiginoss @chegewara @mangooose 

I'm trying to develop a bluetooth network with one client and several servers, but they communicate with the client based on "notify". each server is connected to a sensor (DHT11).
i am able to connect to one server and get the temp characteristic.

i need a code for one esp32 client connecting to multiple ESP32 server .

I'm using the BLE examples

AS123-y, 2021-04-15:
i came across this "multi-connection example’s main source file is gattc_multi_connect.c." [https://github.com/espressif/esp-idf/blob/a6f33a9d366e8179fa7d18a81f205254a15db117/examples/bluetooth/bluedroid/ble/gattc_multi_connect/main/gattc_multi_connect.c](url) 
but it seems its not for arduino IDE

guevaj5, 2021-11-24:
hi @AS123-y 

I am currently attempting to build a bluetooth network of ESP32s one client and about 3 servers, similar to your project. each server would be connected to accelerometers. I just plan on sending the different accelerometer values to the client, which would display them on some LCD display. Were you able to figure out your project?

AS123-y, 2021-11-25:
> 
hi

AS123-y, 2021-11-25:
hi @guevaj5 
no, I still didn't figure out a solution. Please keep me updated on the status of things.

helmym, 2021-11-25:
Can you send me WhatsApp please
00201121106662

Sireevenkat, 2023-02-09:
@luizahiginoss @mangooose @AS123-y   Do you find solution to connect multiple ESP32 BLE servers with one ESP32 BLE client on Arduino IDE?               
@luizahiginoss  [BLE_client.txt] Did you solved the issue?

helmym, 2023-02-09:
Hello
Yes I had do it
Send me your issue and we will discuss it

AS123-y, 2023-02-09:
> @luizahiginoss @mangooose @AS123-y Do you find solution to connect multiple ESP32 BLE servers with one ESP32 BLE client on Arduino IDE? @luizahiginoss [BLE_client.txt] Did you solved the issue?

Hi @Sireevenkat 
i still did not find a solution 
@helmym can you share your findings?

helmym, 2023-02-09:
Send me WhatsApp please 00201121106662

Sireevenkat, 2023-02-13:
Hi,
@helmym,@chegewara,@AS123-y 
 I am working with ESP32-S3 module on Arduino IDE.I need to connect  multiple BLE servers to one client and able send and receive data from client to all servers at a time. Can You share your findings.
Thanks