nkolban/esp32-snippets #539: FTPServer c++ class usage

snahmad, 2018-05-29:
Hi,

How to use FTPServer with example to use with spi flash  and with sd card.

I also getting linker error. getcwd and chdir. I need to give function definitions for these.
Can you provide example function definitions for these.

:/Work/Actisense/esp32/build/cpp_utils\libcpp_utils.a(FTPServer.o):(.literal._ZN9FTPServer19getCurrentDirectoryB5cxx11Ev+0x0): undefined reference to `getcwd'
C:/Work/Actisense/esp32/build/cpp_utils\libcpp_utils.a(FTPServer.o):(.literal._ZN9FTPServer5onCwdERNSt7__cxx1119basic_istringstreamIcSt11char_traitsIcESaIcEEE+0x8): undefined reference to `chdir'
C:/Work/Actisense/esp32/build/cpp_utils\libcpp_utils.a(FTPServer.o): In function `FTPServer::getCurrentDirectory[abi:cxx11]()':
C:/Work/LibDev/esp32/esp-idf/components/cpp_utils/FTPServer.cpp:229: undefined reference to `getcwd'
C:/Work/Actisense/esp32/build/cpp_utils\libcpp_utils.a(FTPServer.o): In function `FTPServer::onCwd(std::__cxx11::basic_istringstream<char, std::char_traits<char>, std::allocator<char> >&)':
C:/Work/LibDev/esp32/esp-idf/components/cpp_utils/FTPServer.cpp:229: undefined reference to `chdir'
collect2.exe: error: ld returned 1 exit status
make: *** [C:/Work/LibDev/esp32/esp-idf/make/project.mk:389: /c/Work/Actisense/esp32/build/app-template.elf] Error 1

squonk11, 2018-05-29:
The FTPServer class is quite new in this repository. So, I am not sure if ever someone used this class except Mr. Kolban. 
Unfortunately I currently do not have the time to work on my ESP32 project - otherwise we could try to get things working together as it seems that we have quite similar targets. My project finally also should use HttpServer with SSL serving a bigger size web page from an SD card. The web page will be made using a higher level Javascript framework and an FTP Server for updating this website. Later on I also would like to add MQTT protocol.

squonk11, 2018-05-29:
no, also this I did not try until now. I am quite in the beginning of my project and I don't know if will ever be able to finish it. 
Until now I just tested HttpServer and mongoose server to serve a smaller web page. This works so far. I decided to stay with HttpServer as it is free SW because if I finish the project it might become a comercial thing. 
Maybe in a few weeks I will have time again to continue on this project.

snahmad, 2018-05-29:
ok. let me know outcome

squonk11, 2018-05-29:
o.k.

snahmad, 2018-12-13:
Have some one used this FTPServer yet. with SD card?

fa1ke5, 2019-03-01:
https://github.com/fa1ke5/ESP32_FTPServer_SD_MMC

lkozinakov, 2019-03-07:
I have tried to use this FTP code as well, but I keep getting the same "undefined reference to getcwd" errors during the linking stage of the build. There must be some problems with the #include <unistd.h> line at the top of the code, but I can't seem to figure out what exactly. I'll post if I find a fix.