nkolban/esp32-snippets #540: Q: Is this project still up to date with the latest expressif32?

neilyoung, 2018-05-30:
as subject says
Thanks

neilyoung, 2018-05-30:
@chegewara Wow, what a quick response :)

OK, this is the state:

I'm having here a pretty old Sparkfun ESP32, but it works well with ordinary stuff, so I can compile and upload and run and debug via Serial.

I was testing this example: https://platformio.org/lib/show/1841/ESP32%20BLE%20Arduino/examples

Here is my platform.ini

```
[platformio]
env_default = main

[env:main]
platform = espressif32
board = esp32thing
framework = arduino
lib_deps = ESP32 BLE Arduino

```

and while compiling I get these error list:
```

[Wed May 30 17:10:53 2018] Processing main (platform: espressif32; board: esp32thing; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 32 > SparkFun ESP32 Thing
SYSTEM: ESP32 240MHz 288KB RAM (1.25MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 19 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP32 BLE Arduino> v0.4.13
Compiling .pioenvs/main/src/main.cpp.o
Compiling .pioenvs/main/lib26f/ESP32 BLE Arduino_ID1841/BLEClient.cpp.o
Compiling .pioenvs/main/lib26f/ESP32 BLE Arduino_ID1841/BLEDescriptor.cpp.o
Compiling .pioenvs/main/lib26f/ESP32 BLE Arduino_ID1841/BLERemoteCharacteristic.cpp.o
Compiling .pioenvs/main/lib26f/ESP32 BLE Arduino_ID1841/BLERemoteDescriptor.cpp.o
Compiling .pioenvs/main/lib26f/ESP32 BLE Arduino_ID1841/BLERemoteService.cpp.o
Compiling .pioenvs/main/lib26f/ESP32 BLE Arduino_ID1841/GeneralUtils.cpp.o
Archiving .pioenvs/main/libFrameworkArduinoVariant.a
Indexing .pioenvs/main/libFrameworkArduinoVariant.a
Compiling .pioenvs/main/FrameworkArduino/Esp.cpp.o
Compiling .pioenvs/main/FrameworkArduino/HardwareSerial.cpp.o
.piolibdeps/ESP32 BLE Arduino_ID1841/src/BLEDescriptor.cpp: In member function 'void BLEDescriptor::handleGATTServerEvent(esp_gatts_cb_event_t, esp_gatt_if_t, esp_ble_gatts_cb_param_t*)':
.piolibdeps/ESP32 BLE Arduino_ID1841/src/BLEDescriptor.cpp:158:53: error: 'struct esp_ble_gatts_cb_param_t::gatts_add_char_descr_evt_param' has no member named 'descr_uuid'
m_bleUUID.equals(BLEUUID(param->add_char_descr.descr_uuid)) &&
^
.piolibdeps/ESP32 BLE Arduino_ID1841/src/BLEClient.cpp: In member function 'bool BLEClient::connect(BLEAddress)':
.piolibdeps/ESP32 BLE Arduino_ID1841/src/BLEClient.cpp:116:2: error: too many arguments to function 'esp_err_t esp_ble_gattc_open(esp_gatt_if_t, uint8_t*, bool)'
);
^
In file included from .piolibdeps/ESP32 BLE Arduino_ID1841/src/BLEClient.cpp:13:0:
/Users/decades/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/bluedroid/esp_gattc_api.h:293:11: note: declared here
esp_err_t esp_ble_gattc_open(esp_gatt_if_t gattc_if, esp_bd_addr_t remote_bda, bool is_direct);
^
.piolibdeps/ESP32 BLE Arduino_ID1841/src/BLEClient.cpp: In member function 'BLERemoteService* BLEClient::getService(BLEUUID)':
.piolibdeps/ESP32 BLE Arduino_ID1841/src/BLEClient.cpp:340:12: error: exception handling disabled, use -fexceptions to enable
throw new BLEUuidNotFoundException;
^
*** [.pioenvs/main/lib26f/ESP32 BLE Arduino_ID1841/BLEDescriptor.cpp.o] Error 1
.piolibdeps/ESP32 BLE Arduino_ID1841/src/BLERemoteCharacteristic.cpp: In member function 'std::__cxx11::string BLERemoteCharacteristic::readValue()':
.piolibdeps/ESP32 BLE Arduino_ID1841/src/BLERemoteCharacteristic.cpp:439:34: error: exception handling disabled, use -fexceptions to enable
throw BLEDisconnectedException();
^
.piolibdeps/ESP32 BLE Arduino_ID1841/src/BLERemoteDescriptor.cpp: In member function 'std::__cxx11::string BLERemoteDescriptor::readValue()':
.piolibdeps/ESP32 BLE Arduino_ID1841/src/BLERemoteDescriptor.cpp:64:34: error: exception handling disabled, use -fexceptions to enable
throw BLEDisconnectedException();
^
.piolibdeps/ESP32 BLE Arduino_ID1841/src/BLERemoteService.cpp: In member function 'BLERemoteCharacteristic* BLERemoteService::getCharacteristic(BLEUUID)':
.piolibdeps/ESP32 BLE Arduino_ID1841/src/BLERemoteService.cpp:156:37: error: exception handling disabled, use -fexceptions to enable
throw new BLEUuidNotFoundException();
^
*** [.pioenvs/main/lib26f/ESP32 BLE Arduino_ID1841/BLEClient.cpp.o] Error 1
*** [.pioenvs/main/lib26f/ESP32 BLE Arduino_ID1841/BLERemoteDescriptor.cpp.o] Error 1
*** [.pioenvs/main/lib26f/ESP32 BLE Arduino_ID1841/BLERemoteCharacteristic.cpp.o] Error 1
*** [.pioenvs/main/lib26f/ESP32 BLE Arduino_ID1841/BLERemoteService.cpp.o] Error 1
========================== [ERROR] Took 2.21 seconds ==========================

```

neilyoung, 2018-05-30:
OK, switched to the older ESP BLE library as suggested:

Problems now:
```

[Wed May 30 17:47:51 2018] Processing main (platform: espressif32; board: esp32thing; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 32 > SparkFun ESP32 Thing
SYSTEM: ESP32 240MHz 288KB RAM (1.25MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 20 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP32 BLE Arduino> v0.4.10
Compiling .pioenvs/main/src/main.cpp.o
Generating partitions .pioenvs/main/partitions.bin
Compiling .pioenvs/main/lib302/ESP32 BLE Arduino_ID1841@0.4.10/BLE2902.cpp.o
Compiling .pioenvs/main/lib302/ESP32 BLE Arduino_ID1841@0.4.10/BLE2904.cpp.o
Compiling .pioenvs/main/lib302/ESP32 BLE Arduino_ID1841@0.4.10/BLEAddress.cpp.o
Compiling .pioenvs/main/lib302/ESP32 BLE Arduino_ID1841@0.4.10/BLEAdvertisedDevice.cpp.o
Compiling .pioenvs/main/lib302/ESP32 BLE Arduino_ID1841@0.4.10/BLEAdvertising.cpp.o
Compiling .pioenvs/main/lib302/ESP32 BLE Arduino_ID1841@0.4.10/BLEBeacon.cpp.o
Compiling .pioenvs/main/lib302/ESP32 BLE Arduino_ID1841@0.4.10/BLECharacteristic.cpp.o
Compiling .pioenvs/main/lib302/ESP32 BLE Arduino_ID1841@0.4.10/BLECharacteristicMap.cpp.o
Compiling .pioenvs/main/lib302/ESP32 BLE Arduino_ID1841@0.4.10/BLEClient.cpp.o
Compiling .pioenvs/main/lib302/ESP32 BLE Arduino_ID1841@0.4.10/BLEDescriptor.cpp.o
Compiling .pioenvs/main/lib302/ESP32 BLE Arduino_ID1841@0.4.10/BLEDescriptorMap.cpp.o
Compiling .pioenvs/main/lib302/ESP32 BLE Arduino_ID1841@0.4.10/BLEDevice.cpp.o
Compiling .pioenvs/main/lib302/ESP32 BLE Arduino_ID1841@0.4.10/BLEExceptions.cpp.o
Compiling .pioenvs/main/lib302/ESP32 BLE Arduino_ID1841@0.4.10/BLEHIDDevice.cpp.o
Compiling .pioenvs/main/lib302/ESP32 BLE Arduino_ID1841@0.4.10/BLERemoteCharacteristic.cpp.o
.piolibdeps/ESP32 BLE Arduino_ID1841@0.4.10/src/BLEClient.cpp: In member function 'bool BLEClient::connect(BLEAddress)':
.piolibdeps/ESP32 BLE Arduino_ID1841@0.4.10/src/BLEClient.cpp:115:2: error: too many arguments to function 'esp_err_t esp_ble_gattc_open(esp_gatt_if_t, uint8_t*, bool)'
);
^
In file included from .piolibdeps/ESP32 BLE Arduino_ID1841@0.4.10/src/BLEClient.cpp:13:0:
/Users/decades/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/bluedroid/esp_gattc_api.h:293:11: note: declared here
esp_err_t esp_ble_gattc_open(esp_gatt_if_t gattc_if, esp_bd_addr_t remote_bda, bool is_direct);
^
.piolibdeps/ESP32 BLE Arduino_ID1841@0.4.10/src/BLEDescriptor.cpp: In member function 'void BLEDescriptor::handleGATTServerEvent(esp_gatts_cb_event_t, esp_gatt_if_t, esp_ble_gatts_cb_param_t*)':
.piolibdeps/ESP32 BLE Arduino_ID1841@0.4.10/src/BLEDescriptor.cpp:158:53: error: 'struct esp_ble_gatts_cb_param_t::gatts_add_char_descr_evt_param' has no member named 'descr_uuid'
m_bleUUID.equals(BLEUUID(param->add_char_descr.descr_uuid)) &&
^
.piolibdeps/ESP32 BLE Arduino_ID1841@0.4.10/src/BLEClient.cpp: In member function 'BLERemoteService* BLEClient::getService(BLEUUID)':
.piolibdeps/ESP32 BLE Arduino_ID1841@0.4.10/src/BLEClient.cpp:337:12: error: exception handling disabled, use -fexceptions to enable
throw new BLEUuidNotFoundException;
^
*** [.pioenvs/main/lib302/ESP32 BLE Arduino_ID1841@0.4.10/BLEDescriptor.cpp.o] Error 1
*** [.pioenvs/main/lib302/ESP32 BLE Arduino_ID1841@0.4.10/BLEClient.cpp.o] Error 1
.piolibdeps/ESP32 BLE Arduino_ID1841@0.4.10/src/BLERemoteCharacteristic.cpp: In member function 'std::__cxx11::string BLERemoteCharacteristic::readValue()':
.piolibdeps/ESP32 BLE Arduino_ID1841@0.4.10/src/BLERemoteCharacteristic.cpp:439:34: error: exception handling disabled, use -fexceptions to enable
throw BLEDisconnectedException();
^
*** [.pioenvs/main/lib302/ESP32 BLE Arduino_ID1841@0.4.10/BLERemoteCharacteristic.cpp.o] Error 1
========================== [ERROR] Took 3.83 seconds ==========================

```

Regarding my ESP32 libs: I made a fresh install of expressif32 today (at least I tried). I updated and upgraded pio as well. Here the versions:

```
~/Documents/Arduino/ESP32Arduino $ pio platform show espressif32
espressif32 ~ Espressif 32
==========================
Espressif Systems is a privately held fabless semiconductor company. They provide wireless communications and Wi-Fi chips which are widely used in mobile devices and the Internet of Things applications.

Version: 0.12.0
Home: http://platformio.org/platforms/espressif32
[... 73 lines cut ...]
esp32doit-devkit-v1   ESP32          240MHz    1.25MB  288KB  DOIT ESP32 DEVKIT V1
espea32               ESP32          240MHz    1.25MB  288KB  April Brother ESPea32
esp32dev              ESP32          240MHz    1.25MB  288KB  Espressif ESP32 Dev Module
lolin32               ESP32          240MHz    1.25MB  288KB  WEMOS LOLIN32
onehorse32dev         ESP32          240MHz    1.25MB  288KB  Onehorse ESP32 Dev Module
nodemcu-32s           ESP32          240MHz    1.25MB  288KB  NodeMCU-32S
esp-wrover-kit        ESP32          240MHz    1.25MB  288KB  Espressif ESP-WROVER-KIT
esp32-gateway         ESP32          240MHz    1.25MB  288KB  OLIMEX ESP32-GATEWAY
hornbill32minima      ESP32          240MHz    1.25MB  288KB  Hornbill ESP32 Minima
esp320                ESP32          240MHz    1.25MB  288KB  Electronic SweetPeas ESP320
[... cut ...]
mhetesp32devkit       ESP32          240MHz    1.25MB  288KB  MH ET LIVE ESP32DevKIT
microduino-core-esp32 ESP32          240MHz    1.25MB  288KB  Microduino Core ESP32
featheresp32          ESP32          240MHz    1.25MB  288KB  Adafruit ESP32 Feather
firebeetle32          ESP32          240MHz    1.25MB  288KB  FireBeetle-ESP32
mhetesp32minikit      ESP32          240MHz    1.25MB  288KB  MH ET LIVE ESP32MiniKit
espino32              ESP32          240MHz    1.25MB  288KB  ESPino32
~/Documents/Arduino/ESP32Arduino $ 

```

chegewara, 2018-05-30:
Please try with older version: 0.4.9 or 0.4.8.

neilyoung, 2018-05-30:
0.4.9:

```
[Wed May 30 18:01:33 2018] Processing main (platform: espressif32; board: esp32thing; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 32 > SparkFun ESP32 Thing
SYSTEM: ESP32 240MHz 288KB RAM (1.25MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 22 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP32 BLE Arduino> v0.4.9
Compiling .pioenvs/main/src/main.cpp.o
Generating partitions .pioenvs/main/partitions.bin
Compiling .pioenvs/main/libb93/ESP32 BLE Arduino_ID1841@0.4.9/BLE2902.cpp.o
Compiling .pioenvs/main/libb93/ESP32 BLE Arduino_ID1841@0.4.9/BLE2904.cpp.o
Compiling .pioenvs/main/libb93/ESP32 BLE Arduino_ID1841@0.4.9/BLEAddress.cpp.o
Compiling .pioenvs/main/libb93/ESP32 BLE Arduino_ID1841@0.4.9/BLEAdvertisedDevice.cpp.o
Compiling .pioenvs/main/libb93/ESP32 BLE Arduino_ID1841@0.4.9/BLEAdvertising.cpp.o
Compiling .pioenvs/main/libb93/ESP32 BLE Arduino_ID1841@0.4.9/BLEBeacon.cpp.o
Compiling .pioenvs/main/libb93/ESP32 BLE Arduino_ID1841@0.4.9/BLECharacteristic.cpp.o
Compiling .pioenvs/main/libb93/ESP32 BLE Arduino_ID1841@0.4.9/BLECharacteristicMap.cpp.o
Compiling .pioenvs/main/libb93/ESP32 BLE Arduino_ID1841@0.4.9/BLEClient.cpp.o
Compiling .pioenvs/main/libb93/ESP32 BLE Arduino_ID1841@0.4.9/BLEDescriptor.cpp.o
Compiling .pioenvs/main/libb93/ESP32 BLE Arduino_ID1841@0.4.9/BLEDescriptorMap.cpp.o
Compiling .pioenvs/main/libb93/ESP32 BLE Arduino_ID1841@0.4.9/BLEDevice.cpp.o
Compiling .pioenvs/main/libb93/ESP32 BLE Arduino_ID1841@0.4.9/BLEExceptions.cpp.o
Compiling .pioenvs/main/libb93/ESP32 BLE Arduino_ID1841@0.4.9/BLEHIDDevice.cpp.o
Compiling .pioenvs/main/libb93/ESP32 BLE Arduino_ID1841@0.4.9/BLERemoteCharacteristic.cpp.o
.piolibdeps/ESP32 BLE Arduino_ID1841@0.4.9/src/BLEClient.cpp: In member function 'BLERemoteService* BLEClient::getService(BLEUUID)':
.piolibdeps/ESP32 BLE Arduino_ID1841@0.4.9/src/BLEClient.cpp:336:12: error: exception handling disabled, use -fexceptions to enable
throw new BLEUuidNotFoundException;
^
*** [.pioenvs/main/libb93/ESP32 BLE Arduino_ID1841@0.4.9/BLEClient.cpp.o] Error 1
.piolibdeps/ESP32 BLE Arduino_ID1841@0.4.9/src/BLERemoteCharacteristic.cpp: In member function 'std::__cxx11::string BLERemoteCharacteristic::readValue()':
.piolibdeps/ESP32 BLE Arduino_ID1841@0.4.9/src/BLERemoteCharacteristic.cpp:439:34: error: exception handling disabled, use -fexceptions to enable
throw BLEDisconnectedException();
^
*** [.pioenvs/main/libb93/ESP32 BLE Arduino_ID1841@0.4.9/BLERemoteCharacteristic.cpp.o] Error 1
========================== [ERROR] Took 11.14 seconds ==========================

```

0.4.8
```

[Wed May 30 18:02:20 2018] Processing main (platform: espressif32; board: esp32thing; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 32 > SparkFun ESP32 Thing
SYSTEM: ESP32 240MHz 288KB RAM (1.25MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 22 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP32 BLE Arduino> v0.4.8
Compiling .pioenvs/main/src/main.cpp.o
Generating partitions .pioenvs/main/partitions.bin
Compiling .pioenvs/main/libae0/ESP32 BLE Arduino_ID1841@0.4.8/BLE2902.cpp.o
Compiling .pioenvs/main/libae0/ESP32 BLE Arduino_ID1841@0.4.8/BLE2904.cpp.o
Compiling .pioenvs/main/libae0/ESP32 BLE Arduino_ID1841@0.4.8/BLEAddress.cpp.o
Compiling .pioenvs/main/libae0/ESP32 BLE Arduino_ID1841@0.4.8/BLEAdvertisedDevice.cpp.o
Compiling .pioenvs/main/libae0/ESP32 BLE Arduino_ID1841@0.4.8/BLEAdvertising.cpp.o
Compiling .pioenvs/main/libae0/ESP32 BLE Arduino_ID1841@0.4.8/BLEBeacon.cpp.o
Compiling .pioenvs/main/libae0/ESP32 BLE Arduino_ID1841@0.4.8/BLECharacteristic.cpp.o
Compiling .pioenvs/main/libae0/ESP32 BLE Arduino_ID1841@0.4.8/BLECharacteristicMap.cpp.o
Compiling .pioenvs/main/libae0/ESP32 BLE Arduino_ID1841@0.4.8/BLEClient.cpp.o
[... 7 lines cut ...]
.piolibdeps/ESP32 BLE Arduino_ID1841@0.4.8/src/BLEHIDDevice.h:18:22: fatal error: HIDTypes.h: No such file or directory

******************************************************************
* Looking for HIDTypes.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:HIDTypes.h"
* Web  > https://platformio.org/lib/search?query=header:HIDTypes.h
*
******************************************************************

compilation terminated.
*** [.pioenvs/main/libae0/ESP32 BLE Arduino_ID1841@0.4.8/BLEHIDDevice.cpp.o] Error 1
.piolibdeps/ESP32 BLE Arduino_ID1841@0.4.8/src/BLEClient.cpp: In member function 'BLERemoteService* BLEClient::getService(BLEUUID)':
.piolibdeps/ESP32 BLE Arduino_ID1841@0.4.8/src/BLEClient.cpp:336:12: error: exception handling disabled, use -fexceptions to enable
throw new BLEUuidNotFoundException;
^
*** [.pioenvs/main/libae0/ESP32 BLE Arduino_ID1841@0.4.8/BLEClient.cpp.o] Error 1
.piolibdeps/ESP32 BLE Arduino_ID1841@0.4.8/src/BLERemoteCharacteristic.cpp: In member function 'std::__cxx11::string BLERemoteCharacteristic::readValue()':
.piolibdeps/ESP32 BLE Arduino_ID1841@0.4.8/src/BLERemoteCharacteristic.cpp:439:34: error: exception handling disabled, use -fexceptions to enable
throw BLEDisconnectedException();
^
*** [.pioenvs/main/libae0/ESP32 BLE Arduino_ID1841@0.4.8/BLERemoteCharacteristic.cpp.o] Error 1
========================== [ERROR] Took 6.03 seconds ==========================

```

chegewara, 2018-05-30:
Good, 0.4.9 will be good. Now last one thing:
https://github.com/nkolban/esp32-snippets/issues/393

neilyoung, 2018-05-30:
0.4.7 compiles successfully. What is wrong with my expressif32 version?

chegewara, 2018-05-30:
Platformio espressif32 is very old(+/- 5 months) if it compiles with 0.4.7.

neilyoung, 2018-05-30:
OK, did I made a mistake on upgrading? I uninstalled the old and re-installed again. What can go wrong there?

Other than that: The 0.4.9 compiles well with #393 applied

Thanks

chegewara, 2018-05-30:
Like i said, im not familiar with platformio. You can ask on their forum or esp32.com.

neilyoung, 2018-05-30:
@chegewara Thanks for your help. The sample works.