nkolban/esp32-snippets #542: difficulty including bootwifi as component in IDF build ("WiFi does not name a type")

hamishcunningham, 2018-05-31:
Great library, thanks!

I'm trying to include bootwifi in an existing IDF build, so I copied this (and cpp_utils) into my components directory, ran make menuconfig and then make, but I get this compile error (which seems to be to do with which Wifi.h gets included?):

CXX build/bootwifi/BootWiFi.o
In file included from .../components/bootwifi/./BootWiFi.cpp:24:0:
.../components/bootwifi/./BootWiFi.h:22:2: error: 'WiFi' does not name a type
  WiFi        m_wifi;
  ^

Any ideas?
Thanks!

chegewara, 2018-05-31:
To make it easier just copy BootWiFi to your main folder. Apart from that, do you have component.mk in bootwifi folder? What is your components folder structure?

hamishcunningham, 2018-05-31:
Thanks, I tried copying the bootwifi directory to my main, but got the same error.
(I could of course copy all the .cpp and .h files needed into there, but that would mean
I couldn't easily work with a fork of esp32-snippets and contribute any additions I make.)

I copied the component.mk from the bootwifi folder -- in fact I copied the whole bootwifi structure. The compiler clearly finds the code for the component, but it perhaps it then includes a WiFi.h that is incompatible with bootwifi?

My components structure is a single "components" subdirectory containint the Arduino layer IDF tree and a components.mk that contains:

```
# "main" pseudo-component makefile

# add all subdirectories as source and include directories
COMPONENT_SRCDIRS := . $(notdir $(shell find -L $(COMPONENT_PATH) -maxdepth 1 -mindepth 1 -type d))
COMPONENT_ADD_INCLUDEDIRS := $(COMPONENT_SRCDIRS)
```

Cheers, H

chegewara, 2018-05-31:
This is project with BootWiFi:
https://github.com/nkolban/ESP32_Explorer