nkolban/esp32-snippets #544: Is the latest ESP32 BLE library usable at the moment with Arduino IDE?

JabberwockPL, 2018-06-03:
I have come back to create another project with ESP32 (the last one, the cycling interface, is still working great, thanks to everyone involved!). However, it seems much more problematic than before.

In particular, I am trying to do a simple BLE HID keyboard. I have found a sample sketch, but first it complained about esp_bt.h, which I have fixed with a fake esp_bt.h referencing bt.h. But now it throws an error related to 'Enable C++ Exceptions'. I have read the related issues here, but I am still confused about the actual state of things. I have tried to make the recommended changes in sdkconfig.h, but it did not work. So I still am not sure if it is usable at the moment or not?

chegewara, 2018-06-03:
If i remember it has been fixed, you just need to update arduino-esp32 stack. Its tricky but its been described few times here and on forum if i remember.

JabberwockPL, 2018-06-03:
OK, I will dig deeper. Thank you for the lightning speed response!

JabberwockPL, 2018-06-03:
I have tried to update the stack according to the instructions, but for some reason it did not work. I had to delete everything and clone it again for it to work.

Now I cannot get HID to work, but that is another story :)

closed 2018-06-03 by JabberwockPL