nkolban/esp32-snippets #545: BLE UART example and IOS disconnecting

PhilColbert, 2018-06-04:
Hi,

Has anyone got this to stay stable, mine runs sporadically and randomly disconnects?

(W (25699) BT_APPL: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0008)

Which i guess is the iphone disconnecting ?

Log

V (25524) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe71cc), owner: <N/A>
D (25542) GeneralUtils:      00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f  ----------------
D (25547) GeneralUtils: 0000 7c                                               |
D (25554) FreeRTOS: Semaphore taking: name: ConfEvt (0x3ffe6c30), owner: <N/A> for notify
D (25562) FreeRTOS: Semaphore taken:  name: ConfEvt (0x3ffe6c30), owner: notify
V (25569) FreeRTOS: >> wait: Semaphore waiting: name: ConfEvt (0x3ffe6c30), owner: notify for notify
V (25570) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe6c30), owner: notify
V (25585) FreeRTOS: << wait: Semaphore released: name: ConfEvt (0x3ffe6c30), owner: notify
V (25585) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe71cc), owner: <N/A>
D (25603) GeneralUtils:      00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f  ----------------
D (25608) GeneralUtils: 0000 7d                                               }
D (25615) FreeRTOS: Semaphore taking: name: ConfEvt (0x3ffe6c30), owner: <N/A> for notify
D (25623) FreeRTOS: Semaphore taken:  name: ConfEvt (0x3ffe6c30), owner: notify
V (25630) FreeRTOS: >> wait: Semaphore waiting: name: ConfEvt (0x3ffe6c30), owner: notify for notify
V (25632) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe6c30), owner: notify
V (25646) FreeRTOS: << wait: Semaphore released: name: ConfEvt (0x3ffe6c30), owner: <N/A>
V (25646) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe71cc), owner: <N/A>
D (25664) GeneralUtils:      00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f  ----------------
D (25669) GeneralUtils: 0000 7e                                               ~
D (25676) FreeRTOS: Semaphore taking: name: ConfEvt (0x3ffe6c30), owner: <N/A> for notify
D (25684) FreeRTOS: Semaphore taken:  name: ConfEvt (0x3ffe6c30), owner: notify
V (25691) FreeRTOS: >> wait: Semaphore waiting: name: ConfEvt (0x3ffe6c30), owner: notify for notify
W (25699) BT_APPL: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0008
V (25699) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe6c30), owner: notify
V (25715) FreeRTOS: << wait: Semaphore released: name: ConfEvt (0x3ffe6c30), owner: <N/A>
V (25715) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe71cc), owner: <N/A>
V (25730) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe6c30), owner: <N/A>
D (25733) GeneralUtils:      00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f  ----------------
V (25737) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe71cc), owner: <N/A>
D (25745) GeneralUtils: 0000 7f                                               .
start advertising

PhilColbert, 2018-06-08:
No one ? :(

Surely someone must be using BLE with the Iphone ! :)

I have tried pcbreflex example aswell, same problem.

Tried increasing and decreasing  the Min and Max interval times, still same issue.

Anyone ? :)

chegewara, 2018-06-08:
Im pretty sure all of them are using iPhone brick:
https://github.com/nkolban/esp32-snippets/issues/430

offtop: sorry but i am iPhone hater

PhilColbert, 2018-06-08:
tried alsorts of values in

 esp_ble_gap_set_prefer_conn_params();

Still disconnecting with the same problem .

think this one has beaten me :(

chegewara, 2018-06-08:
You can try to use nRF connect on android to get those values from logs.

This is my advice about iPhone: flash it in the toilet. But seriously, try to add security layer to your application. Its a chance that iPhone is security sensitive or try to contact with apple devs community.

PhilColbert, 2018-06-08:
Thanks
I wish I could do that regarding the Iphone, but my project is a full flying computer and people use kobos, android phones and unfortunately iphones and theres a lot with iphones wanting to use the system.

So I am stuck.

Any examples on adding a security layer ? Have spent a few days hacking at this and getting nowhere :(

chegewara, 2018-06-08:
https://github.com/nkolban/esp32-snippets/tree/master/cpp_utils/tests/BLETests/Arduino/security

PhilColbert, 2018-06-11:
Well, the hm10 doesnt require a security layer, I am trying to emulate this but can try !

Thanks