nkolban/esp32-snippets #546: MTU problem

robson31, 2018-06-04:
I have a HUZZAH32 dev board with MPU6050 connected to it. I wrote a simple example to have one of the data variables send over BLE to the client. When a client is a phone (Android) there is absolutely no problem with the code working. However, when I try to connect my setup to the PC (Windows 10), it fails. 
Here is the Arduino code: https://pastebin.com/JbwsvkcV
Don't mind line 111 (setMTU function), I tried to experiment with different values with no luck.
Here is the Verbatim from the console: https://pastebin.com/QZdWqm6v

My guess here is that the MTU is too large, following issue @290.

As for client, I use BluetoothLE app, and Bluetooth LE Lab app from the Microsoft store. I also tried different Visual Studio examples, with no luck.

Any help would be much appreciated. Thanks

wegunterjr, 2018-06-04:
Is this helpful:
https://github.com/nkolban/esp32-snippets/issues/325

BLEDevice::setMTU(value);

This piece of code needs to be added right after BLEDevice::init(). Thats
all and that should works.

Value can be any unsigned integer between 23 and 517, this is bluetooth 4.2
specification limitation.

robson31, 2018-06-04:
Hi there, this is what I've tried on line 111 in my code, with different values, but absolutely no luck. Thanks for you comment!

chegewara, 2018-06-04:
Its seems that something is not right, because from logs we can see that MTU is set to 517:
```
[D][BLEUtils.cpp:1766] dumpGattServerEvent(): [conn_id: 0, mtu: 517]
[I][BLEDevice.cpp:112] gattServerEventHandler(): ESP_GATTS_MTU_EVT, MTU 517
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_MTU_EVT
```

One thing i can advise is to play with this line:
https://github.com/nkolban/esp32-snippets/blob/eb25dac7c1408847ceca22f8f3931f27fd5af720/cpp_utils/BLECharacteristic.cpp#L365-L366

When i have write this code i was not sure what is the max MTU value. Try to change 512 to 517.

robson31, 2018-06-04:
This is actually the first thing I've tried, first 517, then some other values, but unfortunately no luck at all. Thanks for your comment!

chegewara, 2018-06-04:
Like i said, without logs when it crash i cant help. For me all my codes are working with windows 10 with no issues.

robson31, 2018-06-05:
Pardon my ignorance, but you mean the logs when the app crashes?

chegewara, 2018-06-05:
About those windows its better to ask somewhere else, i am not using it so i dont know it.

Edit2) i think its normal, bluedroid is not getting disconnect message from client app when it crash and for some time does not know that client app is no longer connected; also i dont know how you are handling connection and sending message

GrooverFromHolland, 2018-06-06:
The BluetoothLe Universal windows samples just don't always work, because they all use 
the deviceInformationDeviceWatcher class to get the device information.
This information is often stale and not up to date.

The deviceWacher is on system level; meaning You have to add the device first in Windows Settings before it can be found.
The best way to deal with this is to modify the the sample and use the advertisementwatcher. 

I made an example on Github explaining the advertisementwatcher and subscribing for notifications:
https://github.com/GrooverFromHolland/SimpleBleExample_by_Devicename
The example is written for visual studio 2017, but you can read the most important methods and getting an idea by opening the MainPage.xaml.cs on Github.
If you download the complete solution, set your IDE solution platform to X86 and local machine to run.

robson31, 2018-06-06:
Thanks, I appreciate you time to write this comment. I'm still on an early stage of understanding how everything works. I've downloaded the complete solution from your repo and ran the code. It crashes on line 87 inside MainPage.xaml.cs:
System.ArgumentOutOfRangeException: 'Specified argument was out of the range of valid values.'

Thanks again for the sample, it's indeed very well commented and very helpful.

closed 2018-06-07 by robson31