nkolban/esp32-snippets #548: Advertised device getServiceData() in loop without rescan

u4mzu4, 2018-06-05:
Hello!
First I would say big thanks for this BLE solution. I'm using Xiaomi Mijia thermo-hygrometer sensor via advertised service (so NOT connected), and my goal would be to read out the advertised data in every 2 sec. Is there any proper solution without BLEScan? Since I already know the BLEAdvertisedDevice address, name, ServiceUUID and really don't want to rescan to find them again. I just want to enjoy the getServiceData in a loop.
I hope my needs are clear, but if you have any question, do not hesitate to contact me.

chegewara, 2018-06-05:
I understand what you mean, but your peripheral can disappear (power loss or moving), so you have to scan to see if that device is still present and advertising itself. You can try to connect to that device without BLEScan but thats all.

u4mzu4, 2018-06-05:
OK, thank you for your suggestions! :-)

closed 2018-06-05 by u4mzu4