nkolban/esp32-snippets #550: BLE Client example unable to connect to Microbit UART BLE

65innovations, 2018-06-05:
Hi,
I used BLE client example to connect ESP32 and Microbit via UART BLE. They unable to establish connection with the following error message. I am able to connect to Microbit BLE UART with iOS nrfconnect app without any pairing.

Starting Arduino BLE Client application...
Forming a connection to c4:91:24:5b:09:ab
 - Created client
E (43111) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0100
E (43111) BT: p_conn is NULL in bta_gattc_conn_cback

E (43112) BT: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x0100
E (43119) BT: p_conn is NULL in bta_gattc_conn_cback
 - Connected to server

chegewara, 2018-06-06:
Its not much of logs. Full logs would be better, just be sure you have set log level debug.

65innovations, 2018-06-06:
Hi Chegewara,

Thanks. More logs below:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11572
entry 0x40078a5c
Starting Arduino BLE Client application...
Forming a connection to c4:91:24:5b:09:ab
[D][BLEDevice.cpp:53] createClient(): >> createClient
[D][BLEDevice.cpp:59] createClient(): << createClient
 - Created client
[D][BLEClient.cpp:90] connect(): >> connect(c4:91:24:5b:09:ab)
[D][BLEClient.cpp:74] clearServices(): >> clearServices
[D][BLEClient.cpp:80] clearServices(): << clearServices
[D][BLEDevice.cpp:142] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_REG_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_REG_EVT
[D][BLEUtils.cpp:1550] dumpGattClientEvent(): [status: ESP_GATT_OK, app_id: 0x0]
E (43208) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0100
E (43209) BT: p_conn is NULL in bta_gattc_conn_cback

E (43210) BT: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x0100
E (43217) BT: p_conn is NULL in bta_gattc_conn_cback

[D][BLEDevice.cpp:142] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_DISCONNECT_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_DISCONNECT_EVT
[D][BLEUtils.cpp:1423] dumpGattClientEvent(): [reason: ESP_GATT_CONN_CONN_CANCEL, conn_id: 0, remote_bda: c4:91:24:5b:09:ab]
[D][BLEDevice.cpp:142] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_OPEN_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_OPEN_EVT
[D][BLEUtils.cpp:1503] dumpGattClientEvent(): [status: ESP_GATT_ERROR, conn_id: 0, remote_bda: c4:91:24:5b:09:ab, mtu: 0]
[D][BLEClient.cpp:122] connect(): << connect(), rc=0
 - Connected to server
[D][BLEClient.cpp:319] getService(): >> getService: uuid: 6e400001-b5a3-f393-e0a9-e50e24dcca9e
[D][BLEClient.cpp:355] getServices(): >> getServices
[D][BLEClient.cpp:74] clearServices(): >> clearServices
[D][BLEClient.cpp:80] clearServices(): << clearServices

chegewara, 2018-06-06:
You are giving just a few pieces of logs. It still not telling whole story.

65innovations, 2018-06-06:
I modified the BLE Client example in the setup() to:
void setup() {
  Serial.begin(115200);
  Serial.println("Starting Arduino BLE Client application...");
  BLEDevice::init("");

  // Retrieve a Scanner and set the callback we want to use to be informed when we
  // have detected a new device.  Specify that we want active scanning and start the
  // scan to run for 30 seconds.
//  BLEScan* pBLEScan = BLEDevice::getScan();
//  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
//  pBLEScan->setActiveScan(true);
// pBLEScan->start(30);
  pServerAddress = new BLEAddress("c4:91:24:5b:09:ab");

  doConnect = true;
} // End of setup.
and theUUID:

// The remote service we wish to connect to.
static BLEUUID serviceUUID("6E400001-b5a3-f393-e0a9-e50e24dcca9e");

// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID("6E400003-b5a3-f393-e0a9-e50e24dcca9e");

Besides that, I disable the pairing and security on microbit. 

I am noob to ble. I am learning.

65innovations, 2018-06-06:
I am trying to learn how to enable ble security and pairing on esp32 ble android library. Any guide that I could refer to learn more?