nkolban/esp32-snippets #552: How to get client's address on BLE Server

parhamdb, 2018-06-06:
Hi,
I'm trying to understand how to get a client's mac address when it is connected to the server.
Remote Address is available on esp32-idf on connect event but in arduino library that is not available and can't figure out how to find it.
Thanks

chegewara, 2018-06-07:
If i remember its not implemented.

chegewara, 2018-07-10:
@parhamdb Its added in master. You can get client address in onConnect() callback, just check source code, because for backward compatibility i there is two onConnect() callbacks in BLEServer.

wincy122, 2022-07-07:
I'm trying to understand how to get a client's mac address when it is connected to the server.
Remote Address is available on esp32-idf on connect event but in arduino library that is not available and can't figure out how to find it.
Thanks

wincy122, 2022-07-07:
hi 
iam working on esp32 ble where i connect to mobile as client and esp32 as server where the communication is going good but my problem is when they both are connected esp32 serial monitor should display the address of mobile phone to which it is connected  please can any one help me or share  any reference
Thanks