nkolban/esp32-snippets #555: [request]mac address of connected device

chegewara, 2018-06-08:
There was few questions about retrieving mac address of connected device in BLEServer mode. I have idea how to do it in very simple manner but it may breaks older applications. Changing this callback can help:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEServer.h#L104

Unless its possible to have two callbacks with different signature?
`virtual void onConnect(BLEServer* pServer, gatts_connect_evt_param params);`

This way we can get conn_id and remote_bda whenever its required. Like in this function:
`esp_ble_gap_update_conn_params`

What do you think?

panargirakis, 2018-06-29:
Did you manage to get esp_ble_gap_update_conn_params to work? I changed the definition of onConnect and tried to update the parameters but I could not get the GAP event to get triggered properly.

chegewara, 2018-06-29:
There was not much interest of this so i didnt work on it, i have my own projects i am working on now, but if you need i can try to help. There is not much of code here to make it just work. Try to change this line:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEServer.cpp#L204
with this code:
`m_pServerCallbacks->onConnect(this, param);`

This line according to my previous post:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEServer.h#L104

and you should be able to use `esp_ble_gap_update_conn_params` in onConnect() callback, just you need to extract bda from param:
https://esp-idf.readthedocs.io/en/latest/api-reference/bluetooth/esp_gatts.html#_CPPv2N24esp_ble_gatts_cb_param_t23gatts_connect_evt_paramE

panargirakis, 2018-07-01:
Thank you for the help. Here are my findings:

I tried the following code as per your suggestion:
```
class MyServerCallbacks : public BLEServerCallbacks
{
    void onConnect(BLEServer *pServer, esp_ble_gatts_cb_param_t *param)
    {
        esp_ble_conn_update_params_t conn_params = {0};
        memcpy(conn_params.bda, param->connect.remote_bda, sizeof(esp_bd_addr_t));
        conn_params.latency = 0;
        conn_params.max_int = 0x0c; // max_int = 0x0c*1.25ms = 15ms
        conn_params.min_int = 0x0c; // min_int = 0x0c*1.25ms = 15ms
        conn_params.timeout = 200;  // timeout = 200*10ms = 2000ms
        //start sent the update connection parameters to the peer device.
        esp_ble_gap_update_conn_params(&conn_params);

        // other
        deviceConnected = true;
    };
etc...
```

However, the update doesn't seem to come into effect. It doesn't feel like the connection latency has changed at all which it should. Also, I see the following message even if I comment out the `esp_ble_gap_update_conn_params`  function call:

` BT: Call back not found for application conn_id=3`

Looks like the lower level stack is pointing to an error. Could it be an issue with the GAP handler?

panargirakis, 2018-07-01:
I do not think such a request is received. Here is the log for a connect-disconnect sequence:

```
20:37:30.368 Connecting BLE Test
20:37:30.369 [MethodCall]connectPeripheral
20:37:30.903 [Callback]Successful connect to peripheral BLE Test
20:37:30.903 Status changed: Connected to BLE Test
20:37:31.097 [Callback]Successful exchange MTU 514 bytes if send without response
20:37:31.097 [Callback]Successful exchange MTU 512 bytes if send with response
20:37:31.098 [Callback]Successful discover service 80865467-9C99-4CCE-A94E-48058D175FED
20:37:31.157 [Callback]Successful discover charact. for service 88E28319-E781-4B31-B34F-5B9C65E4DD74
20:37:31.158 [MethodCall]readCharacteristicValue 88E28319-E781-4B31-B34F-5B9C65E4DD74
20:37:31.158 [MethodCall]discoverDescriptorsForCharact. 88E28319-E781-4B31-B34F-5B9C65E4DD74
20:37:31.158 [Callback]Successful discover charact. for service 011C9658-F282-4DDC-97A9-A1B1FB6C52B9
20:37:31.158 [MethodCall]readCharacteristicValue 011C9658-F282-4DDC-97A9-A1B1FB6C52B9
20:37:31.159 [MethodCall]discoverDescriptorsForCharact. 011C9658-F282-4DDC-97A9-A1B1FB6C52B9
20:37:31.187 [Callback]Successful update value for charact. 88E28319-E781-4B31-B34F-5B9C65E4DD74
20:37:31.217 [Callback]Successful discover descriptors for charact. 88E28319-E781-4B31-B34F-5B9C65E4DD74
20:37:31.262 [Callback]Successful update value for charact. 011C9658-F282-4DDC-97A9-A1B1FB6C52B9
20:37:31.337 [Callback]Successful update value for descriptor 2902
20:48:37.931 [MethodCall]cancelPeripheralConnection
```

chegewara, 2018-07-01:
I mean could you see nRF logs?

panargirakis, 2018-07-01:
The log above is from nRF Connect for mobile. I am not quite sure what you are requesting. Could you elaborate?

chegewara, 2018-07-01:
Sorry, i thought its from arduino. Maybe could you fork this repo and push your changes so i could pull it from your repo with changes and test it?

panargirakis, 2018-07-01:
Yes, I will send it soon

panargirakis, 2018-07-01:
[fork with onConnect changes](https://github.com/panargirakis/esp32-snippets.git)

[my project](https://github.com/panargirakis/esp32-snippets/tree/master/pargirakis_projects/BLE%20Server%20Arduino)

chegewara, 2018-07-01:
It seems to work for me:
[image]

chegewara, 2018-07-01:
What i did is to change BLEServer.cpp:
```
m_pServerCallbacks->onConnect(this);
m_pServerCallbacks->onConnect(this, param);
```
and BLEServer.h:
```
virtual void onConnect(BLEServer* pServer);
virtual void onConnect(BLEServer* pServer, esp_ble_gatts_cb_param_t *param);
```
and no changes to your main.cpp (.ino) file.

panargirakis, 2018-07-01:
Interesting. I cannot replicate this. I am using platformio, so maybe my idf is outdated and something is failing. I believe I have made the exact same changes as you in the fork I posted.

chegewara, 2018-07-01:
It is possible. If you can try with normal arduino-ide would be good.
BTW thanks for test code

panargirakis, 2018-07-10:
Update: I checked again from an Android device and the update of the connection parameters does indeed work. From an iPhone thought it does not get logged. Not sure if the iPhone rejects the request all together or just doesn't log it.

chegewara, 2018-07-10:
@panargirakis Thanks for update and help with tests.

panargirakis, 2018-07-10:
@chegewara Thanks for all the support!