nkolban/esp32-snippets #557: client  passing data to server bug

techno94, 2018-06-10:
Hello i treid to send data to each other client to server

I arrived to send data from server to client
The server arrived to receive data from an app made in app inventor
## **but i don't arrive to received data from the client in the server sketch**

so i think that the bug come from the client maybe here : 
pRemoteCharacteristic->writeValue(txStringArduino.c_str() ,txStringArduino.length() );

here the code of the server : 
https://pastebin.com/bVtAUSxJ

here the code of the client : 
https://pastebin.com/Sveg8Lgc

if someone could help me it will be really  nice
Thank you

chegewara, 2018-06-10:
https://github.com/nkolban/esp32-snippets/issues/50#issuecomment-384597137

techno94, 2018-06-14:
Hello chegewara ,

thanks for the tip
i arrived to notify by replacing : 
pRemoteCharacteristic->writeValue

by this : 
 BLERemoteDescriptor *desc = pRemoteCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2902));
         uint8_t val[] = {0x01,0x01};
        desc->writeValue(val, 2);

but in the serial.print of the client i recieved this : 
Notify callback for characteristic 6e400003-b5a3-f393-e0a9-e50e24dcca9e

but this is the CHARACTERISTIC_UUID_RX

and i want to notify the CHARACTERISTIC_UUID_TX

since 4 days, i try to implemente a : 
static BLERemoteCharacteristic*  writeChar;

with a : 
writeChar=pRemoteService->getCharacteristic(CHARACTERISTIC_UUID_TX);

but when i try to : 
BLERemoteDescriptor *desc = writeChar->getDescriptor(BLEUUID((uint16_t)0x2902)); 
 uint8_t val[] = {0x01,0x01};
        desc->writeValue(val, 2);
the client sketch still went wrong. 
i try to add writeChar->registerForNotify(notifyCallback); // line 44
but still wrong

here is what i've done for the client : 
https://pastebin.com/AxMv1FcH

Thanks a lot for you re help
did you have a code for a server and a client on an esp32 each one receive and send data ?

chegewara, 2018-06-15:
```
BLERemoteDescriptor *desc = writeChar->getDescriptor(BLEUUID((uint16_t)0x2902));
uint8_t val[] = {0x01,0x01};  <-------{0x01, 0x00}
desc->writeValue(val, 2)
```
And move those 3 lines to line 52

techno94, 2018-06-15:
hello,
i've put those 3 lines to line 52 : 
 BLERemoteDescriptor *desc = writeChar->getDescriptor(BLEUUID((uint16_t)0x2902));
uint8_t val[] = {0x01,0x00};  
desc->writeValue(val, 2);

but it went wrong
i think i've a bug somewhere  in writeChar

here is the log : 
https://pastebin.com/1KtGBy27

thanks

techno94, 2018-06-15:
Hello don't you have a sketch of a client and a server in ble who send and receive data to each other.
it will really help me 
thanks