nkolban/esp32-snippets #558: ios unable to find ble device

einiemand, 2018-06-12:
Hello,

my ipad and iphone7 can't find the ble device running ble_uart demo, but some apps like LightBlue can find the device and successfully connect to it.

chegewara, 2018-06-12:
If you have android smartphone with android 6 or better then you can try this:
- scan for esp32,
- use clone option on scanner tab,
- on advertiser tab try different time settings etc
- try to find your fake uart device(from android smartphone).

einiemand, 2018-06-12:
Thanks for your reply, another question: where are all the logs stored? I want to check some details.

chegewara, 2018-06-12:
Which logs?
Esp32 logs you can see when you connect with putty for example, and nRF connect logs you can find on connected tab when you swipe from left to right.