nkolban/esp32-snippets #560: BLE SampleClient can't get services from Server

vuanh-minh, 2018-06-12:
Hi,
After updated esp-idf, and esp32-snippets, i got some problem with SampleClient after it connects to the Server. The connexion is successful, but for some reasons, the client can't get the service, and wait the semaphore : SearchCmplEvt. How can i resolve this problem ?

```
D (3657) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT
D (3664) BLEUtils: [status: 0]
D (3667) Task: >> runTask: taskName=Task
D (3671) BLEDevice: >> createClient
D (3675) BLEDevice: << createClient
D (3678) BLEClient: >> connect(30:ae:a4:80:87:16)
D (3683) FreeRTOS: Semaphore taking: name: RegEvt (0x3ffd9ca4), owner: <N/A> for connect
D (3691) FreeRTOS: Semaphore taken:  name: RegEvt (0x3ffd9ca4), owner: connect
D (3698) BLEClient: >> clearServices
D (3702) BLEClient: << clearServices
D (3706) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_REG_EVT
D (3713) BLEUtils: GATT Event: ESP_GATTC_REG_EVT
D (3718) BLEUtils: [status: ESP_GATT_OK, app_id: 0x0]
V (3723) FreeRTOS: Semaphore giving: name: RegEvt (0x3ffd9ca4), owner: connect
V (3730) FreeRTOS: >> wait: Semaphore waiting: name: RegEvt (0x3ffd9ca4), owner: <N/A> for connect
V (3739) FreeRTOS: << wait: Semaphore released: name: RegEvt (0x3ffd9ca4), owner: connect
D (3748) FreeRTOS: Semaphore taking: name: OpenEvt (0x3ffdac40), owner: <N/A> for connect
D (3756) FreeRTOS: Semaphore taken:  name: OpenEvt (0x3ffdac40), owner: connect
V (3766) FreeRTOS: >> wait: Semaphore waiting: name: OpenEvt (0x3ffdac40), owner: connect for connect
V (3772) FreeRTOS: << wait: Semaphore released: name: ScanEnd (0x3ffd2bac), owner: start
D (3781) BLEScan: << start()
I (3784) SampleClient: Scanning finished
D (3903) nvs: nvs_open_from_partition gatt_a8316ee7 1
D (3903) nvs: nvs_get_str_or_blob gattc_cache_key
D (3904) nvs: nvs_get_str_or_blob gattc_cache_key
D (3907) nvs: nvs_get_str_or_blob gattc_cache_key
D (3912) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_CONNECT_EVT
D (3919) BLEUtils: GATT Event: ESP_GATTC_CONNECT_EVT
D (3924) BLEUtils: [conn_id: 0, remote_bda: 30:ae:a4:80:87:16]
D (3931) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_OPEN_EVT
D (3938) BLEUtils: GATT Event: ESP_GATTC_OPEN_EVT
D (3943) BLEUtils: [status: ESP_GATT_OK, conn_id: 0, remote_bda: 30:ae:a4:80:87:16, mtu: 23]
V (3951) FreeRTOS: Semaphore giving: name: OpenEvt (0x3ffdac40), owner: connect
V (3959) FreeRTOS: << wait: Semaphore released: name: OpenEvt (0x3ffdac40), owner: connect
D (3967) BLEClient: << connect(), rc=1
D (3971) BLEClient: >> getService: uuid: 91bad492-b950-4226-aa2b-4ede9fa42f59
D (3978) BLEClient: >> getServices
D (3981) BLEClient: >> clearServices
D (3985) BLEClient: << clearServices
D (3989) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_SEARCH_RES_EVT
D (3997) BLEUtils: GATT Event: ESP_GATTC_SEARCH_RES_EVT
D (4002) BLEUtils: [conn_id: 0, start_handle: 1 0x01, end_handle: 5 0x05, srvc_id: uuid: 00001801-0000-1000-8000-00805f9b34fb, inst_id: 1
D (4015) BLERemoteService: >> BLERemoteService()
D (4019) BLERemoteService: << BLERemoteService()
D (4024) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_SEARCH_RES_EVT
D (4032) BLEUtils: GATT Event: ESP_GATTC_SEARCH_RES_EVT
D (4037) BLEUtils: [conn_id: 0, start_handle: 20 0x14, end_handle: 28 0x1c, srvc_id: uuid: 00001800-0000-1000-8000-00805f9b34fb, inst_id: 20
D (4050) BLERemoteService: >> BLERemoteService()
D (4055) BLERemoteService: << BLERemoteService()
D (4060) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_SEARCH_RES_EVT
D (4068) BLEUtils: GATT Event: ESP_GATTC_SEARCH_RES_EVT
D (4073) BLEUtils: [conn_id: 0, start_handle: 40 0x28, end_handle: 65535 0xffff, srvc_id: uuid: eb65b6c6-fec3-4ed1-a6fc-9eff755a416c, inst_id: 40
D (4086) BLERemoteService: >> BLERemoteService()
D (4091) BLERemoteService: << BLERemoteService()
D (4096) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_SEARCH_CMPL_EVT
D (4104) BLEUtils: GATT Event: ESP_GATTC_SEARCH_CMPL_EVT
D (4109) BLEUtils: [status: ESP_GATT_OK, conn_id: 0]
V (4114) FreeRTOS: Semaphore giving: name: SearchCmplEvt (0x3ffd9d64), owner: <N/A>
D (4122) FreeRTOS: Semaphore taking: name: SearchCmplEvt (0x3ffd9d64), owner: <N/A> for getServices
D (4131) FreeRTOS: Semaphore taken:  name: SearchCmplEvt (0x3ffd9d64), owner: getServices
V (4139) FreeRTOS: >> wait: Semaphore waiting: name: SearchCmplEvt (0x3ffd9d64), owner: getServices for getServices
W (607456) BT_APPL: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0008
D (607458) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_DISCONNECT_EVT
D (607462) BLEUtils: GATT Event: ESP_GATTC_DISCONNECT_EVT
D (607468) BLEUtils: [reason: ESP_GATT_CONN_TIMEOUT, conn_id: 0, remote_bda: 30:ae:a4:80:87:16]
D (607477) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_CLOSE_EVT
D (607484) BLEUtils: GATT Event: ESP_GATTC_CLOSE_EVT
D (607489) BLEUtils: [status: ESP_GATT_OK, reason:ESP_GATT_CONN_TIMEOUT, conn_id: 0]

```

closed 2018-06-26 by vuanh-minh

ufanders, 2020-05-11:
@vuanh-minh How did you solve this? I am experiencing the same problem.

zaindroid, 2020-06-02:
Hi,  I m having this same issue. Can you please guide how did you resolve it. I m not able to access the server services . it gets stuck at:

 BLERemoteService* pRemoteService = pclient->getService(serviceUUID);

On monitor I m stuck at:

[V][FreeRTOS.cpp:69] wait(): >> wait: Semaphore waiting: name: SearchCmplEvt (0x3ffdf974), owner: getServices for getServices

lightingghost, 2020-08-30:
I am having the same issue -- when waked up from deep sleep. Does anyone have any experience fixing it?