nkolban/esp32-snippets #561: Byte sending issue

robson31, 2018-06-13:
Hi all,

I'm trying to serialize the data I'll be sending over BLE. Let me explain the problem as simple as I am able to. Let's say, we have the following situation:

`uint8_t a = 1;`
`uint8_t b = 255;`  
`uint16_t ab = 0;`
`ab = ab << 8 | a;` 
`ab = ab << 8 | b;`

Above example works fine. However, when `a=0`, then there is an issue that the first byte is all 0's, and libraries treat it an a NULL character. Logs, to prove (some other example, but same point):

`sizeof(temp_test): 4`
`Starting to print temp_test as HEX from buffer: 	0FFFFFF`
`Starting to print temp_test one byte as HEX at a time from buffer: 	0FFFFFF`
`buffer_test size: 4`
`[D][BLECharacteristic.cpp:664] setValue(): >> setValue: length=0, data=, characteristic UUID=6e400003-b5a3-f393-e0a9-e50e24dcca9e`
`[D][BLECharacteristic.cpp:671] setValue(): << setValue`
`[D][BLECharacteristic.cpp:524] notify(): >> notify: length: 0`

I've looked over in the BLECharacteristic files, but I'm not sure how to solve this issue.

Ultimately, the goal is to pack some of the variables from two sensors, attach a packet_ID variable, and send it over BLE. Thanks for any help.

chegewara, 2018-06-13:
Did you try with this overloaded function? It works for me.
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLECharacteristic.h#L76

robson31, 2018-06-14:
Thanks, but this does not answer my question.

robson31, 2018-06-14:
Check my edit, please. Also, how do you use the function that you pointed out if the ab is uint16_t? I'm using Arduino IDE.

chegewara, 2018-06-14:
Pseudocode:
uint16 = 2 x uint8;
val[0] = uint16;
val[1] = uint16 >> 8; <--- right shift (google it)

now you have `uint8_t* val` or `uint8_t val[2]`

robson31, 2018-06-14:
Oh man, saving the day again. Dzięki!

robson31, 2018-06-14:
By the way, in function buildHexData, there is a guard against too much data (now it's set to 100). Isn't this depended on the MTU size? I'm using 517, therefore the max length for me would be 258. Just a note if somebody ever finds this issue.

closed 2018-06-14 by robson31

chegewara, 2018-06-14:
@robson31 If you could point me to that line in code it would be nice.

robson31, 2018-06-14:
@chegewara sure, here:
https://github.com/nkolban/esp32-snippets/blob/937a81c18d742e86305df5184ccfb70f35b77c85/cpp_utils/BLEUtils.cpp#L789