nkolban/esp32-snippets #562: BLECharacteristic get raw data

johngull, 2018-06-14:
Currently, in BLECharacteristic you can get value only as std::string - ```std::string    getValue();```
But what if I need raw data (can contain zeros in my case).
Is there is a workaround?
If not, will you accept pull request with the new function for this? And how fast it will appear in 'ESP32 BLE Arduino'?

Thanks

chegewara, 2018-06-14:
@nkolban

u4mzu4, 2018-06-14:
I was fighting with a similar challenge, but a simple casting to byte, short or uint solved my problem.

johngull, 2018-06-14:
@u4mzu4 that is not a solution for my case.
I can receive e.g. 0x11110011. When it will be transformed into std::string the last byte 0x11 will be lost.

johngull, 2018-06-14:
I made additional tests. And seems I was wrong std::string::length() not cut length by zero byte, so existed function is ok.
Sorry for bothering.

closed 2018-06-14 by johngull