nkolban/esp32-snippets #563: BLE_client receiving data

haarislabs, 2018-06-14:
I have an Esp-Wroom-32 and nrf52832. I'm trying to establish a ble connection between using the esp-32 as client/central and nrf52832 as server/peripheral. In BLE_client example I'm just changing the UUID of service and characteristics. `static BLEUUID serviceUUID("f3641400-00b0-4240-ba50-05ca45bf8abc");
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID("f3641401-00b0-4240-ba50-05ca45bf8abc");` 
In nordic I'm using this example [https://github.com/bjornspockeli/custom_ble_service_example](url)
The problem is im not able to receive any data from nordic. It shows me `BT: bta_gattc_cmpl_cback unknown conn_id =  3, ignore data
` That error. For a better insight of what's going on [https://pastebin.com/TXZdH21T](url)
I'm new to this so any kind of guidance will be helpful

chegewara, 2018-06-14:
Arduino or esp-idf example?

haarislabs, 2018-06-14:
Arduino IDE

chegewara, 2018-06-14:
You code works, you are connected to nrf52832. This is where you are:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/BLE_client/BLE_client.ino#L125-L131
Now instead writing to characteristic try to read from it and print value to serial:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLERemoteCharacteristic.h#L43-L46

haarislabs, 2018-06-14:
I tried using this but it shows nothing on the serial monitor `  std::string value = pRemoteCharacteristic->readValue();
    Serial.print("The characteristic value was: ");
    Serial.println(value.c_str());`

chegewara, 2018-06-14:
Probably because value you read is not printable character:
`The characteristic value was: `
Try to read int.

haarislabs, 2018-06-14:
Thanks that worked but I'm still not getting into the "notifyCallback()" function. Why do I have to read in the main loop everytime ?

chegewara, 2018-06-14:
You dont have to if peripheral is sending notifications. There is workaround for that, search this repo issues for answer.

techno94, 2018-06-15:
hello : haarislabs 

here the code of the server :
https://pastebin.com/bVtAUSxJ

here the code of the client :
https://pastebin.com/Sveg8Lgc

cf : https://github.com/nkolban/esp32-snippets/issues/557

but i don' t arrived the client to send data to the server.
it send data on the CHARACTERISTIC_UUID_RX
but i want to send data to the CHARACTERISTIC_UUID_TX