nkolban/esp32-snippets #565: getManufacturerData() returning garbage (or more likely a garbage programmer)

chrisn-au, 2018-06-15:
Hi,

guys first up thanks for an amazing library. Have got up and running in hours - amazing stuff. As a point of warning I have not written any C++ for 30 years so the problem almost certainly is outside the code :-)

I am using the BLEAdvertisedDeviceCallbacks 

The ManufactureData is great in the advertisedDevice.toString however is garbage on the getManufacturerData() call

Thanks for the help

Code

void onResult(BLEAdvertisedDevice advertisedDevice) {
      StaticJsonBuffer<1000> jsonBuffer;
      char buffer[1000];
      Serial.printf("Advertised Device: %s \n", advertisedDevice.toString().c_str());
      if (advertisedDevice.haveManufacturerData()){
           Serial.printf("Manufacturer Data: %s \n", advertisedDevice.getManufacturerData().c_str());
      }

      if (advertisedDevice.haveServiceUUID()){
           Serial.printf("Service UUID: %s \n", advertisedDevice.getServiceUUID().toString().c_str());
           Serial.printf("Service Data: %s \n", advertisedDevice.getServiceData().c_str());
      }

Output

Advertised Device: Name: , Address: 88:c6:26:19:a9:0c, manufacturer data: 030000184800000570700d0380ea88c62619a90c12, serviceUUID: 0000fe61-0000-1000-8000-00805f9b34fb
Manufacturer Data: ␃
Service UUID: 0000fe61-0000-1000-8000-00805f9b34fb
Service Data:
Advertised Device: Name: , Address: 50:32:37:8d:f2:8f, manufacturer data: 4c0010050110f5eb95
Manufacturer Data: L
Advertised Device: Name: , Address: 32:66:07:db:5f:a8, manufacturer data: 060001092002685161f724c8f5ee777d5278b5f0664b0f07add7e6dc13
Manufacturer Data: ␆
Scan complete!

chegewara, 2018-06-15:
This is your manufacturer data. As you can see not all values are printable values:
`060001092002685161f724c8f5ee777d5278b5f0664b0f07add7e6dc13`
Now what you are trying to do is to convert it to string and print:
`advertisedDevice.getManufacturerData().c_str()'
You have to remember its 0 terminated string, so after printing 06 which is ␆ symbol printing is terminated.

chrisn-au, 2018-06-15:
Thanks for really quick reply. Understand -  Yes I was trying to print it (actually return it in a JSON MQTT msg vi WiFI). I naively assumed that the std::string getManufacturerData();  method would return a null terminated string. How would I do that ?

Thanks again

chegewara, 2018-06-15:
Easy, you have to send null terminated manufacturer data from peer device. If you send json data, you will receive json data.

chrisn-au, 2018-06-15:
Got it thanks

chrisn-au, 2018-06-15:
Ahh lights are coming on. I guess more of a ble question than the library question because what  I was looking at using the manufactures data for was a  unique identifier (I now understand that is not a valid thing to do as it may change and will change for the devices I am looking at ) . The question really is is there any mandatory data in an advertisement packet that is unique to the peripheral that will be unchanging - I think the answer is no and so welcome to ble

chegewara, 2018-06-15:
It all depends on device. Some device are using public mac, some are using private. 
Now, i think, its also possible to distinguish if it is the same device in case when you were bonded to it.
Some device has manufacturer data that is constant.

closed 2018-06-15 by chrisn-au

chrisn-au, 2018-06-17:
Just to close this - rather than doing processing on the ESP32 i wanted to send the manufactures data back via MQTT for processing.  So needed to convert the manufacturers data to a HEX string for  later processing.

so char* pHex
pHex = BLEUtils::buildHexData(nullptr, (uint8_t*)advertisedDevice.getManufacturerData().data(), advertisedDevice.getManufacturerData().length());

persibal, 2019-03-25:
I got this error:
invalid operands of types 'const char [20]' and 'char*' to binary 'operator+'

Can somebody tell me the below proper way to convert manufacturer data to HEX? 
char* pHex
pHex = BLEUtils::buildHexData(nullptr, (uint8_t*)advertisedDevice.getManufacturerData().data(), advertisedDevice.getManufacturerData().length()); 
???

I want to read the manufacturer data only.

Cyrilanthony777, 2020-03-07:
if anyone is still wondering how to get the manufacturer data only here is how.
```
uint8_t* MFRdata;
MFRdata = (uint8_t*)advertisedDevice.getManufacturerData().data();
int len = advertisedDevice.getManufacturerData().length();
```

1technophile, 2020-05-21:
Thanks, and if you want a char *;
```c++
char* manufacturerdata = BLEUtils::buildHexData(NULL, (uint8_t*)advertisedDevice.getManufacturerData().data(), advertisedDevice.getManufacturerData().length());
```

aliman22, 2023-02-10:
> Thanks, and if you want a char *;
> 
> ```c++
> char* manufacturerdata = BLEUtils::buildHexData(NULL, (uint8_t*)advertisedDevice.getManufacturerData().data(), advertisedDevice.getManufacturerData().length());
> ```
Thanks , add this after 
if (manufacturerdata != NULL) {
    free(manufacturerdata001);
}

aliman22, 2023-02-10:
> Thanks, and if you want a char *;
> 
> ```c++
> char* manufacturerdata = BLEUtils::buildHexData(NULL, (uint8_t*)advertisedDevice.getManufacturerData().data(), advertisedDevice.getManufacturerData().length());
> ```

Thanks, add this after
if (manufacturerdata != NULL) {
    free(manufacturerdata);
}