nkolban/esp32-snippets #566: Sending BLE data received from AT response via POST Command

Goutham8128, 2018-06-15:
Hi
I am using a GSM+BLE module which receives BLE data from a BLE sensor and sends data to server using GSM communication.
I am reading the incoming BLE data by sending AT commands to the device.I want to store the BLE AT response and Send the same data to server.
Now , in my case i am posting data via QHTTPPOST command.if I send AT it sending my input "AT" to the server. it should send "OK" to the server.If it executes i can pass my ble receive characteristic command such that it sends my sensor data.
Can u help me out in this?

chegewara, 2018-06-15:
Do you have any source materials about AT commands of this GSM? (pdf, website)
Or if you can, share your code to see logic.

Goutham8128, 2018-06-15:
This is the code which i am using

https://pastebin.com/N7RgvXLL

Goutham8128, 2018-06-15:
`
these are BLE AT commands

[Quectel_MC60_BLE_AT_Commands_Manual_V1.1.pdf](https://github.com/nkolban/esp32-snippets/files/2105752/Quectel_MC60_BLE_AT_Commands_Manual_V1.1.pdf)
`

Goutham8128, 2018-06-15:
these are GSM communication commands

                                           QHTTPPOST Commands

AT
OK

ATE1
OK

AT+CSQ
+CSQ: 26,0
OK

AT+CREG?
+CREG: 1,1
OK

AT+CGREG?
+CGREG: 0, 1
OK

AT+QIFGCNT=0
OK

AT+QICSGP=1,"airtelgprs.com"
OK

AT+QIREGAPP
OK

AT+QIACT
OK

AT+QHTTPURL=58,30
CONNECT
	//Input "http://api.efxnow.com/DEMOWebServices2.8/Service.asmx/Echo?"
OK

AT+QHTTPPOST=18,50,10
CONNECT
		//Input "MESSAGE=HELLOWORLD" and Uncheck Enter key while using QCOM
OK

AT+QHTTPREAD=30
CONNECT
<?xml version="1.0" encoding="utf-8"?>
<string xmlns="https://api.efxnow.com/webservices2.3">Message='HELLOWORLD' ASCII:72 69 76 76 79 87 79 82 76 68 </string>
OK

AT+QIDEACT
DEACT OK

Goutham8128, 2018-06-15:
The main loop starts from setting the url and sending post data and getting response from server
That data is succesfully sent from the server.(dynamic data i.e response collected from AT command)

Goutham8128, 2018-06-15:
sent to the server*

chegewara, 2018-06-19:
I will check it asap, been busy with very hard project. Sorry.

chegewara, 2018-06-19:
@Goutham8128 Can you show the line you are trying to send data to ble server?

chegewara, 2018-06-19:
Ok, i see what are you trying to achieve. I am now like baby in the dark, but i will try to help. 
Write to me on email: imperiaonline4@gmail.com and will try to do something with this and dont spam this repo.