nkolban/esp32-snippets #569: Just one to say thanks and a question about BLE in its own core

chrisn-au, 2018-06-19:
Guys,

Just wanted to say thanks for an extraordinary library.  Started from scratch last Friday and got my BLE/MQTT gateway up and running in a few days. This library was without fault and just worked. I will write up my  library this week and share as it had a few learnings.

A question: To get reliable MQTT sending (so I did not  loose messages, I ended up putting the BLE processing on core 0 while leave WiFi / MQTT on Core 1. I was not able to find any really clear guidance on when this is needed . Any ideas would be appreciated

Any guidance on what conditions/values to set Window/Interval would also be appreciated.

Thanks again

chegewara, 2018-06-19:
What i have learned in past few days is when you are working with wifi + ble is to very carefully control when wifi and when ble is using radio. Ive been trying to pool publish mqtt message every 1000 ms and at the same time connecting to ble server and it did not work, but when i moved publish procedure from loop to on demand with push button all works fine. Probably now i can move back and keep publishing mqtt topics even often than 1000 ms but not at the time when i am trying to accomplish ble connection.

When you know what can go wrong you have space to experiment.
My app is in arduino-ide so i cant move wifi or ble to core 1 (i think so) and my whole code works on core 0. If not counting huge issue with lack of heap then all seems to works fine.

chrisn-au, 2018-06-19:
Thanks for the response. I am only scanning (no connection) however have not noticed the conflict in the dual core mode .

You are able to use dual core in aurduino ide. The core Arduino code runs on core 1 and you have an option to run a task on core 0 (unless I don’t understand your point) . I will post my very poorly written code in a few hours

chegewara, 2018-06-19:
Probably you have better knowledge in arduino codding than i do. I just know that by default in esp-idf wifi and ble stack are tied to core 0, so i assumed that works the same way in arduino but i might be wrong.
Lucky for me my code works when i PinnedToCore 0 all tasks. I did one step farther in my app and right after initial setup i am destroying default task in arduino so my loop() is empty and is only to let my code compile. This way i have more heap recovered.

chrisn-au, 2018-06-20:
The old saying if it ain’t broke don’t touch it - I will try and experiment a bit further however given it is working at the sec I will move forward in adding functionality first

Thanks again - will close the thread

closed 2018-06-20 by chrisn-au