nkolban/esp32-snippets #571: advertisedDevice.getScan()->stop() caused heap corruption

u4mzu4, 2018-06-19:
Hello,

I'm currently working with ESP32_BLE_Arduino, I wrote a dummy advertisedDevice sample based on a BLE_client:
https://pastebin.com/V64c7qtW

If I want to speed up the running time (stop the scan before the timeout, uncomment lines 19/25), it caused me heap corruption error. If I'm waiting patiently until the timeout /pBLEScan->start(timeout)/ , the code just works without any problem. What could go wrong by the line
advertisedDevice.getScan()->stop()
?
Thanks in advance!

chegewara, 2018-06-19:
Try to change this part of code:
```
void loop() {
   BLEScan* pBLEScan = BLEDevice::getScan();
   MyAdvertisedDeviceCallbacks myCallbacks;
   pBLEScan->setAdvertisedDeviceCallbacks(&myCallbacks);
   pBLEScan->setActiveScan(true);
   pBLEScan->start(4);
}
```
Even if you want to use this approach, which can cause a lot issues because scan() is blocking function, move most of it to setup and leave just last line.

chegewara, 2018-06-19:
In your code you are creating hundreds instances of BLEScan, but one instance with scan(0) is enough to scan forever. Now what you need is separate task for scan or with code that will handle all scanned devices and those you are interested in particular.  
In this case you dont even need pScan->stop() command, because you dont want to connect to peripheral. You just need to decode advertising data.

closed 2018-07-18 by u4mzu4