nkolban/esp32-snippets #572: hardware address issue

kapil97, 2018-06-20:
I am using your BLE library for ESP32 in arduino ide.
I am able to print the address of all the devices but not of my iBeacon tag.
Please give some solution for the above issue asap.

My iBeacon is CC2541 based HM10 module.

Thanks and Regards
Kamleshwar

chrisn-au, 2018-06-20:
Kamleshwar,

Does your beacon show up in the nefConnect app ?

kapil97, 2018-06-20:
Yes, it shows all the information

kapil97, 2018-06-20:
[image]

chegewara, 2018-06-20:
Try to run scan example code and watch logs. You will see if this ibeacon will be discovered and logs will be printed with or without address.

markiv25, 2018-06-20:
where are the log files created in it& where are it stored!

chegewara, 2018-06-20:
Use putty to connect esp32 and print logs. Just remember to change log level to verbose or at least to debug.

markiv25, 2018-06-20:
we are using Arduino IDE not ESP-IDF so please go through the issue again and carefully.

chegewara, 2018-06-20:
Thats fine, you can use arduino-ide. In Tools menu you have Core Debug Level, set it to debug or verbose.
Now you can use putty or serial monitor to see logs, like i mention before.

chegewara, 2018-06-20:
You have nice instructions here:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md#switching-on-debugging

markiv25, 2018-06-20:
thank you !! i had seen the logs ,
though could you help me to get the adddress ! & the it is ignorred to!
[image]

chegewara, 2018-06-20:
As you can see everything works, beacon is discovered and MAC is printed in logs. Now here you can filter devices by name or by UUID (this is just example how to filter with UUID):
https://github.com/nkolban/ESP32_BLE_Arduino/blob/af865a916795289c8e7e09b091ff2140c33fc3fe/examples/BLE_client/BLE_client.ino#L78

and then get address:
https://github.com/nkolban/ESP32_BLE_Arduino/blob/af865a916795289c8e7e09b091ff2140c33fc3fe/examples/BLE_client/BLE_client.ino#L84

markiv25, 2018-06-20:
What if we don't know the UUID?
Can we get the MAC without it? because i was able to do so in esp-idf

markiv25, 2018-06-20:
i tried the client.ino file as you mentioned though it is unable to get the address

chegewara, 2018-06-20:
How do you know? How did you try to print address to console log when you get it with `pServerAddress = new BLEAddress(advertisedDevice.getAddress());`

markiv25, 2018-06-20:
yes i tried it & hereby is the screenshots
[image]
[image]

u4mzu4, 2018-06-20:
Try this one:
advertisedDevice.getAddress().toString().c_str()
as above in your code, instead of simple getAddress()

chegewara, 2018-06-20:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEAdvertisedDevice.cpp#L490

markiv25, 2018-06-20:
nope nothing !
[image]
[image]

chegewara, 2018-06-20:
Serial.println("Add **%s**", advertisedDevice.getAddress().toString().c_str());

markiv25, 2018-06-20:
nothing again
[image]

markiv25, 2018-06-20:
bhai tujhe kuch nahi ata 
though thanks !

markiv25, 2018-06-20:
thank you friend
4 !

chegewara, 2018-06-20:
then:
`Serial.printf("Add %s", advertisedDevice.getAddress().toString().c_str());`
or
`Serial.println( advertisedDevice.getAddress().toString().c_str());'

markiv25, 2018-06-21:
thank you got it!

hansmbakker, 2018-12-04:
@kapil97 this issue seems solved, could you please close it to keep the issue list clean?

cc @chegewara

closed 2018-12-09 by chegewara