nkolban/esp32-snippets #574: What is the "data32" parameter in setValue(float& data32) function?

pgrlopes, 2018-06-21:
Hello.

I'm reading sensors on my PSOC and transmiting the data to my ESP32 where I created a BLE server to send the info to my phone. The data so far comes in as an array of 4 bytes that I transform into a float, but while trying to use the characteristic.setValue(float&) function I keep getting an error saying that it isn't recognised, and when I checked the BLECharacteristic.h file I found out there is a "data32" passing as a parameter.

What is the data32 and how can I pass it through so I can call the setValue(float&) function?

Than you.

chegewara, 2018-06-21:
Its just a variable name and most likely person that wrote this code used it to point it is 32 bit value.

pgrlopes, 2018-06-21:
So if I do:

float variable;
testCharacteristics.setValue(variable);

It should work? Cause it isn't working, it just says "no matching function for call to 'BLECharacteristic::setValue(float&)'". I'm using the Arduino IDE.

chegewara, 2018-06-21:
How do you create testCharacteristics? If its pointer then you need to acces with -> beside that it should works.

pgrlopes, 2018-06-21:
Yeah it was just an example, the test is created fine because I already got it working with a uint8_t variable before. I just went ahead and checked my BLECharacteristic.h file and I don't have the setValue(float&) function, I must have an older version.

Do you know whats the easiest way to update it? Just replacing the files in the destination folder maybe?

chegewara, 2018-06-21:
You can try, most likely it can works if you just replace .h and .cpp.

pgrlopes, 2018-06-21:
Yeah it worked out. However, im getting weird values being read in my NRF app, maybe the characteristic does not support float values?

chegewara, 2018-06-21:
Im guessing you are getting 4 separate bytes in nRF app because 4 bytes are sent and nRF does not know it is float value. You have 2 options:
- test with your app that will decode those 4 bytes into float value 
- add descriptor that will inform client it is float value.

pgrlopes, 2018-06-21:
I don't actually have my own app, so I can't do that client side I guess, as the idea is to get this working and then send the info to a cloud service of some sort through a gateway. What you are saying is that even though im sending a float value, what's actually being sent are 4 different bytes to form the 32 bit value?

So how would I go about using the descriptor to transform it into a float value in the nRF app?

chegewara, 2018-06-21:
https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/src/BLE2904.cpp

pgrlopes, 2018-06-28:
Well yeah I know how to create the descriptor, my issue is understanding how it affects the way I'm reading the values, cause according to the Bluetooth documentation, the environmental sensing characteristic has a temperature parameter that has a format of "sint16", and I'm trying to pass a float for higher precision, as the int16 only gives me a round number.

How would a descriptor help in this case? My environmental sensing characteristic value would still be wrong right?