nkolban/esp32-snippets #581: not really issue, but help needed the UUID 2901/2901

mitchjs, 2018-06-25:
Neil, first off to your wonderful work here
be completely lost without it... aka not even using an ESP32!

so.. I created a Characterestic descriptor for 0x2901
user description, and added it to my characteristics, it shows up
but it shows as READ | WRITE

mitchjs, 2018-06-25:
oh the rest of the code to add it to the characteristic

```
//MyLEDControlDescription.setAccessPermissions(ESP_GATT_PERM_READ);
 MyLEDControlDescription.setValue("Led Control");
 MyLEDControlCharacteristic.addDescriptor(&MyLEDControlDescription);
```

chegewara, 2018-06-25:
You can set this when you create characteristic:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/SampleServer.cpp#L26-L31

mitchjs, 2018-06-25:
that still confuses me, the characteristic is read/write
its the descriptor 0x2901, that needs to be read only
maybe its an issue with Nordics nRF Connect
the 0x2902 one comes up read only

chegewara, 2018-06-25:
Not really, 0x2902 is read/write. When you want to turn on/off notifications/indications then you write to 0x2902 descriptor.

https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.gatt.characteristic_user_description.xml
> If the Writable Auxiliary bit of the Characteristics Properties is set then this descriptor is written. Only one User Description descriptor exists in a characteristic definition.

Im not sure i understand this properly(english is not my native lang), but in my understanding descriptor is writable if characteristic is writable (correct me please if im wrong).

mitchjs, 2018-06-25:
got it sorted.. its was all how nRF connect presented the info
(incorrectly)
switched to LightBlue on my ipad… and the info is correct
in fact it shows it MUCH BETTER

[image]

mitchjs, 2018-06-25:
you can see my USER DESCRIPTIONS "Led Control" and & "Config"
odd that the order is not what I thought it would be...
maybe the last ones I add to the service comes in 1st?
I would think maybe sort by guid(which is hard to do)

one more odd, thing... on the discovery screen, shows no services
[image]

chegewara, 2018-06-25:
Last issue is caused by advertisement packet size which is 31 bytes and your name is very long so there is no space to advertise 128 bit service UUID. To fix this use short name (5 characters) or advertise name only in scan response advertising packet. 
Probably order is like you suspect by adding order which is connected with attribute handle.

mitchjs, 2018-06-26:
I don't think that's it I did 3chars as the name, and it still shows NO services
something is not right in the ADVERTISEMENT DATA
I bet something is missing

one thing I see, the alta's advertisement data has more
my devices has:
1. device is connectable
2. local name
3. tx power level

the Alta has:
1. device is connectable
2. local name
3. service data [
4. service UUIIDs (and shows 1 uuid)
5. tx power level

neil any idea?

mitchjs, 2018-06-26:
ok, I did this

mitchjs, 2018-06-26:
i get it, its not just 31 bytes of data to consider, you got the header bytes
so its cut down a few bytes... 
so with a short name as "DSLED"
also i see the same data used again for the scan response
that explains the UUID showing up 2x

any easy way to not have it do a scan response
other than i comment it out in start()

or is this really not a big deal... and im letting it bother me how it looks in LightBlue Explorer

chegewara, 2018-06-26:
Its no big deal unless you want to use scan resp to advertise other data. But there is class and proper functions to do this. You can also try to use BLEAdvertisementData class to set empty data (never been trying it):
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEAdvertising.cpp

hansmbakker, 2018-12-15:
@mitchjs there has been no activity in this thread for several months. If your issue is solved, could you please close it to keep the issue list clean?
cc @chegewara

closed 2018-12-30 by chegewara