nkolban/esp32-snippets #584: Changing BLE advertisement type

vuanh-minh, 2018-06-26:
Hello,

I am actually using BLE with ESP32 to interact between an application and a server.

I want the advertising to be in **directed** mode, to connect to a specific adress. I saw that different modes exist : 

```
/// Advertising mode
typedef enum {
    ADV_TYPE_IND                = 0x00,
    ADV_TYPE_DIRECT_IND_HIGH    = 0x01,
    ADV_TYPE_SCAN_IND           = 0x02,
    ADV_TYPE_NONCONN_IND        = 0x03,
    ADV_TYPE_DIRECT_IND_LOW     = 0x04,
} esp_ble_adv_type_t;
```

But how can i change my advertisement to ADV_TYPE_DIRECT_IND_HIGH mod, and how can i specify the address of the device i want to connect ? 

Thank you

Thank you

chegewara, 2018-06-26:
Hi,
im afraid its not possible at the moment in this stack, it is hardcoded.

chegewara, 2018-06-26:
Try to read about this `ADV_FILTER_ALLOW_SCAN_ANY_CON_ANY`:
https://esp-idf.readthedocs.io/en/latest/api-reference/bluetooth/esp_gap_ble.html#_CPPv220esp_ble_adv_filter_t

vuanh-minh, 2018-06-27:
I tried to implement whitelist address in my program with : 
`
BLEDevice::whitelistadd(BLEAddress)`
and
`pAdvertising->setScanFilter(true,true);`

However, i still can see the device in my phone, with the Bluetooth menu ( on Android ) or with nRf connect. Is that normal ?

chegewara, 2018-06-27:
You will see it, but can you see scan response or can you connect esp32 from device that is not whitelisted?

vuanh-minh, 2018-06-28:
It's seems to work, but not all the time : Sometimes, the client cannot scan or connect to the ESP32, but when i reset the ESP32 server, the client ( which is constantly scanning and connect for a specific UUID ) can connect to the ESP32 again, that's weird. 

So i tried to flash the server with SampleServer.cpp, just adding the line to scan whitelist address : `pAdvertising->setScanFilter(true,true)`, and it seems to work correctly : even if i reset the server, the client cannot see it or connect to it.

vuanh-minh, 2018-06-28:
Nevermind, now it is working fine ( it was due to a bug in my code ) : when i set setScanFilter(false,false), devices can see or connect to the server, and when i set to setScanFilter(true,true), no one can see it.

The problem is : even the device with the address in the white list cannot see the server, and cannot connect to it. Do you know why ?

This is how my program works : i want the server to memorize the client adress at it first connection, to advertise only to it after bonding. 

-At it first connection, we store the device's address with : 
`BLEDevice::whiteListAdd(pClientAddress)`

-At Server booting, if the ESP32 is paired with a device, _scanRequestWhitelistOnly_ & _connectWhitelistOnly_ are set to **true**, to advertise only to the bonded device. If the server is not paired yet, those two parameters are set to **false**.

MehmetCagriK, 2018-06-30:
@vuanh-minh I had the same problem with advertisement type, you may want to check how I modified the given library and made it work somehow for discoverable/non-connectable. You may create a solution for directed advertising problem of yours.

vuanh-minh, 2018-06-30:
> @vuanh-minh I had the same problem with advertisement type, you may want to check how I modified the given library and made it work somehow for discoverable/non-connectable. You may create a solution for directed advertising problem of yours.

It could be great if you have an example of your code, how you made discoverable/non-connectable mode ! Im a currently working on it, but i don't really know how to do it.

MehmetCagriK, 2018-06-30:
@vuanh-minh Yeah, I forgot to link to my issue: #589

hansmbakker, 2018-12-15:
@vuanh-minh it seems that @MehmetCagriK gave you some guidance. Can this issue then be closed? That way we can keep the issue list clean so that it is easier to see which issues still need attention.

cc @chegewara

closed 2018-12-30 by chegewara