nkolban/esp32-snippets #586: Need help with the Ultrasonic HC-SR04 Sensor interfacing with ESP32

prashanthumbre4, 2018-06-27:
Hello,
I want to interface Ultrasonic HC-SR04 Sensor with ESP32. I am using eclipse and esp-idf SDK.
My current status is I am able to trigger a signal as per the protocol and receive the echo signal.
But I am having difficulty in measuring the time interval between them. The thing is it's a microsecond interval time. What should be the procedure to measure it? Is there any example that I can refer?
Thanks.

chegewara, 2018-06-27:
I think you have more chance to get answer if you ask this question on [forum](esp32.com)

prashanthumbre4, 2018-06-27:
Ok thanks, I will do that.

hansmbakker, 2018-12-04:
@prashanthumbre4 could you please close your issue to keep the issue list clean?

cc @chegewara