nkolban/esp32-snippets #587: Question about type compatibility between platforms

ccfoo242, 2018-06-27:
Hello! Thanks for the BLE library! 

I've been reading up on existing characteristics that might be useful to me and I wonder if there are problems translating an arduino float or int to something on Android or iOS. 

The Bluetooth standard lists **uint8** and **uint16**, etc for int's. Are these the same, bitwise, as **uint8_t** and **uint16_t** that are found in Arduino.h? 

I see 4 different float types defined for bluetooth:
float32 - IEEE-754 32-bit floating point 
float64 - IEEE-754 64-bit floating point
SFLOAT - IEEE-11073 16-bit 
FLOAT - IEEE-11073 32-bit

I believe I read that arduino 'float' is IEEE-754, correct?

Are these float types compatible with Android and Ios float? I don't think I will be using float right now, but would like to know for the future.

Perhaps I am over thinking this. I just recall a long time ago dealing with converting Intel INT to Motorola INT.

ccfoo242, 2018-06-27:
Reading further I see I can call htons() or htonl() from the arduino/esp32 side to get it in network/big endian order for ints. And ntohs() or ntohl() to convert back.

chegewara, 2018-06-27:
If you have good android/ios client then this can help:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLE2904.cpp

closed 2018-06-29 by ccfoo242

bilogic, 2024-11-02:
is there a built-in way to convert to and from SFLOAT?

chegewara, 2024-11-04:
Probably not, but its easy to do it. Just cast the value to float.