nkolban/esp32-snippets #588: How to update connection parameters from BLEServer

panargirakis, 2018-06-29:
Hello!

I am trying to **update the minimum and maximum BLE connection interval from a BLEServer**. (Note: I am not referring to the advertising interval)

I am aware that I ESP-IDF provides a function to update the connection parameters such as the min and max connection interval. How can I do this using the Arduino version?

I noticed there is a GAP event handler in BLEUtils.cpp (BLEUtils::dumpGapEvent) but I do not know how to trigger an ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT event.

Any help is appreciated. Thank you.

chegewara, 2018-06-29:
I am afraid there is no such functionality implemented yet, even more, there is no option to use `esp_ble_gap_update_conn_params` on your own, because to do such it is required central/client mac and this is not implemented yet.
https://github.com/nkolban/esp32-snippets/issues/555

panargirakis, 2018-06-29:
Thank you for letting me know.

chegewara, 2018-07-10:
@panargirakis its merged now and you can do it in onConnect() callback.

panargirakis, 2018-07-10:
@chegewara Great, thank you!

chegewara, 2018-07-10:
@panargirakis Just wait for quick fix PR merge

closed 2018-11-13 by chegewara