nkolban/esp32-snippets #590: Capacitive Touch?

andyrbarrow, 2018-07-02:
Hi Neil, 
Thanks so much for publishing your ESP32 book. It's a truly amazing piece of work!
I was wondering if you have done anything with the capacitive touch sensors on ESP32? I've been looking around for a way to do external water level monitoring of a plastic water tank. This is done commercially by sticking two parallel copper strips to the outside of a tank, and the sensor outputs a changing voltage level based on the change in capacitance as water level changes. Everything I've read so far on the capacitive sensors on the ESP32 has been about touch sensors - essentially on-off sensing.
Do you have any thoughts on this?

Thanks again.
Andy

chegewara, 2018-07-02:
Hi,
if i would have to do something like that i would consider 2 options:
- if you dont need very accurate readings, more like step readings, use all 10 pads with different sensing each pad,
- for more accurate readings consider real time adjusting touch sensitivity for 2 pads, one slightly above water level and one slightly below; when upper pad is triggered then move sensitivity level up, when lower pad is no longer triggering then move sensitivity level down (or opposite)

andyrbarrow, 2018-07-02:
I was thinking, after reviewing research by others, that there might be a
way to detect a variation in the sensor, rather than an on-off.

I think a plastic bottle and some copper strips are in my future... Thanks.

ᐧ

hansmbakker, 2018-12-15:
@andyrbarrow there has been no activity in this thread for several months. If your issue is solved, could you please close it to keep the issue list clean?
cc @chegewara

closed 2018-12-30 by chegewara