nkolban/esp32-snippets #591: Weird conversion

robson31, 2018-07-03:
Hi there,

I have a situation where I have a struct of the data that I want to send over from my Arduino device to a PC:
`struct data {
  uint64_t  data1;
  int16_t   data2;
  int16_t   data3;
};
`

After setting up the values for the struct, I have:
`uint8_t dataToBeSend[sizeof(data)];`
`memcpy(&dataToBeSend, &datastruct, sizeof(data));`
`Characteristic->setValue(dataToBeSend, sizeof(data));`

And this works perfectly, until I add more things to the data struct:

`struct data {
  int8_t dataadded;
  uint64_t  data1;
  int16_t   data2;
  int16_t   data3;
};
`

And from now on, on the other side, I receive dataadded as a string, and the rest as integers, which is not the way I wanted it. I would like to keep dataadded as an integer. Any help would be much appreciated. Thanks.

chegewara, 2018-07-03:
Its 2 possible problems:
- its client side issue, because you are sending bunch of bytes and client is converting it to string,
- another possible issue is here: `struct data { int8_t dataadded` <--- signed integer and you are converting it to uint here: `uint8_t dataToBeSend[sizeof(data)];`

chegewara, 2018-07-03:
Just to make sure whats going on use nRF connect and see what is send.

robson31, 2018-07-03:
Thanks for the tip, but the nRF treats the whole thing as string, so it just prints a bunch of weird characters.

chegewara, 2018-07-03:
nRF is printing both as string and hex values.

robson31, 2018-07-03:
Ok, I'm 100% sure it is the library that is sending wrong stuff.
When everything works fine, here's what the data send looks like IN HEX:
0400 0000 0000 0000 3001 88ff

After I change the struct here it is:
01 0000 0000 0000 0004 0000 00

If you add the appropriate amount of bytes to be send it comes as:
01 0000 0000 0000 0004 0000 0000 0000 0030 0188 ff

So yeah... what's the issue?

chegewara, 2018-07-03:
For some reason int8_t value is stored as `01 0000 0000 0000`, but int8_t is just one byte long. Try this:
`int8_t dataToBeSend[sizeof(data)];` or try outside this structure to figure out why int8_t is stored as 7 bytes. Its also possible its some weird internal bytes alligment, because this seems to be alligned to 4bytes: `01 0000 0000 0000 0004 0000 00`, so you can try int16_t instead, or put it at the end in your structure.

robson31, 2018-07-06:
Nice, thanks. I have to say though, I'm kind of confused on how should I associate the descriptor with the characteristic in the code?

On the other note, __attribute__((__packed__)) on the struct seems to pack the bits the way I want, but still treats it as a char.

chegewara, 2018-07-06:
In your case it can be hard to use BLE2904, because you have mixed struct, but you can try with this value:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLE2904.h#L61
`descriptor->setFormat(FORMAT_OPAQUE)` 
Im guessing that client like nRF should treat received value transparently as bunch of bytes.

robson31, 2018-07-06:
This has no effect. Thanks for support.

closed 2018-07-06 by robson31

mitchjs, 2018-07-06:
can I ask how do I apply the BLE2904 class
can I see some example code to create the class and add it to the characteristic

thanks
mitch

chegewara, 2018-07-07:
@mitchjs you can use it just like BLE2902:
```
BLE2904 *desc = new BLE2904();
desc->setFormat(FORMAT_OPAQUE);
characteristic->addDescriptor(desc);
```

Well, there is few more options you can set, but you have to read about it:
https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.gatt.characteristic_presentation_format.xml

mitchjs, 2018-07-07:
ah.. duh... im newish with c++ (cant stant pointers!!)
thanks
ok.. so im going to use that for my byte array
got it all working... 
thanks!